espressif/arduino-esp32 #505: Problems reading SD cards via SPI

APokorny, 2017-07-12:
I am trying to integrate a micro sd card reader via SPI. The most recent commit contained a fix for the SD card initialization. I am curious what type of cards you used for testing?

So far, with the SD/SPI example I tried two different micro SD cards. One is a 64GB sandisk ultra SDXC class 10 card (only formatted with a 4GB fat32 partition). But with the current esp32-idf and arduino-eps32 it fails when sending GO_IDLE_STATE.

The other card I tested is a 'cheap' paradise 16GB SDHC class 10 card. There the initialization procedure succeeds, and it is detected as a SDHC card. Then when the fatfs requests the sector count only zeroes are read. The same happens when f_mount tries to get the first sector to identify the file system/partition.

We do not have to support a broad range of cards. Just one working type would be fine. Which one was working for you?

APokorny, 2017-07-12:
Small correction the card is not detected as SDHC. Also the second sdTransaction(..READ_OCR...) yields zero. From there on all follow up requests fail

melvinpmathew, 2017-07-12:
Which pins did you use for the SPI on the esp32 device?

melvinpmathew, 2017-07-12:
Gives me an error : 
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034
Card Mount Failed

Must be something to do with SPI connections, but I have had this working earlier, now it doesn't want to work :\

APokorny, 2017-07-12:
I am using this one: 
https://cdn.sparkfun.com/assets/learn_tutorials/5/0/7/esp32-thing-graphical-datasheet-v02.png
I used the pins  22 19 18 23 5  and one of the 3.3V and the GND pins. Looking at the code again - it makes use of pin2 - are the pins 2 and 5 somewhat connected? Because I do see the LED go off during the init.. 
An osciloscope showed that the CS LOW flank  somewhat after activity on MOSI ..
Will try again tomorrow.

APokorny, 2017-07-12:
.. I did enable warning logs in esp-idf and added some more traces to see where it bails out depending on the cards I use.. But yeah it ends with Mount Failed..

melvinpmathew, 2017-07-12:
Thanks for the datasheet. I am using the ESP32 dev board, also have it almost the same, except for the CS pin (
[image]
). 
Don't know if its some electrical or faulty wiring, I rewiring the whole setup 3-4 times and finally it worked. Now its able to talk to the SDcard on SPI flawlessly.

APokorny, 2017-07-13:
Yikes .. I used the wrong power pin. The Catalex micro SD Card Adapter neesd 5V instead 3.3V
Both the SD and SDXC card work.

closed 2017-07-13 by APokorny