espressif/arduino-esp32 #507: ubuntu installation error

scitechindian, 2017-07-12:
System: Linux, Info: Linux-4.10.0-26-generic-x86_64-with-Ubuntu-17.04-zesty
Platform: x86_64-pc-linux-gnu
Tool xtensa-esp32-elf-linux64-1.22.0-61-gab8375a-5.2.0.tar.gz already downloaded
Extracting xtensa-esp32-elf-linux64-1.22.0-61-gab8375a-5.2.0.tar.gz
Traceback (most recent call last):
  File "get.py", line 138, in <module>
    get_tool(tool)
  File "get.py", line 103, in get_tool
    unpack(local_path, '.')
  File "get.py", line 56, in unpack
    tfile.extractall(destination)
  File "/usr/lib/python2.7/tarfile.py", line 2081, in extractall
    self.extract(tarinfo, path)
  File "/usr/lib/python2.7/tarfile.py", line 2118, in extract
    self._extract_member(tarinfo, os.path.join(path, tarinfo.name))
  File "/usr/lib/python2.7/tarfile.py", line 2194, in _extract_member
    self.makefile(tarinfo, targetpath)
  File "/usr/lib/python2.7/tarfile.py", line 2235, in makefile
    copyfileobj(source, target)
  File "/usr/lib/python2.7/tarfile.py", line 266, in copyfileobj
    shutil.copyfileobj(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 49, in copyfileobj
    buf = fsrc.read(length)
  File "/usr/lib/python2.7/tarfile.py", line 831, in read
    buf += self.fileobj.read(size - len(buf))
  File "/usr/lib/python2.7/tarfile.py", line 743, in read
    return self.readnormal(size)
  File "/usr/lib/python2.7/tarfile.py", line 758, in readnormal
    return self.__read(size)
  File "/usr/lib/python2.7/tarfile.py", line 748, in __read
    buf = self.fileobj.read(size)
  File "/usr/lib/python2.7/gzip.py", line 268, in read
    self._read(readsize)
  File "/usr/lib/python2.7/gzip.py", line 315, in _read
    self._read_eof()
  File "/usr/lib/python2.7/gzip.py", line 354, in _read_eof
    hex(self.crc)))
IOError: CRC check failed 0x18be049e != 0x6fa1e491L

copercini, 2017-07-12:
looks like a corrupted file
try remove ~/Arduino/hardware/espressif/esp32/tools/dist/xtensa-esp32-elf-linux64-1.22.0-61-gab8375a-5.2.0.tar.gz

and try again =)

copercini, 2017-09-09:
@scitechindian solved?

copercini, 2017-10-13:
Closed due no response =/

closed 2017-10-13 by copercini

koummich, 2018-09-26:
I have problemTraceback (most recent call last):
  File "my_experiments/tp1.py", line 24, in <module>
    trainer.run(MiniBatches(MnistDataset(), batch_size=2))
  File "/home/fati/deepy/deepy/dataset/mnist.py", line 33, in __init__
    self._train_set, self._valid_set, self._test_set = cPickle.load(gzip.open(path, 'rb'))
  File "/usr/lib/python2.7/gzip.py", line 268, in read
    self._read(readsize)
  File "/usr/lib/python2.7/gzip.py", line 315, in _read
    self._read_eof()
  File "/usr/lib/python2.7/gzip.py", line 354, in _read_eof
    hex(self.crc)))
IOError: CRC check failed 0x2acf9dae != 0xd7ebbb4bL

can you help me please! thanks