espressif/arduino-esp32 #508: Using LedControl.h library: avr/pgmspace.h: No such file or directory

grplyler, 2017-07-13:
Hi,

I am trying to use the Arduino LedControl library to drive a `max7219` led matrix display
but when I compile, there is no `avr/pgmspace.h`

Many led control libraries use `pgmspace.h`. Is this available or possible on esp32?

I am using an [Adafruit HUZZAH32 ESP32 Feather Board](https://learn.adafruit.com/adafruit-huzzah32-esp32-feather/overview)

Any advice or direction on this matter would be greatly appreciated.
Thanks!

Ryan

bbx10, 2017-07-13:
Try #include <pgmspace.h>.

grplyler, 2017-07-13:
Yep, that worked. But had to change the code in the `LedControl.h` file in the library not the Arduino sketch.

Added this:
```
#if (defined(__AVR__))
#include <avr\pgmspace.h>
#else
#include <pgmspace.h>
#endif
```

Where `#include <avr\pgmspace.h>` was.

Found this solution on an issue that the ESP8266 had:
https://github.com/SodaqMoja/Sodaq_DS3231/issues/5

closed 2017-08-31 by copercini

gregfox151, 2021-02-21:
Hi, I added to Adafruit_SH1106.cpp
 #if (defined(__AVR__))
#include <avr\pgmspace.h>
#else
#include <pgmspace.h>
#endif
But still get the error:
fatal error: avr/pgmspace.h: No such file or directory

It was working a few weeks ago but not anymore.
Using an Wroom32 and a 128x96 OLED
Tried it with a Arduino mega same story.