espressif/arduino-esp32 #510: How to disable the touch feature

atzanis, 2017-07-14:
It seems that when I get close to the touch pins and digitalRead it returns HIGH, any ideas on how I can disable the touch feature?

copercini, 2017-07-14:
In which pins?

atzanis, 2017-07-14:
I was hoping to disable it for all the pins (the nodeMCU pinout which shows the touchpins http://www.shenzhen2u.com/NodeMCU-32S)

copercini, 2017-07-14:
One solution is use:
`pinMode(myPin, INPUT_PULLDOWN);`
or
`pinMode(myPin, INPUT_PULLUP);`

instead of just INPUT

atzanis, 2017-07-14:
So here is what I've tried so far:
- pinMode(34, INPUT_PULLDOWN);
- pinMode(34, INPUT_PULLUP);
- gpio_pad_select_gpio(GPIO_NUM_34);
   gpio_set_direction(GPIO_NUM_34, GPIO_MODE_INPUT);
   gpio_set_pull_mode(GPIO_NUM_34, GPIO_PULLDOWN_ONLY);
- gpio_pad_select_gpio(GPIO_NUM_34);
   gpio_set_direction(GPIO_NUM_34, GPIO_MODE_INPUT);
   gpio_set_pull_mode(GPIO_NUM_34, GPIO_PULLUP_ONLY);

Here is my code:
```
void setup() {
  pinMode(34, INPUT_PULLDOWN);
  pinMode(13, OUTPUT); 

  // Debug console
  Serial.begin(115200);
  while(!Serial) { ; }
  Serial.println();
  Serial.println();
  delay(500);
}

void loop() {
  if(digitalRead(34)==HIGH){
    digitalWrite(13, HIGH);
  }else{
    digitalWrite(13, LOW);
  }
}
```

Whenever I get close the pin 34, my LED in 13 will still light up.

atzanis, 2017-07-14:
Ok, it seems that GPIO 34 is not even touch. So, I think I just have to have a pulldown resistor on the GPIO.

copercini, 2017-07-14:
"GPIO pins 34-39 are input-only. These pins do not feature an output driver or internal pull-
up/pull-down circuitry."
http://espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf page 39

atzanis, 2017-07-14:
Thanks for your help.

closed 2017-07-14 by atzanis

schieck0, 2023-05-09:
Any solution with that? I also reading high when touch in pins that are not touch. With mode INPUT or INPUT_PULLUP they ever read high even though i dont touch