espressif/arduino-esp32 #511: WiFi crash after 12 hours

mitchsf, 2017-07-14:
This happens repeatedly, I've been testing over the last week.  It is compiled with the debug option set. I think it's crashing when WiFi disconnects, and tries to reconnect. The program basically tests the WiFi connection, then gets NTP time every five minutes. Processframe() is an ISR firing every 10ms. Possibly calling the ISR during a particular WiFi event is causing the crash. 

> 
stime & get gps or ntp
after ntp
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 1 - UNSPECIFIED
Connecting to SSID: Greyhound-2.4
Password: xxxxxxxxxxxxx

stop get wifi
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 8 - ASSOC_LEAVE
Connecting to SSID: Greyhound-2.4
Password: xxxxxxxxxxxxx

stop get wifi
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Register dump:
PC      : 0x400d0fc0  PS      : 0x00060034  A0      : 0x40081e79  A1      : 0x3ffc0be0  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x000000ff  A5      : 0x400860e8  
A6      : 0x00000020  A7      : 0x3ffca0b8  A8      : 0x800810fc  A9      : 0xe00a1400  
A10     : 0x00000000  A11     : 0x3ffc127c  A12     : 0x20000000  A13     : 0x00000001  
A14     : 0x00000400  A15     : 0xffffffff  SAR     : 0x00000014  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d0fc0:0x3ffc0be0 0x40081e79:0x3ffc0c00

CPU halted.

Exception decoder - 

0x400d0fc0: processFrame() at C:\Users\Mitchell\SkyDrive\Arduino\GC-1000ESP/GC-1000ESP.ino line 431
0x40081e79: _xt_lowint1 at xtensa_vectors.o line ?
0x400860e8: _frxt_int_enter at ?? line ?
0x400d0fc0: processFrame() at C:\Users\Mitchell\SkyDrive\Arduino\GC-1000ESP/GC-1000ESP.ino line 431
0x40081e79: _xt_lowint1 at xtensa_vectors.o line ?

haroonahmed97, 2017-07-17:
In my case same error is coming whenever capacitive touch PCB is connected to the esp32 board. I am using WiFi, pubSubClient(MQTT) and Preferences libraries.

lonerzzz, 2017-08-15:
@mitchsf @haroonahmed97 Have you tried with the latest code? I have had WiFi running for days at a time without issue. Please confirm if the issue is still present in your case.

mitchsf, 2017-08-16:
The problem I'm having is not related to WiFi directly. I think it has something to do with WiFi and an ISR fired every 10ms. When I switch to polling to run the function every 10ms, it runs perfectly without crashing, so far for two weeks. I'll check to make sure I have the latest updates and I'll try the ISR again.

mitchsf, 2017-08-19:
I'm using the latest update, and it is still not working. I believe I know when it crashes, now. First, the program connects to the access point, then it starts the ISR. Everything initializes and runs perfectly. When I purposely disconnect the access point the program attempts to connect every ten seconds, After five minutes or so the access point comes up and a successful attempt to connect crashes with the error below. 

uru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Register dump:
PC      : 0x400d1070  PS      : 0x00060034  A0      : 0x40081e75  A1      : 0x3ffc0be0  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x000000ff  A5      : 0x400860e4  
A6      : 0x00000020  A7      : 0x3ffcad10  A8      : 0x8008115c  A9      : 0xe00a1400  
A10     : 0x00000000  A11     : 0x3ffc13d0  A12     : 0x20000000  A13     : 0x00000001  
A14     : 0x00000400  A15     : 0x00000001  SAR     : 0x00000016  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d1070:0x3ffc0be0 0x40081e75:0x3ffc0c00

CPU halted.
Decoding 5 results
0x400d1070: processFrame() at C:\Users\Mitchell\SkyDrive\Arduino\GC-1000ESPV3/GC-1000ESPV3.ino line 508
0x40081e75: _xt_lowint1 at xtensa_vectors.o line ?
0x400860e4: _frxt_int_enter at ?? line ?
0x400d1070: processFrame() at C:\Users\Mitchell\SkyDrive\Arduino\GC-1000ESPV3/GC-1000ESPV3.ino line 508
0x40081e75: _xt_lowint1 at xtensa_vectors.o line ?

tobozo, 2017-08-20:
Some access points have their own NTP service, is yours acting as a captive portal ? 
Also do you `udp.begin()` on connection loss too ?

mitchsf, 2017-08-20:
Thanks for the comments. udp.begin() is executed by the NTP library that I am using. My access point does not run an NTP service.

igrr, 2017-08-20:
Based on the backtrace, this is related to https://github.com/espressif/arduino-esp32/issues/489. Try adding `IRAM_ATTR` to the declaration of your ISR (`processFrame`), and make sure you call only functions declared with `IRAM_ATTR` attribute from it.

mitchsf, 2017-08-20:
Thank you. I"m using the Arduino IDE. This compiles:

void IRAM_ATTR processFrame() {

but errors on timerAttachInterrup with 'processFrame' was not declared in this scope. 

 // intialize timer
  timer = timerBegin(0, 80, true); // timer_id = 0; divider=80; countUp = true;
  timerAttachInterrupt(timer, &processFrame, true); // edge = true
  timerAlarmWrite(timer, 10000, true); //10 ms

The ISR does call digitalWrite. Otherwise the few variables it uses are declared volatile. Please let me know where to go from here. Thanks.

closed 2017-08-22 by mitchsf