espressif/arduino-esp32 #512: Do I have to repeat WiFi.begin() myself?

AndreKR, 2017-07-14:
I'm waking up from deep sleep and trying to connect to WiFi with this code:
```arduino
void onWifiConnected(WiFiEvent_t event)
{
  Serial.printf("Wifi event: %i\n", event);

  if (event != SYSTEM_EVENT_STA_GOT_IP)
    return;

  Serial.println("Connected!");
  goToSleep();
}

void setup() {
  Serial.begin(115200);
  Serial.println("Start");
  WiFi.onEvent(onWifiConnected);
  WiFi.begin(ssid, password);
  delay(10000);
  Serial.println("Timeout");
  goToSleep();
}
```

Most of the time I get:
```
Start
Wifi event: 2
Wifi event: 4
Wifi event: 7
Connected!
```

But sometimes I get:
```
Start
Timeout
```

The strange thing is, whenever I get "Connected", it is within a second or so. If it doesn't happen within the first second, I can be sure to get "Timeout" after 10 seconds. It will never connect after 5 seconds or so.

So I'm wondering... when I call `WiFi.begin()`, does it continuously try to connect for 10 seconds until I send it back to deep sleep? Or does `WiFi.begin()` trigger some kind of one-time action (like a "request to connect") and if that one is lost, it will wait forever?

So then I would have to call WiFi.begin() in a loop every second or something like that?

me-no-dev, 2017-07-17:
I get this thing sometimes too, but have not yet been able to zero on the cause. I get AP and ETH starting fine, but STA does not start properly so it does not connect.

AndreKR, 2017-07-23:
Indeed it seems this is _not_ a problem of having to call WiFi.begin() several times.

I compared the power usage for a number of cases.

Successful connection:
[image]
(My application goes back to deep sleep in case of a successful connection.)

Failure:
[image]

AP unreachable:
[image]

In case of a failure `SYSTEM_EVENT_STA_START` will not be fired while in case of an unreachable Wifi it will.

closed 2019-08-15 by stale[bot]