espressif/arduino-esp32 #515: is esp32 support Boards Manager？

0x1abin, 2017-07-16:


lonerzzz, 2017-08-01:
@0x1abin Can you describe in more detail what you are looking for so there is no ambiguity?

0x1abin, 2017-08-10:
is can work now.

0x1abin, 2017-08-11:
exec: "C:\\Users\\Zibin\\AppData\\Local\\Arduino15\\packages\\M5Stack\\hardware\\esp32\\0.0.1/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++": file does not exist
Error compiling for board M5Stack-Core-ESP32.

It must to manual run the get.exe download the xtensa-esp32-elf compiling tools.
How to set up to automatically download?

copercini, 2017-08-11:
There is a task in [roadmap](https://github.com/espressif/arduino-esp32/projects/1) about it:

> Integration - installation via boards manager (can use the same ESP8266 package json file, so the people that already uses ESP8266 don’t need to include another link, just install ESP32 on boards manager)

Due it is a administrative task, depends more of Espressif than the users can do

tedder, 2018-05-10:
@me-no-dev @copercini perhaps you can lock this issue to prevent the useless +1 noise?

People who are considering leaving a +1: subscribe to the issue to see progress, and add an emoji to the top comment instead of spamming us.

Testato, 2018-05-17:
@0x1abin 
has the Board Manager been implemented?

me-no-dev, 2018-05-17:
it's in implementation stage actually right now. We have been working on it for some days now to ensure that all will be automated and working smooth ;) Maybe next week it will be done

Testato, 2018-05-17:
So why close the issue ?
When it will be released we can close it

tedder, 2018-05-17:
Closed or not, please comment here when it's released. I think a lot of are following this ticket as a proxy for it.

epicfoox, 2018-06-04:
Task is in the "Seeking Contributors" column. This means that it is on hold atm?

jantje, 2018-06-27:
This one works in windows and probably on mac.
I don't know which files I need for linux so that one fails:
https://raw.githubusercontent.com/jantje/arduino-esp32/master/package/package_Espressif_esp32_index.json

jantje, 2018-07-01:
@unmateria Which os did you use?

beegee-tokyo, 2018-07-02:
@jantje I used your package_Espressif_esp32_index.json and it installs the main core of the Arduino-ESP32, but it fails to install the AzureIoT and BLE libraries which are not in the main repo, but link to dependency repos **_`AzureIoT @ 67dfa4f` `BLE @ 7951347`_**. Any way you can extend the json file to get these libraries as well?
Working on Windows 7 with ArduinoIDE 1.8.5

jantje, 2018-07-02:
@beegee-tokyo 
Thanks for the feedback.
I'm in no way connected to espressif nor esp32. I'll be attending a espressif sponsored event with esp32 based badges. So I just want a easy way to get esp32 installed in Arduino IDE and Sloeber.
IMHO these are separate libraries so I think they should be installed separately as well. Arduino IDE has support for installing libraries based on the github url so that should be fine.
For Sloeber I will add them to my libraries.json

beegee-tokyo, 2018-07-02:
@jantje no complain from my side (using platformio anyway). But maybe it should be noted somewhere that these separate libs need to be installed manually.

Thanks for your workaround solution that makes it easier for ArduinoIDE users to install the ESP32 framework.

I will add instructions about how to install missing libraries into my [ARDUINO-ESP32 wiki](https://desire.giesecke.tk/index.php/2018/07/01/add-esp32-to-boards-manager-workaround/) 

Keep up your great work!

me-no-dev, 2018-07-02:
packager is ready. just making sure that there are no breaking bugs (like one that I fixed the other day). As a matter of fact I will cut a prerelease right now ;)

me-no-dev, 2018-07-02:
You can now try the prerelease package using this URL: https://dl.espressif.com/dl/package_esp32_dev_index.json please test and let me know if there are any installation/compilation errors on Windows and Linux :)

PaoloP74, 2018-07-02:
Invalid version found: da31d9d2 (Windows 7 - Arduino IDE 1.8.5)
I think the IDE does not like the ESPtool version string: Incorrect or unsupported format.

me-no-dev, 2018-07-02:
@PaoloP74 installed and compiled fine on win 10... what would be different on win7?

lbernstone, 2018-07-02:
I got two messages with that "invalid version found: da31d9d2" during the boards manager install, but they did not seem to be fatal, and uploads worked just fine.

unmateria, 2018-07-02:
@jantje I'm on Windows 10 and arduino 1.8.5
I have the same problem with AzureIoT @ 67dfa4f BLE @ 7951347, but since I don't use it I didn't looked the cause. If I can help, just tell me :-)

pacav69, 2018-07-03:
@me-no-dev 
I installed on my Mac using IDE 1.8.5, it worked fine

[image]

PaoloP74, 2018-07-03:
@me-no-dev I used, on Windows 7 64 bit, the Arduino IDE zip version for non admin install.
But the second time I uninstalled and reinstalled the core, no warning appeared. Confused.

me-no-dev, 2018-07-03:
@PaoloP74 I'm confused too... let's see if anyone will come up with similar error

Brubacker, 2018-07-03:
Did a quick test on a Win10 x64 machine with Arduino 1.8.5 and it seems to work.

beegee-tokyo, 2018-07-03:
@PaoloP74 @me-no-dev same config, same error, for the installation I run ArduinoIDE as administrator.

Invalid version found: da31d9d2 (Windows 7 - Arduino IDE 1.8.5)

Second installation runs without problems and sketches compile

me-no-dev, 2018-07-03:
@beegee-tokyo maybe it is expecting something like 1.2.3 as version... will do :)

jantje, 2018-07-03:
> Invalid version found: da31d9d2

Is probably when using https://dl.espressif.com/dl/package_esp32_dev_index.json
which contains
`"version": "da31d9d2"`
A version should be something like Major.minor.build-something
Only something can be text and 1.1.1-XX is considered new"er than 1.1.1

me-no-dev, 2018-07-03:
yes, it's what I suspect also. Will change to x.x.x version in the next RC and see if that helps

copercini, 2018-07-28:
Released! https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md

closed 2018-07-28 by copercini