espressif/arduino-esp32 #519: base64.cpp error

wperw, 2017-07-16:
Hi to all.
I'm trying to us the base64 library present into esp32/cores/esp32, but when i use it to encode data received from the serial and output it on another serial or send with a HTTPClient.POST to a server mi sketch crash a lot of time.
I analyzed the stack trace with "ESP Exception Decoder" and the error some times is into the free(buffer) inside base64.cpp and very often the error is in WString library.
Wath do you need for tracking this issue?

P.S. I've modified the main.cpp under esp32/core giving it more stack (16384) but problem continue

lonerzzz, 2017-08-01:
@wperw Can you provide the exact trace for this that you were hitting so that we can get the type of error as well. As for something to try, can you check the size of your stack with the free_rtos commands?

wperw, 2017-08-03:
Hi @lonerzzz in the weekend I'll reproduce the error and try to post the stack trace with the sketch, what is the  freeRTOS command for checking the free stack?

wperw, 2017-08-06:
Hi @lonerzzz

This is the .ino file

```
#include <WiFi.h>
#include <HTTPClient.h>
#include <base64.h>
#include <ArduinoJson.h>

uint64_t chipid;  

#define ssid "<your BSSID>"
#define password "<your Password>"
const char* server = "your NODE RED IP";
const int port = 1880;
const char* endpoint = "/remotizer";

#define USE_SERIAL Serial

const size_t bufferSize = JSON_OBJECT_SIZE(2) + 68300;
DynamicJsonBuffer jsonBuffer(bufferSize);

String UniqueID(uint64_t id)
{
  char tmp_h[9];
  char tmp_l[9];

  sprintf(tmp_h, "%08X", (uint32_t)(id>>32));
  sprintf(tmp_l, "%08X", (uint32_t)(id));

  return String(tmp_h) + String(tmp_l);
}

String uint8ToString(uint8_t* buff, size_t len)
{
  String ret;
  char tmp[2];

  for(int i = 0; i <= len - 1; i++)
  {
    USE_SERIAL.printf("buffer at %d is %02X", i, buff[i]);
    sprintf(tmp, "%02X", buff[i]);
    ret = String(ret + tmp);
  }

  return ret;
}

String uint8ToJSONArray(uint8_t* buff, size_t len)
{
  String ret;
  char tmp[2];

  ret = String("[");
  for(int i = 0; i <= len - 1; i++)
  {
    USE_SERIAL.printf("buffer at %d is %02X", i, buff[i]);
    sprintf(tmp, "%02X", buff[i]);
    if(i < len - 1) ret = String(ret + tmp + ",");
    else ret = String(ret + tmp);
  }
  ret =  String(ret + "]");
  return ret;
}

void setup() {
  USE_SERIAL.begin(115200);
  
  chipid=ESP.getEfuseMac();//The chip ID is essentially its MAC address(length: 6 bytes).
  USE_SERIAL.printf("ESP32 Chip ID = %04X",(uint16_t)(chipid>>32));//print High 2 bytes
  USE_SERIAL.printf("%08X\n",(uint32_t)chipid);//print Low 4bytes.

  USE_SERIAL.print("Connecting to ");
  USE_SERIAL.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
  }

  USE_SERIAL.println("");
  USE_SERIAL.println("WiFi connected");
  USE_SERIAL.println("IP address: ");
  USE_SERIAL.println(WiFi.localIP());

}

void loop() {

  HTTPClient http;

  if(true)
  {
    USE_SERIAL.print("[HTTP] begin...\n");
    http.begin(String("http://") + server + ":" + port + endpoint); //HTTP
    
    size_t len = 36;
    USE_SERIAL.printf("len is %d\n", len);
    char buff[len] = "QWERTYUIOPASDFGHJKLZXCVBNM123456789";

    USE_SERIAL.printf("buff to base64 is 64 is %s\n", String(base64::encode((uint8_t *)buff, len)));

    USE_SERIAL.print("[HTTP] POST...\n");
    http.addHeader(F("Content-Type"), F("application/json"), false, false);
    USE_SERIAL.print(" header added\n");
    USE_SERIAL.print(String(buff) + "\n");//at 68 chars the sting become strange...
    int httpCode = http.POST(String("") + "{\"data\":\"" + String(base64::encode((uint8_t *)buff, len)) + "\"}");//<- some times dies here
    USE_SERIAL.print("...POST DONE");

    if(httpCode > 0) {
            // HTTP header has been send and Server response header has been handled
            USE_SERIAL.printf("[HTTP] POST... code: %d\n", httpCode);

            // file found at server
            if(httpCode == HTTP_CODE_OK) {
                String payload = http.getString();
                JsonObject& root = jsonBuffer.parseObject(payload);
                const char* data = root["data"];
                USE_SERIAL.printf("il dato è %s\n", data);
                String receiveData = String(data);
            }
        } else {
            USE_SERIAL.printf("[HTTP] POST... failed, error: %s\n", http.errorToString(httpCode).c_str());
        }

        http.end();
        USE_SERIAL.printf("Free heap is %d\n", ESP.getFreeHeap()); 
    delay(100);
  }
}
```

this is the node-red code

```
[
    {
        "id": "3e49107f.928de",
        "type": "tab",
        "label": "Flow 1"
    },
    {
        "id": "c76a3e37.a4fdc",
        "type": "http in",
        "z": "3e49107f.928de",
        "name": "",
        "url": "/remotizer",
        "method": "post",
        "upload": false,
        "swaggerDoc": "",
        "x": 101,
        "y": 118,
        "wires": [
            [
                "ffadfd19.ce903",
                "392a6a4a.c0ff26"
            ]
        ]
    },
    {
        "id": "ffadfd19.ce903",
        "type": "debug",
        "z": "3e49107f.928de",
        "name": "",
        "active": true,
        "console": "false",
        "complete": "true",
        "x": 326,
        "y": 95,
        "wires": []
    },
    {
        "id": "392a6a4a.c0ff26",
        "type": "http response",
        "z": "3e49107f.928de",
        "name": "",
        "statusCode": "",
        "headers": {},
        "x": 353,
        "y": 188,
        "wires": []
    }
]
```

and this is the exception tath appear only if i use the base64::encoding

```
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400014fd  PS      : 0x00060030  A0      : 0x800d6188  A1      : 0x3ffccd60  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffcd110  
A10     : 0x00000000  A11     : 0x40082204  A12     : 0x3ffc13f0  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3ffcc380  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400014fd:0x3ffccd60 0x400d6188:0x3ffccd70 0x400d7899:0x3ffcd080 0x400d78d1:0x3ffcd110 0x400d29ac:0x3ffcd150 0x400d1217:0x3ffcd1f0 0x40112a7b:0x3ffcd2f0

CPU halted.

```

Hope this can help

me-no-dev, 2017-08-17:
How about you change this line:
```arduino
int httpCode = http.POST(String("") + "{\"data\":\"" + String(base64::encode((uint8_t *)buff, len)) + "\"}");//<- some times dies here
```
To:
```arduino
String payload = "{\"data\":\"";
payload += base64::encode((uint8_t *)buff, len);
payload += "\"}";
int httpCode = http.POST(payload);
```

Also:
```arduino
size_t len = 36;
char buff[len] = "QWERTYUIOPASDFGHJKLZXCVBNM123456789";
```
Can become:
```arduino
const char * buff = "QWERTYUIOPASDFGHJKLZXCVBNM123456789";
size_t len = strlen(buff);
```

lonerzzz, 2017-08-25:
@wperw Any update on the issue?

wperw, 2017-08-26:
Hi @lonerzzz and @me-no-dev , I was busy for work.
Today, I've modified the code like suggested by @me-no-dev and the code now is like this:

```Arduino
#include <WiFi.h>
#include <HTTPClient.h>
#include <base64.h>
#include <ArduinoJson.h>

uint64_t chipid;  

#define ssid "<your BSSID>"
#define password "<your Password>"
const char* server = "your NODE RED IP";
const int port = 1880;
const char* endpoint = "/remotizer";

#define USE_SERIAL Serial

const size_t bufferSize = JSON_OBJECT_SIZE(2) + 68300;
DynamicJsonBuffer jsonBuffer(bufferSize);

String UniqueID(uint64_t id)
{
  char tmp_h[9];
  char tmp_l[9];

  sprintf(tmp_h, "%08X", (uint32_t)(id>>32));
  sprintf(tmp_l, "%08X", (uint32_t)(id));

  return String(tmp_h) + String(tmp_l);
}

String uint8ToString(uint8_t* buff, size_t len)
{
  String ret;
  char tmp[2];

  for(int i = 0; i <= len - 1; i++)
  {
    USE_SERIAL.printf("buffer at %d is %02X", i, buff[i]);
    sprintf(tmp, "%02X", buff[i]);
    ret = String(ret + tmp);
  }

  return ret;
}

String uint8ToJSONArray(uint8_t* buff, size_t len)
{
  String ret;
  char tmp[2];

  ret = String("[");
  for(int i = 0; i <= len - 1; i++)
  {
    USE_SERIAL.printf("buffer at %d is %02X", i, buff[i]);
    sprintf(tmp, "%02X", buff[i]);
    if(i < len - 1) ret = String(ret + tmp + ",");
    else ret = String(ret + tmp);
  }
  ret =  String(ret + "]");
  return ret;
}

void setup() {
  USE_SERIAL.begin(115200);
  
  chipid=ESP.getEfuseMac();//The chip ID is essentially its MAC address(length: 6 bytes).
  USE_SERIAL.printf("ESP32 Chip ID = %04X",(uint16_t)(chipid>>32));//print High 2 bytes
  USE_SERIAL.printf("%08X\n",(uint32_t)chipid);//print Low 4bytes.

  USE_SERIAL.print("Connecting to ");
  USE_SERIAL.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
  }

  USE_SERIAL.println("");
  USE_SERIAL.println("WiFi connected");
  USE_SERIAL.println("IP address: ");
  USE_SERIAL.println(WiFi.localIP());

}

void loop() {

  HTTPClient http;

  if(true)
  {
    USE_SERIAL.print("[HTTP] begin...\n");
    http.begin(String("http://") + server + ":" + port + endpoint); //HTTP
    
    const char * buff = "QWERTYUIOPASDFGHJKLZXCVBNM123456789";
    size_t len = strlen(buff);

    USE_SERIAL.printf("buff to base64 is 64 is %s\n", String(base64::encode((uint8_t *)buff, len)));

    USE_SERIAL.print("[HTTP] POST...\n");
    http.addHeader(F("Content-Type"), F("application/json"), false, false);
    USE_SERIAL.print(" header added\n");
    USE_SERIAL.print(String(buff) + "\n");//at 68 chars the sting become strange...
    int httpCode = http.POST(String("") + "{\"data\":\"" + String(base64::encode((uint8_t *)buff, len)) + "\"}");//<- some times dies here
    USE_SERIAL.print("...POST DONE");

    if(httpCode > 0) {
            // HTTP header has been send and Server response header has been handled
            USE_SERIAL.printf("[HTTP] POST... code: %d\n", httpCode);

            // file found at server
            if(httpCode == HTTP_CODE_OK) {
                String payload = http.getString();
                JsonObject& root = jsonBuffer.parseObject(payload);
                const char* data = root["data"];
                USE_SERIAL.printf("il dato è %s\n", data);
                String receiveData = String(data);
            }
        } else {
            USE_SERIAL.printf("[HTTP] POST... failed, error: %s\n", http.errorToString(httpCode).c_str());
        }

        http.end();
        USE_SERIAL.printf("Free heap is %d\n", ESP.getFreeHeap()); 
    delay(100);
  }
}
```

but the problem persist, and after 3 minuts the software crash with this error:

```
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400014fd  PS      : 0x00060e30  A0      : 0x800d6164  A1      : 0x3ffccd90  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffcd140  
A10     : 0x00000000  A11     : 0x40082204  A12     : 0x3ffc13f0  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3ffcc380  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400014fd:0x3ffccd90 0x400d6164:0x3ffccda0 0x400d7875:0x3ffcd0b0 0x400d78ad:0x3ffcd140 0x400d2988:0x3ffcd180 0x400d11f7:0x3ffcd220 0x40112a57:0x3ffcd2f0

CPU halted.

```

I goed deeper and used the [ESP Exception decoder](https://github.com/me-no-dev/EspExceptionDecoder) from @me-no-dev (it is really a great work!) and this is the reslt of the Exception decoder

```
Decoding 11 results
0x40082204: _malloc_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/./syscalls.c line 26
0x400d6164: _svfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c line 1529
0x400d7875: _vsnprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c line 73
0x400d78ad: vsnprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c line 42
0x400d2988: Print::printf(char const*, ...) at C:\Program Files (x86)\Arduino\hardware\espressif\esp32\cores\esp32/Print.cpp line 227
0x400d11f7: loop() at C:\Users\Telemaco\Documents\Progetti\Espressif\Arduino\test\test.ino/test.ino.ino line 119
0x40112a57: loopTask(void*) at C:\Program Files (x86)\Arduino\hardware\espressif\esp32\cores\esp32/main.cpp line 18 (discriminator 1)
```
It seems to be a problem with a malloc and some function of newlib.

wperw, 2017-09-28:
@lonerzzz @me-no-dev 
Hi, any news?

me-no-dev, 2017-09-29:
sorry I might have missed your previous comment. What is the last line printed before the exception happens?

wperw, 2017-09-30:
Hi @me-no-dev  these are the last lines before de exception
```
[HTTP] begin...
buff to base64 is 64 is Ph⸮??
[HTTP] POST...
 header added
QWERTYUIOPASDFGHJKLZXCVBNM123456789
...POST DONE[HTTP] POST... code: 200
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400014fd  PS      : 0x00060230  A0      : 0x800d6164  A1      : 0x3ffccd90  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffcd140  
A10     : 0x00000000  A11     : 0x40082204  A12     : 0x3ffc13f0  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3ffcc380  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400014fd:0x3ffccd90 0x400d6164:0x3ffccda0 0x400d7875:0x3ffcd0b0 0x400d78ad:0x3ffcd140 0x400d2988:0x3ffcd180 0x400d11f7:0x3ffcd220 0x40112a57:0x3ffcd2f0

CPU halted.
```

me-no-dev, 2017-10-01:
change ```const char* data = root["data"];``` to ```String data = root["data"];```

wperw, 2017-10-01:
Hi @me-no-dev , it's working for more than 1 hour.
I Can't understand why now it work.
What was the problem?

me-no-dev, 2017-10-01:
did you make the change I outlined?

wperw, 2017-10-01:
yes

me-no-dev, 2017-10-02:
that is the difference :) String is an object and const char * is not. ArduinoJson returns String

closed 2017-10-02 by me-no-dev