espressif/arduino-esp32 #520: mDNS_Web_Server.ino: Invalid service type

antonio-fiol, 2017-07-17:
mDNS_Web_Server.ino includes the line:
`    MDNS.addService("http", "tcp", 80);`
This causes avahi-browse on Linux to fail with "Invalid service type". Changing that line to the following fixes the issue.
`    MDNS.addService("_http", "_tcp", 80);`

sansillusion, 2017-08-29:
Thanks ! Now I need to know how to get chrome to work with it like my D-link range boosters
ie: open http://lumiere.local
but it does not work.
In ZeroConf Browser for android my D-Links have a extra listing under (_DHNAP) ?
is _DHNAP what's needed and does it exist under esp32 ?

antonio-fiol, 2017-08-29:
@sansillusion as far as I can tell, you do actually not need any service entry for a browser to be able to open said URL. The service entry is only for discoverability.
To set the host name and make it available on the network for mDNS resolution, in the same example file, the line
```if (!MDNS.begin("esp32")) {```
allows you to find it on esp32.local.
I imagine you already changed that to
```if (!MDNS.begin("lumiere")) {```

sansillusion, 2017-08-29:
@antonio-fiol Yes I did already changed it.
I kind of got it working by installing Apple Bonjour V3 (extracted from iTunes installer) but it leaves me confused because : 
1: My D'Link range extenders did not need Bonjour installed to work
2: It does not work with Edge only in chrome (perfect for me but some people use Edge)

But at least it somwhat works now

antonio-fiol, 2017-08-29:
Sorry, I can't help further: I do not use Windows so I never had to fiddle
with mDNS on Windows.

sansillusion, 2017-08-29:
@antonio-fiol  Thanks a lot, at least it somewhat works now.
I guess the library works with the Bonjour protocol and windows and chrome must have a different implementation of the mDNS protocol.

me-no-dev, 2017-08-30:
new mdns is coming soon ;) it follows all official rules so it should be fine.

sansillusion, 2017-08-30:
@me-no-dev Thank-you for the info !!! You rock !

copercini, 2018-02-13:
@sansillusion is it working with the new mdns?

sansillusion, 2018-02-13:
I cant test right now, it works with apple bonjour installed. But I hope it does without having to install apple bonjour like for the esp8266.

I will try later tonight to see and report back.

sansillusion, 2018-02-13:
Still seem the same here, Will not wirk without installing apple Bonjour V3, it does not show up in chrome MDNS browser nor in edge

sansillusion, 2018-02-22:
I seem to have had some weird issue with my setup.
I noticed that even if git GUI sometimes had some updates when doing the update procedures, I had many different files (old versions) than what is shown on GitHub. So I manually downloaded the repo and copied it over my (documets/arduino/hardware/espressif/esp32) folder, than ran get.exe and "$ git submodule update --init --recursive" in git bash and then recompiled my code ( there is a warning about invalid BLE library wich I dont use anyway) and my code jumped from around 30% of program storage to 62% but everything seems to work better even mDNS !!! it shows up in chrome mdns browser plugin YAY !

I guess I will have to cleanup my PC and redo the whole esp32 setup.

closed 2018-04-19 by everslick