espressif/arduino-esp32 #523: Guru Meditation Error on WifiClientSecure

mceyzar, 2017-07-17:
Using WifiClientSecure would sometime yield in:
`
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered 
Register dump:
PC      : 0x4000bff0  PS      : 0x00060236  A0      : 0x80085174  A1      : 0x3ffcd110  
A2      : 0x00000000  A3      : 0x00060223  A4      : 0x00060220  A5      : 0x00000040  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x8008406d  A9      : 0x3ffcd1b0  
A10     : 0x00000003  A11     : 0x00060223  A12     : 0x00060223  A13     : 0x00000a00  
A14     : 0x00000002  A15     : 0x3ffcbf00  SAR     : 0x0000001f  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4000bff0:0x3ffcd110 0x40085174:0x3ffcd1e0 0x400843e8:0x3ffcd200 0x40082294:0x3ffcd240 0x40082336:0x3ffcd270 0x400f97c9:0x3ffcd290 0x400f99ad:0x3ffcd2b0 0x400f9c94:0x3ffcd2f0 0x400f9e1f:0x3ffcd330 0x400f9cbc:0x3ffcd380 0x400fde78:0x3ffcd3c0 0x400fe222:0x3ffcd3f0 0x400fe342:0x3ffcd470 0x400f96de:0x3ffcd490 0x400f9a84:0x3ffcd4c0 0x400f3213:0x3ffcd510 0x400f3f70:0x3ffcd540 0x400f4150:0x3ffcd980 0x400f1f1a:0x3ffcd9b0 0x400f1c5e:0x3ffcd9e0 0x400f1dd2:0x3ffcda00 0x400f8bf4:0x3ffcda40 0x400f8e10:0x3ffcdae0 0x400f9056:0x3ffcdb80 0x400f7cba:0x3ffcdbc0 0x4010a88f:0x3ffcdc00 0x400f5ad8:0x3ffcdc20 0x400f5b18:0x3ffcdc40 0x400d2ee5:0x3ffcdc60 0x400d2bc0:0x3ffcdec0 0x400d2c30:0x3ffcdee0 0x400d2c4e:0x3ffcdf10 0x400d27bf:0x3ffcdf30 0x4013374b:0x3ffce060

CPU halted.
`

decoded to:

`Decoding 52 results
0x40080000: _iram_start at ?? line ?
0x40080034: _iram_start at ?? line ?
0x40080000: _iram_start at ?? line ?
0x40080034: _iram_start at ?? line ?
0x40080000: _iram_start at ?? line ?
0x40080034: _iram_start at ?? line ?
0x40080000: _iram_start at ?? line ?
0x40080034: _iram_start at ?? line ?
0x40080000: _iram_start at ?? line ?
0x40080034: _iram_start at ?? line ?
0x40085174: vTaskExitCritical at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4420
0x400843e8: xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1992
0x40082294: lock_release_generic at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/./locks.c line 197
0x40082336: _lock_release at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/./locks.c line 203
0x400f97c9: esp_mpi_release_hardware at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/port/esp_bignum.c line 308
0x400f99ad: mpi_mult_mpi_failover_mod_mult at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/port/esp_bignum.c line 308
0x400f9c94: mbedtls_mpi_mul_mpi at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/port/esp_bignum.c line 483
0x400f9e1f: mpi_mult_mpi_overlong at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/port/esp_bignum.c line 623
0x400f9cbc: mbedtls_mpi_mul_mpi at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/port/esp_bignum.c line 489
0x400fde78: mbedtls_mpi_mul_int at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/bignum.c line 2152
0x400fe222: mbedtls_mpi_div_mpi at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/bignum.c line 2152
0x400fe342: mbedtls_mpi_mod_mpi at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/bignum.c line 2152
0x400f96de: calculate_rinv at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/port/esp_bignum.c line 308
0x400f9a84: mbedtls_mpi_exp_mod at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/port/esp_bignum.c line 360
0x400f3213: mbedtls_rsa_public at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/rsa.c line 1607
0x400f3f70: mbedtls_rsa_rsassa_pkcs1_v15_verify at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/rsa.c line 1607
0x400f4150: mbedtls_rsa_pkcs1_verify at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/rsa.c line 1607
0x400f1f1a: rsa_verify_wrap at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/pk_wrap.c line 56
0x400f1c5e: mbedtls_pk_verify at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/pk.c line 361
0x400f1dd2: mbedtls_pk_verify_ext at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/pk.c line 361
0x400f8bf4: x509_crt_verify_top at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/x509_crt.c line 1251
0x400f8e10: x509_crt_verify_child at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/x509_crt.c line 1251
0x400f9056: mbedtls_x509_crt_verify_with_profile at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/x509_crt.c line 1251
0x400f7cba: mbedtls_ssl_parse_certificate at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/ssl_tls.c line 6123
0x4010a88f: mbedtls_ssl_handshake_client_step at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/ssl_cli.c line 3303
0x400f5ad8: mbedtls_ssl_handshake_step at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/ssl_tls.c line 6123
0x400f5b18: mbedtls_ssl_handshake at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mbedtls/library/ssl_tls.c line 6123
0x400d2ee5: start_ssl_client(sslclient_context*, unsigned int, unsigned int, char const*, char const*, char const*) at /Users/sakana/Documents/Arduino/hardware/espressif/esp32/libraries/WiFiClientSecure/src/ssl_client.cpp line 155
0x400d2bc0: WiFiClientSecure::connect(IPAddress, unsigned short, char const*, char const*, char const*) at /Users/sakana/Documents/Arduino/hardware/espressif/esp32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp line 196
0x400d2c30: WiFiClientSecure::connect(char const*, unsigned short, char const*, char const*, char const*) at /Users/sakana/Documents/Arduino/hardware/espressif/esp32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp line 196
0x400d2c4e: WiFiClientSecure::connect(char const*, unsigned short) at /Users/sakana/Documents/Arduino/hardware/espressif/esp32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp line 196
0x400d27bf: loop() at /Users/sakana/Documents/Arduino/CAMActivator/CAMActivatorLCD.ino/CAMActivatorLCD.ino.ino line 256
0x4013374b: loopTask(void*) at /Users/sakana/Documents/Arduino/hardware/espressif/esp32/cores/esp32/main.cpp line 18 (discriminator 1)`

copercini, 2017-07-17:
It's looks like something in your/website certificates...

mceyzar, 2017-07-17:
The thing is, every other time it connects fine! (about 50% success rate)
Also connecting with openssl s_client to the server never fails.

zjykymt, 2017-08-01:
@mceyzar 
I want to learn how you do the decoding of the first part of the data?
Looking forward to your reply, thank you for your guidance.

copercini, 2017-08-01:
It's an out of stack problem

Just increase 4096 value to a bigger one (8192 is ok) here: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/main.cpp#L25

lonerzzz, 2017-08-15:
@mceyzar Were you able to work past the issue by increasing the stack size?

copercini, 2017-08-31:
The stack of loop task was increased by default here https://github.com/espressif/arduino-esp32/commit/4495659ac53bcdc4f8e59d6918b3e472903d63a7 

This solves until many new features need an extra addition and so it goes....

closed 2017-08-31 by copercini