espressif/arduino-esp32 #524: SD Card errors

job-1994, 2017-07-18:
I am trying to set up the basic sd card operation, and using the begin() function to initialize the SD card, however it continuously comes up with errors. After constantly checking the wiring and such, the code finally settled on this error:

*[W][sd_diskio.cpp:143] sdCommand(): crc error
[W][sd_diskio.cpp:479] ff_sd_initialize(): APP_OP_COND failed: 1
[E][sd_diskio.cpp:721] sdcard_mount(): f_mount failed 0x(3)
Card Mount Failed*

Using the following code:
`   SPI.begin(18,19,23);
    if(!SD.begin(5)){
        Serial.println("Card Mount Failed");
        return;
    }`

The wiring follows the recommended wiring diagram in the Sample Wiring Diagram provided

Is the issue in the model of the SD card? I have formatted a 32GB microSD HC card to FAT32.

jackfury413, 2017-07-29:
Same to me,
I've tried with micro SD card module and also wiring the card directly to the SPI pin (via a SD card adapter) and still cannot mount.
Don't know what to to now.
My cards are 4GB & 8GB FAT32

jackfury413, 2017-07-29:
I've tested with another new Sandisk SDHC1 card (different with SDHC) and it works like a charm.
Not sure if this is a bug in the library ?

me-no-dev, 2017-08-01:
Whatever it is, I have yet to stumble upon such card here :( I tried cards from 1GB to 32GB and all mount well.

me-no-dev, 2017-08-01:
you can try to comment "goto" line here: https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/src/sd_diskio.cpp#L502 and see if that will help

jackfury413, 2017-08-02:
Just tried as your suggestion, and the result is as below:

`[W][sd_diskio.cpp:143] sdCommand(): crc error
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:479] ff_sd_initialize(): APP_OP_COND failed: 255
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:490] ff_sd_initialize(): READ_OCR failed: FF8000
[E][sd_diskio.cpp:721] sdcard_mount(): f_mount failed 0x(3)`

marshalltaylorSFE, 2017-08-02:
@jackfury413 I'm experiencing some SD difficulty which may be related... can I ask what pin you're using for chip select?

I've found (on the sparkfun esp32 thing) that pin 2 can't be used because it interferes with loading, pin 5 won't work, but pin 4 (a randomly chosen gpio) does.

Keep in mind I've only been looking at this for a few hours.

Thanks

capedra, 2017-08-03:
@marshalltaylorSFE not working with pin 4/5 on MH-ET LIVE ESP32 Minikit.

capedra, 2017-08-03:
@marshalltaylorSFE hello again, thank you for the tip. it mounted successfully until I decided to press the reset button and it does not mount anymore.

jackfury413, 2017-08-03:
Hi @marshalltaylorSFE , my wiring diagram is the same as the example.
@capedra It seems like your initialization process is not quite stable. According to ELM Chan (FATFS author), the minimum SPI connection of SD card requires a pull up resistor on MISO line. Please try it.

capedra, 2017-08-03:
@jackfury413 ok, i'm going to try that.
for now, I tested it using micropython and it didn't work as well. it shows an oserror that there is no SD card on tested CS pins 4/5/33 and also tested 3V3/5V. it convinces me even more than it's a hardware issue.

capedra, 2017-08-03:
@jackfury413 Still not working on MicroPython.

jackfury413, 2017-08-03:
@capedra do you use any SD card module ?

clowrey, 2017-08-03:
My SD card has been working fine for a few weeks. Be careful you don't have any other peripherals setup on the same pins. I originally had LEDC running on one of the same pins by accident which of course made it not work.
Also I have seen some SD breakout boards that put 1K resistors in series with some of the datalines and heard this can cause problems just something you might want to check for. I'm using all the default SPI pins.

This person has compiled an even more verbose pinout diagram that includes the SPI pins, it is not the top images but about 2/3rd of the way down the page. These are the pins I'm using.

https://gojimmypi.blogspot.com/2017/03/jtag-debugging-for-esp32.html

G6EJD, 2018-02-03:
Just to confirm, if your using the MH-T Live ESP32 board, or any ESP32 board with an external SD-Card reader, you must ensure there is a pull-up resistor on the MISO line, typically 1k to 4k7 will do. If you fit this and then use the MH-T Live ESP32 board and plug in a Wemos SD-Card adapter/shield, add the resistor and it all works. Use these settings in your sketch:
SPI.begin(18,19,23); 
if(!SD.begin(5)) { Serial.println("Card Mount Failed");}

G6EJD, 2018-02-03:
pinMode(23,INPUT_PULLUP); solves the problem too, no external resistor needed.

everslick, 2018-04-19:
This issue is closed, because it looks as if it is not a bug or problem with the ESP32 Arduino core or its support libraries. For general API usage questions or help on specific coding challenges, please visit the [arduino-esp32 Gitter channel](https://gitter.im/espressif/arduino-esp32). If you feel this issue was closed in error, reopen it and comment, why you think this is a bug in the Arduino-Core.

closed 2018-04-19 by everslick

dalwinder1578singh, 2019-07-15:
i have same issue it started giving error of   } required at the end of in put, even set up is empty and loop is empty, as soon as i include  arduino esp masterzip that error starts poping up when i delete this sketch is fune. surly it has bug but how to fix it i do not know.if any body has clue , or alternative library. thanl
ks in advance if some one can help.

he3117, 2020-05-11:
for anybody else that has the same problem with the micro SD module you should connect the VCC pin of micro SD card module to pin Vin of ESP32 not 3.3 V out.

Naphier, 2020-06-21:
@he3117 thank you!! I wonder how much confusion happens due to this? I think a lot of the Arduino micro SD card modules like this one:
[image]
actually seem to need to be powered by VIN / 5v channel with my ESP32 WROOM 32 / devkit v1.
I'm not sure if it's the breakout board or not. I'm pretty sure I've powered these breakout boards with 3v3 before with Nanos.

chegewara, 2020-07-21:
@Naphier Thanks man, 5V is solution in my case

bhupiister, 2020-08-09:
> for anybody else that has the same problem with the micro SD module you should connect the VCC pin of micro SD card module to the pin Vin of ESP32, not the 3.3 V out. Some micro SD card modules need 5 volts to work.

Thanks man, i had the same module as yours. It works perfectly fine with 5V input. God knows what all i tried before coming to this solution.