espressif/arduino-esp32 #525: Porting of Esp8266 EEPROM library

pbecchi, 2017-07-18:
For legacy compatibility it will be very useful to have the equivalent of Esp8266 EEPROM.h

everslick, 2017-07-19:
I hacked something up for myself to ease my own migration pains a little..

HEADER
```
#ifndef _EEPROM_H_
#define _EEPROM_H_

#include <stddef.h>
#include <stdint.h>
#include <string.h>

#include <Preferences.h>

#define MAX_EEPROM_SIZE 2048

class EEPROM {

public:

  EEPROM(void);

  void begin(size_t size = MAX_EEPROM_SIZE);
  uint8_t read(int address);
  void write(int address, uint8_t val);
  bool commit();
  void end();

  uint8_t * getDataPtr();

  template<typename T>
  T &get(int address, T &t) {
    if (address < 0 || address + sizeof(T) > _size)
      return t;

    memcpy((uint8_t*) &t, _data + address, sizeof(T));
    return t;
  }

  template<typename T>
  const T &put(int address, const T &t) {
    if (address < 0 || address + sizeof(T) > _size)
      return t;

    memcpy(_data + address, (const uint8_t*) &t, sizeof(T));
    _dirty = true;
    return t;
  }

protected:

  uint8_t *_data;
  size_t   _size;
  bool     _dirty;

private:

  Preferences preferences;

};

#endif
```

IMPLEMENTATION
```
#include "Arduino.h"

#include "eeprom.h"

EEPROM::EEPROM(void) {
  _data  = 0;
  _size  = 0;
  _dirty = false;

  preferences.begin("EEPROM", false);
}

void EEPROM::begin(size_t size) {
  if (size <= 0) return;

  if (size > MAX_EEPROM_SIZE) size = MAX_EEPROM_SIZE;

  if (_data) {
    delete[] _data;
  }

  _data = new uint8_t[size];
  _size = size;

  // read data from flash
  if (preferences.getBytes("config", _data, _size) == 0) {
    memset(_data, 0, _size);
    preferences.putBytes("config", _data, _size);
  }
}

void EEPROM::end() {
  if (!_size) return;

  commit();

  if (_data) {
    delete[] _data;
  }

  _data = 0;
  _size = 0;

  preferences.end();
}

uint8_t EEPROM::read(int address) {
  if (address < 0 || (size_t)address >= _size) {
    return (0);
  }

  if (!_data) {
    return (0);
  }

  return (_data[address]);
}

void EEPROM::write(int address, uint8_t value) {
  if (address < 0 || (size_t)address >= _size) {
    return;
  }

  if (!_data) {
    return;
  }

  // Optimise _dirty. Only flagged if data written is different.
  uint8_t* pData = &_data[address];

  if (*pData != value) {
    *pData = value;
    _dirty = true;
  }
}

bool EEPROM::commit() {
  if (!_size)  return (false);
  if (!_dirty) return (true);
  if (!_data)  return (false);

  // write data to flash
  if (preferences.putBytes("config", _data, _size) == 0) {
    return (false);
  }

  _dirty = false;

  return (true);
}

uint8_t *EEPROM::getDataPtr() {
  _dirty = true;

  return (&_data[0]);
}
```

igrr, 2017-07-19:
The underlying NVS library should give an error if you try to store a blob larger than 1984 bytes (at least in the current ESP-IDF).

everslick, 2017-07-19:
Thanks for the hint, so it would be wise to change the define to

#define MAX_EEPROM_SIZE 1984

pbecchi, 2017-07-19:
Thanks for the suggestions, but ,in order to avoid the max size limitation that is not consistent with Arduino and Esp8266 Max EEPROM size, I was thinking to use flash_spi_read and flash_spi_right to a flash sector located on the highest sector of NVS memory. To do this , do I have to modify the partition table?
To something like this:
Name	Type	SubType	Offset	Size	Flags
nvs	data	nvs	0x9000	0x4fff
EEPROM data 99 0xdfff 0x1
otadata	data	ota	0xe000	0x2000	
app0	app	ota_0	0x10000	0x140000	
app1	app	ota_1	0x150000	0x140000	
spiffs	data	spiffs	0x290000	0x170000	
Can this be done?

pbecchi, 2017-07-19:
thanks for the suggestions,
but , In order to avoid this limitation and be fully compatible with Esp8266 EEPROM Size, I was thinking to use flash_spi_write and flas_spi_read to a sector located to the highest NVS memory.
I was thinking to modify the partition table this way : 

# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x4000,
eeprom,data,99,0xd000,0x1000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x140000,
app1,     app,  ota_1,   0x150000,0x140000,
spiffs,   data, spiffs,  0x290000,0x170000,

is this feasible?

igrr, 2017-07-19:
Yes, that should work. Since you plan to use the partition table, I suggest using `esp_partition_{read,write,erase_range}` instead of `spi_flash_*` functions, as they take partition offsets and sizes into account.

me-no-dev, 2017-07-19:
I would suggest that eeprom go right before spiffs though.

pbecchi, 2017-07-19:
Thanks a lot
I will try following your suggestions and I will post the library after debug.

pbecchi, 2017-07-21:
The EEPROM library is ready and is attached
[EEPROM.zip](https://github.com/espressif/arduino-esp32/files/1165147/EEPROM.zip)
It require a partition with label EEPROM that need to be added to the partition table.

I have now to do a pull request ? Sorry but I am new to this!

lonerzzz, 2017-07-21:
@pbecchi Are you new to git for version control or getting your content into a shared project?

pbecchi, 2017-07-22:
Yes .....it is first time......  but I think a can do a pull request with new library folder and modified partition list  .CSV file

pbecchi, 2017-07-23:
I have just created a pull request that contain new library and modified partition table default.csv.
If successfull issue can be closed.

everslick, 2017-07-24:
The PR from pbecchi works for me. #535

lonerzzz, 2017-07-27:
Confirm pbecchi PR works for me as well.

lonerzzz, 2017-08-01:
https://github.com/espressif/arduino-esp32/pull/535 has been merged, closing issue...

closed 2017-08-01 by lonerzzz