espressif/arduino-esp32 #526: Not compiling anymore when using latest IDF libs

Raienryu97, 2017-07-19:
So I used to use esp-idf's `make menuconfig` and arduino's default sdkconfig file and make changes to it as per my needs, take those libs and use them with Arduino IDE for a customised build.
It used to work, I last tried maybe two weeks ago, not sure. 
I tried today with the latest commit from IDF [d998bfc](d998bfc7d62c9ef17a25ea2375e699152f36cb80) and it's now not compiling.
I would love it if I could get some help on if I'm missing anything.
Thanks.

This here is the compilation error. 
```
"/home/pranav/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc" -nostdlib "-L/home/pranav/Arduino/hardware/espressif/esp32/tools/sdk/lib" "-L/home/pranav/Arduino/hardware/espressif/esp32/tools/sdk/ld" -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority  -Wl,--start-group "/tmp/arduino_build_22707/sketch/TestCompile.ino.cpp.o" "/tmp/arduino_build_22707/arduino.ar" -lgcc -lstdc++ -lapp_update -lbootloader_support -lbt -lbtdm_app -lc -lc_nano -lcoap -lcoexist -lcore -lcxx -ldriver -lesp32 -lethernet -lexpat -lfatfs -lfreertos -lhal -ljsmn -ljson -llog -llwip -lm -lmbedtls -lmdns -lmicro-ecc -lnet80211 -lnewlib -lnghttp -lnvs_flash -lopenssl -lphy -lpp -lrtc -lsdmmc -lsmartconfig -lsoc -lspi_flash -ltcpip_adapter -lulp -lvfs -lwear_levelling -lwpa -lwpa2 -lwpa_supplicant -lwps -lxtensa-debug-module -Wl,--end-group -Wl,-EL -o "/tmp/arduino_build_22707/TestCompile.ino.elf"
/home/pranav/Arduino/hardware/espressif/esp32/tools/sdk/lib/libesp32.a(cpu_start.o):(.literal.main_task+0xc): undefined reference to `heap_caps_enable_nonos_stack_heaps'
/home/pranav/Arduino/hardware/espressif/esp32/tools/sdk/lib/libesp32.a(cpu_start.o):(.iram1.literal+0x90): undefined reference to `heap_caps_init'
/home/pranav/Arduino/hardware/espressif/esp32/tools/sdk/lib/libesp32.a(cpu_start.o): In function `main_task':
/home/pranav/esp/esp-idf/components/esp32/./cpu_start.c:348: undefined reference to `heap_caps_enable_nonos_stack_heaps'
/home/pranav/Arduino/hardware/espressif/esp32/tools/sdk/lib/libesp32.a(cpu_start.o): In function `DPORT_READ_PERI_REG':
/home/pranav/esp/esp-idf/components/soc/esp32/include/soc/dport_access.h:93: undefined reference to `heap_caps_init'
/home/pranav/Arduino/hardware/espressif/esp32/tools/sdk/lib/libnewlib.a(syscalls.o):(.iram1.literal+0x0): undefined reference to `heap_caps_malloc'
/home/pranav/Arduino/hardware/espressif/esp32/tools/sdk/lib/libnewlib.a(syscalls.o):(.iram1.literal+0x4): undefined reference to `heap_caps_free'
/home/pranav/Arduino/hardware/espressif/esp32/tools/sdk/lib/libnewlib.a(syscalls.o):(.iram1.literal+0x8): undefined reference to `heap_caps_realloc'
/home/pranav/Arduino/hardware/espressif/esp32/tools/sdk/lib/libnewlib.a(syscalls.o): In function `_malloc_r':
/home/pranav/esp/esp-idf/components/newlib/./syscalls.c:28: undefined reference to `heap_caps_malloc'
/home/pranav/Arduino/hardware/espressif/esp32/tools/sdk/lib/libnewlib.a(syscalls.o): In function `_free_r':
/home/pranav/esp/esp-idf/components/newlib/./syscalls.c:33: undefined reference to `heap_caps_free'
/home/pranav/Arduino/hardware/espressif/esp32/tools/sdk/lib/libnewlib.a(syscalls.o): In function `_realloc_r':
/home/pranav/esp/esp-idf/components/newlib/./syscalls.c:38: undefined reference to `heap_caps_realloc'
/home/pranav/Arduino/hardware/espressif/esp32/tools/sdk/lib/libnewlib.a(syscalls.o): In function `_calloc_r':
/home/pranav/esp/esp-idf/components/newlib/./syscalls.c:43: undefined reference to `heap_caps_malloc'
/home/pranav/Arduino/hardware/espressif/esp32/tools/sdk/lib/libfreertos.a(port.o): In function `pvPortMalloc':
/home/pranav/esp/esp-idf/components/freertos/./port.c:453: undefined reference to `heap_caps_malloc'
/home/pranav/Arduino/hardware/espressif/esp32/tools/sdk/lib/libfreertos.a(port.o): In function `vPortFree':
/home/pranav/esp/esp-idf/components/freertos/./port.c:458: undefined reference to `heap_caps_free'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32 Dev Module.
```

igrr, 2017-07-19:
There's a new component, `heap`, which you are not linking into your output file. Add `-lheap` to linker flags and copy `libheap.a` from the build directory of IDF project.

arvindr21, 2017-07-19:
@igrr @me-no-dev Looks like IDF libs are updating very frequently and a lot of good stuff is being added. Anyway we can put a script together in this repo like `update-to-latest-idf.sh` which will do what @Raienryu97 is doing, then adding new libraries will be easy? - Just a thought

me-no-dev, 2017-07-19:
no need ;) i have scripts I just need to update the base.

arvindr21, 2017-07-19:
@me-no-dev 👍

Raienryu97, 2017-07-19:
Hey @igrr , thanks for that, I was already copying libheap, but adding -lheap flag solved the compilation error. 
Good to know @me-no-dev already worked on the scripts 

But a simple print of "Hello world" is failing with the new libs, it keeps rebooting and I can't seem to understand why, everything is set to verbose debug. Would like some help on this if possible.
Thanks.

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034
[0;32mI (455) cpu_start: Pro cpu up.[0m
[0;32mI (455) cpu_start: Starting app cpu, entry point is 0x40081154[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (464) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (485) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
D (503) heap_init: New heap initialised at 0x3ffaff10[0m
[0;32mI (519) heap_init: At 3FFC6240 len 00019DC0 (103 KiB): DRAM[0m
D (538) heap_init: New heap initialised at 0x3ffc6240[0m
[0;32mI (554) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (574) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (593) heap_init: At 40092834 len 0000D7CC (53 KiB): IRAM[0m
D (612) heap_init: New heap initialised at 0x40092834[0m
[0;32mI (628) cpu_start: Pro cpu start user code[0m
D (665) clk: RTC_SLOW_CLK calibration value: 3347482[0m
V (694) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args[0m
V (694) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE[0m
D (742) intr_alloc: Connected src 46 to int 2 (cpu 0)[0m
V (789) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args[0m
V (848) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE[0m
D (921) intr_alloc: Connected src 56 to int 3 (cpu 0)[0m
V (969) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args[0m
V (1027) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE[0m
D (1101) intr_alloc: Connected src 16 to int 12 (cpu 0)[0m
V (1150) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args[0m
V (1210) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x40E[0m
D (1286) intr_alloc: Connected src 24 to int 13 (cpu 0)[0m
[0;32mI (1336) cpu_start: Starting scheduler on PRO CPU.[0m
V (930) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args[0m
V (936) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting flags 0x40E[0m
D (944) intr_alloc: Connected src 25 to int 2 (cpu 1)[0m
[0;32mI (949) cpu_start: Starting scheduler on APP CPU.[0m
D (954) heap_init: New heap initialised at 0x3ffe0440[0m
D (960) heap_init: New heap initialised at 0x3ffe4350[0m
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034
[0;32mI (674) cpu_start: Pro cpu up.[0m
[0;32mI (674) cpu_start: Starting app cpu, entry point is 0x40081154[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (683) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (704) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
D (723) heap_init: New heap initialised at 0x3ffaff10[0m
[0;32mI (739) heap_init: At 3FFC6240 len 00019DC0 (103 KiB): DRAM[0m
D (758) heap_init: New heap initialised at 0x3ffc6240[0m
[0;32mI (774) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (793) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (813) heap_init: At 40092834 len 0000D7CC (53 KiB): IRAM[0m
D (832) heap_init: New heap initialised at 0x40092834[0m
[0;32mI (848) cpu_start: Pro cpu start user code[0m
D (885) clk: RTC_SLOW_CLK calibration value: 3353190[0m
V (913) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args[0m
V (914) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE[0m
D (961) intr_alloc: Connected src 46 to int 2 (cpu 0)[0m
V (1009) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args[0m
V (1068) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE[0m
D (1142) intr_alloc: Connected src 56 to int 3 (cpu 0)[0m
V (1191) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args[0m
V (1250) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE[0m
D (1324) intr_alloc: Connected src 16 to int 12 (cpu 0)[0m
V (1373) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args[0m
V (1433) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x40E[0m
D (1508) intr_alloc: Connected src 24 to int 13 (cpu 0)[0m
[0;32mI (1558) cpu_start: Starting scheduler on PRO CPU.[0m
V (933) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args[0m
V (939) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting flags 0x40E[0m
D (947) intr_alloc: Connected src 25 to int 2 (cpu 1)[0m
[0;32mI (952) cpu_start: Starting scheduler on APP CPU.[0m
D (957) heap_init: New heap initialised at 0x3ffe0440[0m
D (963) heap_init: New heap initialised at 0x3ffe4350[0m
ets Jun  8 2016 00:22:57
```

Raienryu97, 2017-08-02:
IDF Libs updated to 2.1 
Much appreciated, thank you.

closed 2017-08-02 by Raienryu97