espressif/arduino-esp32 #528: UART2 receive - 2 stop bit, not responding

padmaraj1008, 2017-07-19:
Hi,
We are working with ESP wroom 32 and communicating to TI's ultrasonic AFE PGA460 through UART2 interface. 
We are unable to get data  back from PGA chip on UART2 of ESP32 with 2 stop bits, we are able to communicate with proper data on UART2 interface with pic, MSP and all other controllers but not with ESP 32. 
We are struggling with this issue since many days, still didn't got any solution about this. 
We are able to send data from esp32 but no response from PGA(serial available not getting any bytes)

Kindly requesting you to reply asap with the solution to fix the problem.
Thank you in advance.

copercini, 2017-07-19:
Are you running the last git version?
It looks like this has been fixed few days ago: https://github.com/espressif/arduino-esp32/pull/461/

padmaraj1008, 2017-07-20:
Hello,
We did changes to latest version but still PGA is not responding for two stop bits.

copercini, 2017-07-20:
Just for check, how are you starting the Serial2 ?

copercini, 2017-07-26:
Try something like this:

```cpp
HardwareSerial Serial2(2);

Serial2.begin(9600, SERIAL_8N2, 16, 17); //baud rate, 8 data bits - no parity - 2 stop bits, RX pin, TX pin
```

padmaraj1008, 2017-07-29:
Hello,

Thanks for your valuable suggestions,
Tx and Rx pins from ESP wroom32 are internally pulled down because of this reason UART2 not responding,I included  pull up resistor of 10k on both pins to get it working now it is responding.

Thank you.

closed 2017-07-29 by padmaraj1008

sundar19, 2021-02-10:
Hi , can you please tell me how did you achieve it? I am trying to interface pga460q1psm evm with arduino mega. I am using the modified energia libraries provided by TI but it didnt help us out