espressif/arduino-esp32 #530: RF24_config.h issue with ESP-WROOM-32

wepaking, 2017-07-19:
I saw a recent issue here where someone had the same issue with the compiler giving a avr/pgmspace.h error this file does not exist.  I fixed that by using that the method of 
#if (defined(__AVR__))
#include <avr\pgmspace.h>
#else
#include <pgmspace.h>
#endif
in the RF24_config.h file
when I did that it fixed that issue but generated a series of other errors saying 
C:\Users\User\Documents\Arduino\libraries\RF24/nRF24L01.h:114:0: warning: "NOP" redefined.  This generates of series of "NOP" errors and does not compile can anyone with my knowledge than me sway me in the right direction to fix this issue.  Thank you guys.  I greatly appreciate it.

wepaking, 2017-07-19:
This is the full list of compiler errors.

In file included from C:\Users\User\Desktop\esptranstest\esptranstest.ino:10:0:

C:\Users\User\Documents\Arduino\libraries\RF24/nRF24L01.h:114:0: warning: "NOP" redefined

 #define NOP           0xFF

 ^

In file included from C:\Users\User\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:35:0,

                 from sketch\esptranstest.ino.cpp:1:

C:\Users\User\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal.h:47:0: note: this is the location of the previous definition

 #define NOP() asm volatile ("nop")

 ^

In file included from C:\Users\User\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:35:0,

                 from C:\Users\User\Documents\Arduino\libraries\RF24\RF24_config.h:63,

                 from C:\Users\User\Documents\Arduino\libraries\RF24\RF24.cpp:10:

C:\Users\User\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal.h:47:0: warning: "NOP" redefined

 #define NOP() asm volatile ("nop")

 ^

In file included from C:\Users\User\Documents\Arduino\libraries\RF24\RF24.cpp:9:0:

C:\Users\User\Documents\Arduino\libraries\RF24\nRF24L01.h:114:0: note: this is the location of the previous definition

 #define NOP           0xFF

 ^

C:\Users\User\Documents\Arduino\libraries\RF24\RF24.cpp: In member function 'uint8_t RF24::get_status()':

C:\Users\User\Documents\Arduino\libraries\RF24\RF24.cpp:348:19: error: 'NOP' was not declared in this scope

   return spiTrans(NOP);

                   ^

exit status 1
Error compiling for board ESP32 Dev Module.

copercini, 2017-07-19:
[nRF24L01 library](https://github.com/maniacbug/RF24/blob/master/nRF24L01.h#L113) have the same name macro than [ESP32 hal](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal.h#L47)

So the compiler interprets it as a redefinition.

copercini, 2017-07-19:
You can rename the "NOP" on the whole RF24 library 
or
just delete it and replace: `status = SPI.transfer( NOP );` by `status = SPI.transfer( 0xFF );`

wepaking, 2017-07-19:
I am sorry I am just not understanding if I delete the NOP where or what header file do I place the status code you posted?  Thanks sorry for all the questions.

copercini, 2017-07-19:
Fasten your seat belt:

In C language a macro is a "name" and whenever this name is used, it is replaced by the contents of the macro, like:
`#define [identifier name] [value]` 
Where [identifier name], will be replaced by [value]

Lets think in a example:
You have a mathematical script which makes a lot of operations with pi, so instead of rewrite pi a lot of times, make a macro of PI name : `#define PI 3.14159265`
so just use it in your program:

```
circle_area = PI * circle_radius * circle_radius;
sphere_volume = 4/3 * PI * sphere_radius * sphere_radius * sphere_radius
```

----------------------------
The problem here is that ESP32 and RF24 library have a same name macro, named `NOP ` so the compiler don't know which to replace where.

In this case is more easy change the RF24 library than ESP32, because it appears fewer times and never is a good idea rename anything of the core hahahaha

So in the RF24 it's declared in `RF24-master\nRF24L01.h` and just used on the line 176 of the file `RF24-master\RF24.cpp` , one way is rename that macro and another way is just replace this line by the value.

wepaking, 2017-07-20:
Gotcha man thanks for all your help I did the change and it works now and supper happy man thanks bunhes you saved me from banging my head alot more than usual although something interesting in my libary there was some descrepency where it didnt have that line of code you mention all it had was a return NOP so i saw what you were talking about when I went to github and look at the library for some reason there was a difference between the two so I downloaded it again and made the changes and it worked.  What a weird issue this was seems like the library got corrupt as well somehow.  Thanks man cheers!

wepaking, 2017-07-20:
I will never look at a macro the same again hahaha.  We are frenemies now.

copercini, 2017-07-20:
;)

copercini, 2017-08-18:
Solved!

closed 2017-08-18 by copercini

HannHank, 2018-01-07:
I also have the same problem, but when I looked at the RF24.cpp I found at the line 176 this:  while ( len-- )
    _SPI.transfer(*buf++);
  endTransaction();
. And there is no redefinition , maybe they change the library ?

GOLLAVILLISANJAYKUMAR, 2022-12-05:
Could you please provide me with the Folder that worked for you(RF24_config.h issue with ESP-WROOM-32)  @wepaking