espressif/arduino-esp32 #531: SD with other SPI modules?

dstoiko, 2017-07-20:
Hi,
I'm trying to make my µSD card module work with other SPI modules on the same bus. My current setup is as follows:
MOSI -> 23
MISO -> 19
SCK -> 18

- SD module on CS pin 5 (default on official library)
- LED matrix driver module (MAX7219) on CS pin 16 (tried also with pins 13, 14 and 15)

I'm using the [LedControl](https://github.com/wayoda/LedControl) library to control my LED matrix via MAX7219.

Both SPI modules work separately, but some issue is preventing them from working together.

I can't tell where the issue comes from, because both libraries I use are doing things correctly in terms of selecting / deselecting the slave on the SPI bus: 
- SD library uses the `sdDeselectCard` function in `sd_diskio.cpp` that sets the SD CS pin to HIGH each time an SPI transfer occurs
- LedControl library uses a `digitalWrite(CS, HIGH)` as well each time it commands the controller via SPI

What else can I check? Might there be an issue with the clock divider?

copercini, 2017-07-20:
It doesn't initializes or initializes and bugs in run time?

dstoiko, 2017-07-20:
Ok so I used a scope to figure out what was wrong and found 2 things:
- the frequency of the clock on SD was set to 4MHz in the official library
- the frequency for the LED driver lib was 2.6MHz

So I set the serial clock frequency to a uniform 2.6MHz.
Didn't solve my issue, so I looked at the data being passed to the LED matrix driver when running with SD code. I found that no data was passed through (by hooking an external trigger on the CS line of the module).
Turns out the issue is that I'm using hardware SPI for the SD module (by default) and the library uses *software* SPI for the LED matrix module... which don't mix well because the data lines can't be used as GPIOs when used for hardware SPI.
I need to replace the `shiftOut` function used in the library with a proper SPI transfer function, which is my next problem: it's pretty clear on the arduino standard SPI reference, but I can't use that here apparently, so what should I use instead? In the SD library they use a `writeBytes` function
Here is the code from the library:
```
void LedControl::spiTransfer(int addr, volatile byte opcode, volatile byte data) {
    //Create an array with the data to shift out
    int offset=addr*2;
    int maxbytes=maxDevices*2;

    for(int i=0;i<maxbytes;i++)
        spidata[i]=(byte)0;
    //put our device data into the array
    spidata[offset+1]=opcode;
    spidata[offset]=data;
    //enable the line 
    digitalWrite(SPI_CS,LOW);
    //Now shift out the data 
    for(int i=maxbytes;i>0;i--)
        // Using software SPI...
        shiftOut(SPI_MOSI,SPI_CLK,MSBFIRST,spidata[i-1]);
        // HOW TO USE HARDWARE SPI HERE?
    //latch the data onto the display
    digitalWrite(SPI_CS,HIGH);
}    
```
I tried that:
```
void LedControl::spiTransfer(int addr, volatile byte opcode,
                             volatile byte data) {
  // Create an array with the data to shift out
  int offset = addr * 2;
  int maxbytes = maxDevices * 2;

  for (int i = 0; i < maxbytes; i++) spidata[i] = (byte)0;
  // put our device data into the array
  spidata[offset + 1] = opcode;
  spidata[offset] = data;
  // enable the line
  digitalWrite(SPI_CS, LOW);
  SPI.writeBytes(spidata, sizeof(spidata));
  // latch the data onto the display
  digitalWrite(SPI_CS, HIGH);
}
```
...but I get a `Guru meditation` segfault 👎 
Any ideas?

closed 2019-08-15 by stale[bot]