espressif/arduino-esp32 #532: Arduino IDE expressif installation location info

rrobinet, 2017-07-20:
Hi,
I would like to mention that the installation should not always be located under:
~/Documents/Arduino/.... typically ~/Documents/Arduino/hardware/espressif
But under the location defined in the Arduino preferences has
Sketchbook location:
For example in my case defined as:
/Users/Robert/Data/Our Documents/Projets Techniques/Domotic/Arduino/Sketch
The practical location should be:
/Users/Robert/Data/Our Documents/Projets Techniques/Domotic/Arduino/hardware/espressif
Robert

MrJustPaul, 2017-07-27:
Thanks Robert.  I'm wondering if my SEVERAL attempts to get my ESP32 setup to work is because of location issues - I'm getting lots of "no such file" or duplicates of things, etc.

markonian, 2017-08-25:
@MrJustPaul, did you get your setup to work?

I too have a non-default Sketchbook location. That explains why the Installation Instructions for Mac don't work as-is. As is pointed out in OP, the [installation script](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/mac.md) is hard coded for the Sketchbook location to be `~/Documents/Arduino`.

For historical reasons, my Sketchbook location is `~/Developer/Projects/Arduino/Arduino-trunk`. So I changed the script to: 

```
mkdir -p ~/Developer/Projects/Arduino/Arduino-trunk/hardware/espressif && \
cd ~/Developer/Projects/Arduino/Arduino-trunk/hardware/espressif && \
git clone https://github.com/espressif/arduino-esp32.git esp32 && \
cd esp32/tools/ && \
python get.py

```
and executed it. The following was printed to the Terminal screen:

```
mkdir -p ~/Developer/Projects/Arduino/Arduino-trunk/hardware/espressif && \
> cd ~/Developer/Projects/Arduino/Arduino-trunk/hardware/espressif && \
> git clone https://github.com/espressif/arduino-esp32.git esp32 && \
> cd esp32/tools/ && \
> python get.py
Cloning into 'esp32'...
remote: Counting objects: 5098, done.
remote: Total 5098 (delta 0), reused 0 (delta 0), pack-reused 5098
Receiving objects: 100% (5098/5098), 80.17 MiB | 1.10 MiB/s, done.
Resolving deltas: 100% (2615/2615), done.
Checking connectivity... done.
System: Darwin, Info: Darwin-16.7.0-x86_64-i386-64bit
Platform: x86_64-apple-darwin
Downloading xtensa-esp32-elf-osx-1.22.0-61-gab8375a-5.2.0.tar.gz
Done
Extracting xtensa-esp32-elf-osx-1.22.0-61-gab8375a-5.2.0.tar.gz
Downloading esptool-a420774-macos.tar.gz
Done
Extracting esptool-a420774-macos.tar.gz
Done

```

As far as I can tell, the installation was successful. However, when I run the Arduino IDE, I don't see anything that indicates that arduino-esp32 is available. For example, under the Tools/Board: menu, I assume that there should be some esp32 targets. Questions follow...

Is there a arduino-esp32 focused forum where I can go to get help? If so, I'll go there.

If not, then...

Was my installation attempt successful? How can I tell?

I'm using macOS 10.12.6. Tested with Arduino IDE versions 1.8.1, 1.8.2, and 1.8.4.

rrobinet, 2017-08-26:
Hi,
All I can say is that was the trick for me:
Most of my alien (no-Arduino products) libraries are under my sketchbook location /hardware.
Please verify if this i actually the case: 
My Sketchbook location is: 
/Users/Robert/Data/Our Documents/Projets Techniques/Domotic/Arduino/Sketch
and the espressif set is under:
/Users/Robert/Data/Our Documents/Projets Techniques/Domotic/Arduino/Sketch/Hardware/espressif/esp32

You have to restart Arduino to actually see it under the Tools menu.

Robert

closed 2017-09-01 by rrobinet