espressif/arduino-esp32 #534: Proposal: exceptions should reboot instead of halting the CPU by default

everslick, 2017-07-22:
CPU exceptions (a.k.a. Guru Meditation Errors :-D ￼) now halt the cpu, instead of triggering a reboot. There seems to be a way to configure that, but that requires recompilation of ESP-IDF (ESP32_PANIC_PRINT_REBOOT/ESP32_PANIC_PRINT_HALT/ESP32_PANIC_SILENT_REBOOT/ESP32_PANIC_GDBSTUB) and copying over the libesp32.a file to the Arduino tree by hand.

IMHO the default behaviour should be to reboot instead of halting the CPU, because in production systems this will be the only sensible thing to do. But I dunno what other people think about it. Maybe it's only me.

everslick, 2017-07-22:
AFAIK, there are already some other fixes pending in ESP-IDF, so maybe this could be slipped into the next sync with upstream as well?

lonerzzz, 2017-07-24:
@everslick Have you raised this with the esp32 project? I agree that choosing how to handle these situations would make sense and allowing autoreboot makes sense, but with that project might be the better place to raise the issue. If it was configured, then this project could add an Arduino wrapper to it.

lonerzzz, 2017-08-03:
@everslick This seems to be in place with the latest code using the updated IDF base version. Please test your use case so we know if work needs to be done.

me-no-dev, 2017-08-03:
you know this is already set and will be in the next release, right? :)

everslick, 2017-08-03:
Thanks @me-no-dev. yes I read your comment on Gitter, that the change is
pending.

lonerzzz, 2017-08-03:
@me-no-dev I am seeing something of this functionality now. In sample code I created to help with another issue, the CPU halts turned into resets after upgrading to the latest code Aug. 1st.

lonerzzz, 2017-08-15:
This is now the default behaviour.

closed 2017-08-15 by lonerzzz

sansillusion, 2018-02-18:
Anybody can help me ? I get rare random crashes in my project. It is at a point where I am happy with it and would just like it to reboot when it has a core panic but it just hangs. I have the latest version of arduino-esp32 and have kept up to date but seem to not have the same behavior has expected ?

my code : https://github.com/sansillusion/LampeMeteo