espressif/arduino-esp32 #536: esp_wifi_stop 802 wifi is not init

Ultimatefire54, 2017-07-26:
Hello, thank you for your work on the esp32 and arduino.
I am having a problem withe esp as when I try to view the monitor output from the hello_world example included and mentioned in the esp idf setup, I get these errors:

`E (11820) wifi: esp_wifi_stop 802 wifi is not init
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:5300
load:0x40078000,len:0
load:0x40078000,len:12652
entry 0x40078f44
[0;32mI (45) boot: ESP-IDF v3.0-dev-168-gd515eeac-dirty 2nd stage bootloader[0m
[0;32mI (45) boot: compile time 11:05:36[0m
[0;32mI (49) boot: Enabling RNG early entropy source...[0m
[0;32mI (65) boot: SPI Speed      : 80MHz[0m
[0;32mI (78) boot: SPI Mode       : DIO[0m
[0;32mI (90) boot: SPI Flash Size : 4MB[0m
[0;32mI (103) boot: Partition Table:[0m
[0;32mI (114) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (137) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (160) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (183) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (207) boot: End of partition table[0m
[0;32mI (220) boot: Disabling RNG early entropy source...[0m
[0;32mI (237) boot: Loading app partition at offset 00010000[0m
[0;32mI (255) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x07c98 ( 31896) map[0m
[0;32mI (311) esp_image: segment 1: paddr=0x00017cc0 vaddr=0x3ffb0000 size=0x02dbc ( 11708) load[0m
[0;32mI (324) esp_image: segment 2: paddr=0x0001aa84 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (338) esp_image: segment 3: paddr=0x0001ae8c vaddr=0x40080400 size=0x05184 ( 20868) load[0m
[0;32mI (387) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x26190 (156048) map[0m
[0;32mI (530) esp_image: segment 5: paddr=0x000461b0 vaddr=0x40085584 size=0x0d528 ( 54568) load[0m
[0;32mI (590) esp_image: segment 6: paddr=0x000536e0 vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (624) cpu_start: Pro cpu up.[0m
[0;32mI (625) cpu_start: Starting app cpu, entry point is 0x40080dec[0m
[0;32mI (626) cpu_start: App cpu up.[0m
[0;32mI (634) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (655) heap_init: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM[0m
[0;32mI (674) heap_init: At 3FFB66E0 len 00029920 (166 KiB): DRAM[0m
[0;32mI (693) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (712) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (732) heap_init: At 40092AAC len 0000D554 (53 KiB): IRAM[0m
[0;32mI (751) cpu_start: Pro cpu start user code[0m
[0;32mI (807) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (810) cpu_start: Starting scheduler on APP CPU.[0m
Hello world!
This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external flash
`

The program compiles and flashes as it's supposed to, but in the Arduino and esp-idf serial monitor, this is the error that I receive. I have looked for anyone with a similar issue to no avail. I tried make erase_flash in esp-idf, but it did not fix the issue. Any help would be appreciated. Thank you.

copercini, 2017-07-26:
> Hello world!
This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external flash

It's working fine!

These messages are just for debug and appear by default when ESP boot

Ultimatefire54, 2017-07-26:
So is that how the hello_world example is supposed to work? Should I be worried about the esp_wifi_stop?  That response is such a relief.

copercini, 2017-07-26:
Sorry, I deleted my previous answer unintentionally.

yeah, It's fine =)

About `wifi: esp_wifi_stop 802 wifi is not init` is because the [esp_restart(); function on the example](https://github.com/espressif/esp-idf/blob/master/examples/get-started/hello_world/main/hello_world_main.c#L39)
triggers [esp_wifi_stop();](https://github.com/espressif/esp-idf/blob/master/components/esp32/system_api.c#L245) but you aren't used wifi, so it can't stop one thing that has not started...

The another messages are just for debug

copercini, 2017-07-27:
@Ultimatefire54 may be related with this https://github.com/espressif/arduino-esp32/issues/207

Ultimatefire54, 2017-07-27:
@copercini I tried this but I get the same result, the flash erase works, I open the serial monitor and it states there is a flash error (I think this is expected since there is no program to boot from after the flash_erase). I upload a WiFi sketch and get the same result (tx and rx busy but nothing in the serial monitor).

Ultimatefire54, 2017-07-27:
I loaded the power_save and wpa2_enterprise examples and now I am getting this error along with some logs

[0;33mW (1137) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration[0m

Brownout detector was triggered

The esp then goes through a reset loop. Maybe this can help to narrow down the issue.

copercini, 2017-07-27:
> Brownout detector was triggered

How are you powering your ESP32?

Ultimatefire54, 2017-07-27:
I am powering it via the ftdi mmodule which is connnected to my computer.

copercini, 2017-07-27:
So probably it's giving less power than the module needs to work fine

Ultimatefire54, 2017-07-27:
Ok, I will power it from a Li-Po and 3.3v regulator and let you know how that goes.

Ultimatefire54, 2017-07-28:
The lipo battery in conjunction with the 3.3v regulator did not work, but for some reason a 18650 with a pot diff of 3.68v and no regulator worked. The esp32 seems to require an abnormally large amount of current when initializing rf, as my esp12 is fine with the lipo regulator setup. Anyway, thank you for your help on both of the issues I had :)

closed 2017-07-28 by Ultimatefire54