espressif/arduino-esp32 #538: CPU Halted

architmuchhal12, 2017-07-27:
I am using Sparkfun ESP32 to have an MQTT Connection using FONA808. The serial monitor is giving me this vague error which I have no idea about.
Please help me through this.

**OUTPUT-**

```
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU1)
Register dump:
PC      : 0x400d7584  PS      : 0x00060c34  A0      : 0x80084ad1  A1      : 0x3ffc5430  
A2      : 0x00000008  A3      : 0x00000001  A4      : 0x00060023  A5      : 0x3ffc5b98  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffc2244  A9      : 0x3ffc5410  
A10     : 0x00000000  A11     : 0x00060c20  A12     : 0x800835f8  A13     : 0x3ffc4a30  
A14     : 0x00000000  A15     : 0x3ffc4fa0  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d7584:0x3ffc5430 0x40084ad1:0x3ffc5450

CPU halted.

```

Thanks

lonerzzz, 2017-07-27:
If you have not done so, add the ESP Exception Decoder to the Arduino IDE. Then paste this output into the decoder window and let us know what the error was.

architmuchhal12, 2017-07-27:
I did install the ESP Execution Decoder but on opening it, I get an error as -

`Not Supported on AVR`

What does it mean?

lonerzzz, 2017-07-27:
You likely need to appropriately choose your board type as ESP32 under the Tools menu.

architmuchhal12, 2017-07-27:
Ohh thanks! I just did not notice when the board type got changed!
Anyway, I did paste the output in the ESP Execution Decoder, this is what it gives-

**Decoder Output -**
```

Decoding 3 results
0x400d7584: esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./freertos_hooks.c line 52
0x400d7584: esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./freertos_hooks.c line 52
0x40084ad1: prvIdleTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4420

```

lonerzzz, 2017-07-29:
@architmuchhal12 Sorry, for the delayed response. Hopefully someone can assist on this. The only suggestion I have is to see if putting a delay in your loop affects whether the idle hook is called.

haroonahmed97, 2017-07-30:
I have same issue of CPU Halted. I am using esp32 as MQTT client.

lonerzzz, 2017-07-30:
@haroonahmed97 The CPU will halt for any number of reasons. Are you getting the exact same stack trace? If no, I suggest capturing your own stack track and raising a different issue.

copercini, 2017-07-30:
@architmuchhal12  Are you using software or hardware serial?
Please post some code

lonerzzz, 2017-08-01:
I played with some code and managed to recreate this exact stack trace. In my case, it required multiple threads to occur and happened sometime after these threads were routinely running delay( 100 ); while doing nothing else.

@architmuchhal12 Was this occurring for you with one thread or more than one and were you calling delay?

Is it possible that the esp_vApplicationIdleHook being not assigned is causing the issue? I noticed this:  EXCVADDR: 0x00000000 in the core dump.

architmuchhal12, 2017-08-01:
@lonerzzz The CPU Halting is occurring when I use one thread in addition to the main program. In another instance, I tried to modify the program by using the thread as a subroutine and it did work. But, I am not sure as to why it is an issue when threads are introduced.

copercini, 2017-08-01:
I have some guess:

When vTaskDelay is called, the task is putted in blocked mode, when all the tasks are in blocked mode, nothing is running on CPU so esp_vApplicationIdleHook is called to save power.

Somehow it's triggering WDT, maybe because Arduino core design...
Maybe @igrr or @me-no-dev can explain more

me-no-dev, 2017-08-01:
are you using Serial or SoftSerial?

lonerzzz, 2017-08-01:
@me-no-dev In my reproduction of the issue, I was using Serial. There were occasional printouts every ten seconds or so from the different threads, each calling Serial. I checked in Serial and it appears to have proper multi-threading support.

@architmuchhal12 Can you comment on this same question in your situation?

lonerzzz, 2017-08-01:
@architmuchhal12 Upon updating to the latest wide-scale code change submitted by @me-no-dev, I am no longer encountering a CPU halt. I am getting a restart with the sample code and will have to determine if the cause is likely the same or different. Can you confirm whether an update eliminates your problem or converts it into a problem of restarts?

architmuchhal12, 2017-08-02:
I am using HardwareSerial for my task in the following way - 

Main program -
```
// Default pins for Feather 32u4 FONA
#define FONA_RST 25
HardwareSerial fonaSS(1);
Adafruit_FONA fona = Adafruit_FONA(FONA_RST);
```

Thread program:

`extern HardwareSerial fonaSS(1);`

@lonerzzz I will update to the latest code-change and try again to check if it eliminates the problem.

Thanks for the help!

architmuchhal12, 2017-08-02:
I have updated to the latest code-change and I don't have any issues now.
The CPU Halt issue has been resolved!

Thanks!

closed 2017-08-02 by architmuchhal12