espressif/arduino-esp32 #539: Fails to wake from DEEPSLEEP_RESET all the time

scramble45, 2017-07-28:
I have the start of reading a temp / humity sensor and this code works, but over time it get stuck from the DEEPSLEEP_RESET and only makes it so far before bailing out any ideas? The access point is up and so is the TCP server its talking too.

**This is where it gets stuck:**
````
rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034
ESP32 Chip ID = D80A83C40A24
````

**Code:**
````
/*
 * ESP32 I2C - wireless gateway sensor
 * Humidity / Temperature
 */

#include "SparkFun_Si7021_Breakout_Library.h"
#include <Wire.h>
#include <WiFi.h>
#include <WiFiMulti.h>
#include <esp_deep_sleep.h>
#include <ArduinoJson.h>

#define SDA 21 // GPIO21 on ESP-32 dev module
#define SCL 22 // GPIO22 on ESP-32 dev module

uint64_t chipid;

// Time to sleep (in seconds):
const int sleepTimeSecs = 10;

// WiFi settings
const char* ssid = "FVUGW_270113184508694970";
const char* password = "digital123";

float humidity = 0;
float tempf = 0;

//Create Instance SI7021 temp and humidity sensor
Weather sensor;

WiFiMulti WiFiMulti;

void setup()
{
    Serial.begin(115200);
    chipid = ESP.getEfuseMac(); // The chip ID is essentially its MAC address(length: 6 bytes).
    Serial.printf("ESP32 Chip ID = %04X%08X\n",(uint16_t)(chipid>>32),(uint32_t)chipid);//print High 2 bytes

    
    
    //Initialize the I2C sensors and ping them

    // We start by connecting to a WiFi network
    WiFiMulti.addAP(ssid, password);

    Serial.println();

    delay(3000);
    if(WiFiMulti.run() == WL_CONNECTED) {
            Serial.println("");
            Serial.println("WiFi connected");
            Serial.println("IP address: ");
            Serial.println(WiFi.localIP());
    }

    sensor.begin();
    Serial.printf("Got here\n");
}

void loop()
{
    const uint16_t port = 1337;
    const char * host = "10.0.0.1"; // ip or dns

   if(WiFiMulti.run() != WL_CONNECTED) {
        Serial.println("WiFi not connected!");
        delay(1000);
    } else {
      Serial.print("connecting to: ");
      Serial.println(host);
  
      // Use WiFiClient class to create TCP connections
      WiFiClient client;
  
      if (!client.connect(host, port)) {
          Serial.println("connection failed");
          Serial.println("waiting 5 seconds...");
          delay(5000);
          return;
      }

      // Battery voltage
      /*
       * Use a 27 kOhm + 100 kOhm voltage divider to bring the 4.2 V maximum of a single cell LiPo battery down to 3.3 V,
       * which is the reference voltage (AFAIK) of the ESP32 board I am using. 
       * I have the voltage divider connected to pin 36, and I read the pin as follows:
       */
      float vbat = (127.0f/100.0f) * 3.30f * float(analogRead(36)) / 4096.0f;  // LiPo battery
      
      // This will send the request to the server
      int humidity = sensor.getRH();
      int tempf = sensor.getTempF();
      String output;
      StaticJsonBuffer<200> jsonBuffer;
      JsonObject& root = jsonBuffer.createObject();
      root["sensorid"] = "001";
      root["temperature"] = tempf;
      root["humidity"] = humidity;
      root["battery"] = vbat;
      root.printTo(output);

      client.flush();
      client.print(output);

      

      
      
      Serial.println(output);
      //client.print("Send this data to server");
  
      //read back one line from server
      //String line = client.readStringUntil('\r');
      //client.println(line);
      
  
      Serial.println("closing connection");
      client.stop();
      WiFi.disconnect(true);
  
      // Allow transmission to complete
      Serial.println("wait 10 sec...");
      delay(10000);
      
      Serial.println("ESP32 in sleep mode");
      ESP.deepSleep(sleepTimeSecs * 1000000);

    }

    if(WiFiMulti.run() == WL_DISCONNECTED) {
      Serial.println("WiFi was disconnected...");
      ESP.restart(); // restart
    }

    if(WiFiMulti.run() == WL_CONNECTION_LOST) {
      Serial.println("WiFi connection was lost...");
    }

}
````

pipi61, 2017-07-28:
Hi!
in loop() this where to return?

 if (!client.connect(host, port)) {
          Serial.println("connection failed");
          Serial.println("waiting 5 seconds...");
          delay(5000);
          return;
      }

scramble45, 2017-07-28:
I will try taking out the return see if that does anything. Seems like it runs for a while and then gets stuck at some point in the night and wont re-establish a connection via WiFi.

scramble45, 2017-07-28:
same problem.

scramble45, 2017-07-31:
Fixed my problem:

````
    WiFi.begin(ssid, password);

//    if(WiFiMulti.run() == WL_CONNECTED) {
//            Serial.println("");
//            Serial.println("WiFi connected");
//            Serial.println("IP address: ");
//            Serial.println(WiFi.localIP());
//    }
 while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }
````

closed 2017-07-31 by scramble45

josephsr, 2019-01-30:
put the wifi to sleep using esp_wifi_stop before deep sleep and on wake use esp_wifi_start