espressif/arduino-esp32 #540: WiFi.Config() setting wrong IP Address

BrycePeterson, 2017-07-28:
The following issue is present for me when using WiFi.config.
Code:
```

IPAddress chipIP = (192,168,10,22); 
WiFi.config(chipIP,WiFi.gatewayIP(),WiFi.subnetMask());

```
Result of code is:

> IP Address: 22.0.0.0

Seems like every number I put in the last parameter of IPAddress is assigned as the first number of the new IP address, all other parameters are discarded.

lonerzzz, 2017-07-29:
How about when you use this syntax 'IPAddress chip( 192, 168, 10, 22 )' ?

everslick, 2017-07-29:
I think this is related to https://github.com/espressif/arduino-esp32/issues/500 which was closed by the original reporter. Anyway, I cannot set static IP addresses as well. So probably the issue should be kept open for other people to find it.

BrycePeterson, 2017-07-31:
@lonerzzz
> How about when you use this syntax 'IPAddress chip( 192, 168, 10, 22 )' ?

This seemed to work, thanks for the help.

everslick, 2017-08-01:
After IDF has been updated last night, I gave it a shot again. But I see rather confusing behaviour when trying to set a static IP address.

First of all, the IP address can be set without any issue, but name resolution fails afterwards. Even though, when i read back the config the DNS server(s) have been configured correctly.

Second the SoftAP sometimes ignores the configured IP address and defaults to 192.168.4.1 without me being able to recognize a pattern, when it happens. But I think when it happens, the stack canary exception will trigger the next time the WiFi Event callback is called.

I would appreciate if someone could comment, if static IP configuration DOES indeed work for them INCLUDING name resolution and SoftAP (or at least name resolution). THX

me-no-dev, 2017-08-01:
```cpp
// STA Static IP
    if(!WiFi.config(IPAddress(169, 254, 1, 3), IPAddress(10, 0, 0, 1), IPAddress(255, 255, 0, 0), IPAddress(10, 0, 0, 1))){
        Serial.println("STA Failed to configure");
    }
    if(WiFi.begin(STA_SSID, STA_PASS) == WL_CONNECT_FAILED){
        Serial.println("STA Failed to start");
    }

// AP Static IP
    if(!WiFi.softAPConfig(IPAddress(192, 168, 5, 1), IPAddress(192, 168, 5, 1), IPAddress(255, 255, 255, 0))){
        Serial.println("AP Config Failed");
    }
    if(!WiFi.softAP(AP_SSID)){
        Serial.println("AP Start Failed");
    }
```

@everslick this is what worked for me. Did not notice issues with resolution

everslick, 2017-08-25:
Static IP configuration started working after IoT was updated a view days ago, so I suggest closing this issue.

The second problem regarding WiFi.softAPConfig() not setting the configuration turned out to be a race condition between calling WiFi.softAP() and then setting the config with WiFi.softAPConfig(). The solution is to:

1) Start SoftAP with WiFi.softAP()
2) Wait for SYSTEM_EVENT_AP_START
3) Set the config with WiFi.softAPConfig()

Not waiting for the SYSTEM_EVENT_AP_START event _might_ overwrite the config one has just set for the interface.

lonerzzz, 2017-08-25:
Closing since the original and the related issue seem resolved.

closed 2017-08-25 by lonerzzz