espressif/arduino-esp32 #541: analogRead() triggers unrelated interrupt

MaxSNi, 2017-07-28:
I am experiencing a strange bug:

Whenever i use analogRead(), an unrelated interrupt will also trigger, and by changing around a little i could even get 2 interrupts to trigger with every analogRead().

I have not testet every analog pin, but so far this bug happens with A0, A3, A4, A5, A6, A7, but not A10, A11, A12, A13, A14.

I can reproduce the issue with the code below - every 2 seconds it will trigger AnalogRead() and this will trigger one of the button Interrupts. I have have multiple components hooked up as well, but this did not seem to change anything when i tried removing some of them, or having the button GPIOs unconnected.

```C++
const int BUTTON_1 = 25;
const int BUTTON_2 = 26;

void setup() 
{
  pinMode(BUTTON_1, INPUT_PULLUP);  
  pinMode(BUTTON_2, INPUT_PULLUP);
  
  attachInterrupt(BUTTON_1, int1, FALLING);
  attachInterrupt(BUTTON_2, int2, FALLING);
  
  Serial.begin(115200);
}

void loop() 
{
  delay(2000);  
  float derp = analogRead(A3);  
  Serial.println(derp);
}

void int1() 
{
  Serial.println("1 Interrupt triggered");
}

void int2() 
{
  Serial.println("2 Interrupt triggered");
}
```

Raienryu97, 2017-07-31:
Hey
Tried the code you posted along with just the board, and yes, the interrupts are being triggered.
If you're looking for a workaround
Change the Pull type to PULLDOWN  (and of course you should change the button trigger from GND to Vcc since it's pulled down)

I tried with Pulldown and the problem is solved.

```
  pinMode(BUTTON_1, INPUT_PULLDOWN);
  pinMode(BUTTON_2, INPUT_PULLDOWN);
```

Also, maybe you already know this , but just in case, it's never good to `Serial.println()` inside of an ISR. The best idea would be to update a `volatile bool` inside of ISR and based on the bool, print in `void loop()`

KevinVanthuyne, 2019-10-22:
Had this issue as well, the workaround works 👍 Very weird that this is still happening though

VojtechBartoska, 2023-04-28:
@PilnyTomas Please check if this is still relevant now. There have been several updates to AnalogRead so maybe this is already covered.

PilnyTomas, 2023-05-10:
I have tried this with ESP32, C3, S2 and S3 and could not reproduce...
I'm closing this issue. If anyone experiences this error again, fell free to reopen this issue, or open a new one. Please don't forget to add as much info as possible.

closed 2023-05-10 by PilnyTomas