espressif/arduino-esp32 #542: RX2 - TX2 Problem [ ESP32s DevKit V1 ]

DaveOff, 2017-07-29:
hi, i connect Arduino Mega To ESP32.
Everything is ok but When i restart Arduino Serial2.available() Goes True and Serial get few bytes.
my ESP32 picture : http://www.pazhelec.com/image/cache/data/Elecrow%20NodeMCU-32S%20Lua%20WiFi%20ESP32S%20Development%20Board%20WiFi%20Bluetooth%20Ultra%20Low%20Power%20Consumption%20Dual%20Cores%20ESP32%20DIY%20Kit-500x500.jpg

this is my code 
ESPS32 

`void setup(){
  Serial.begin(9600);
  Serial2.begin(9600);     //Start serial communication at 9600bps
  pinMode(23, OUTPUT);

} //End Setup

// LOOP *************************************************************************
void loop()
{

if(Serial2.available() > 0) {

    incomingByte = Serial2.read();
    
    if(incomingByte == 49) { digitalWrite(23, 1); }
    else if(incomingByte == 50) { digitalWrite(23, 0); }
    else {

        while(1>0) {
         Serial.print( incomingByte );
         delay(5000);
          digitalWrite(23, 1);
          delay(80);
          digitalWrite(23, 0);
          delay(80); 
        }

    }
}`

Arduino :

void setup() {
  // initialize serial communication at 9600 bits per second:
  Serial.begin(9600);
  Serial3.begin(9600);
  // Serial3.write(49); 
   pinMode(LED_BUILTIN, OUTPUT);
}

// the loop routine runs over and over again forever:
void loop() {

if(Serial.available() > 0) {
char t = Serial.read();
   Serial3.write(49);
   Serial.print( "send" );

}

          digitalWrite(LED_BUILTIN, 1);
          delay(80);
          digitalWrite(LED_BUILTIN, 0);
          delay(80); 

}

padmaraj1008, 2017-07-29:
your issue seems similar to the the issue That i was facing.Solution is in the following issue thread

UART2 receive - 2 stop bit, not responding #528

DaveOff, 2017-07-29:
hi, tnks for answer,

are you connected tx2,rx2 to resistor of 10k ?

padmaraj1008, 2017-07-29:
Hello,
yes 10k pull up for both.

hallard, 2017-08-05:
Guys, 
Got the same problem and google drive me here, I digged into the core code and I saw we can enable internals pullup but no option do do it in library so what I've done with success is to configure pull up on RX pin after Serial1.begin()
I'm using RX only but should works on TX also

```cpp
#define RX_PIN 33

HardwareSerial Serial1(1);  // UART1/Serial1 

void setup() {
  Serial1.begin(1200, SERIAL_7E1, RX_PIN);
  pinMode(RX_PIN, INPUT_PULLUP);
}
```

lonerzzz, 2017-08-13:
@DaveOff Did the information provided answer your question?

closed 2018-04-19 by everslick