espressif/arduino-esp32 #543: Commit process unclear

lonerzzz, 2017-07-29:
Hello,

For everyone contributing content, what is the process to get something onto the main branch? And who is ensuring the pull requests get there? I am happy to document this for other contributors, but right now there are several pull requests that have been sitting and it is unclear what they are waiting on. If people are needed for this, I will volunteer and can solicit others for the role.

Input would appreciated.

me-no-dev, 2017-08-01:
So are you asking why some commits were delayed or how to submit a PR?

lonerzzz, 2017-08-01:
@me-no-dev I am asking more about commits being delayed but not that specific question :) I would just like to know the process for something getting onto the main branch, not the technical instructions. From igrr, I learned that you are the person responsible for accepting PRs onto the main branch and are the key reviewer so the commits are dependent on your time availability as well as the technical content.

Is there something that other volunteers like myself can do to make it easier for you. ie code cleanup, code comments, adherence/relation to a roadmap (https://github.com/espressif/arduino-esp32/projects/1) etc.

me-no-dev, 2017-08-01:
Well, I am usually here all of the time, but the last few weeks have been busy with other Espressif business and places with poor internet. So I can not always be prompt and merge or answer immediately.

lonerzzz, 2017-08-01:
@me-no-dev No need to defend yourself on this. That you are an Espressif employee/representative is something that was not certain. I am just looking to see how we can help you, if possible.

me-no-dev, 2017-08-01:
One possible thing that you could do is separate user issues from actual bugs that need to be fixed. That can help me focus on the bugs while the community tries to handle the user issues.
Once most bugs are fixed, we can move onto implementing those features that are so much desired.
Probably in this order:
1. Ethernet
2. SPIFFS (in IDF and Arduino)
3. Everything else
4. Async

copercini, 2017-08-01:
I also can help with issues management, documentation, a bit of code and something else =) 

About SPIFFS, how do you plan implement it? I really want to know how improve my code quality like I see in your libs.

lonerzzz, 2017-08-01:
@me-no-dev I will work to help with users issues and encourage others to do so. Thanks for the feedback and your hard work on the code. If you find other areas where assistance is a help, do let us know.

me-no-dev, 2017-08-01:
@copercini SPIFFS has to be implemented in IDF like fatfs. Then provide a simple mount/unmount/format... so on interface, like it's done for SDMMC in Arduino. That is it :)
VFS is already implemented in Arduino ;)

mrengineer7777, 2023-04-13:
Cleaning up old issue

closed 2023-04-13 by mrengineer7777