espressif/arduino-esp32 #544: DOIT ESP32 DEVKIT V1 Board

dbachko, 2017-07-29:
Can you add this one? Pins doesn't match with this one: `variants/doitESP32devkitV1/pins_arduino.h`
[image]
[image]

copercini, 2017-07-29:
You can use it as "ESP32 Dev Module", D13 is GPIO13, D14 is GPIO14 and so on...

Anyway, here is a DOIT author github page: https://github.com/Nicholas3388/LuaNode, have the old board schematic and you can ask for the new one

liqngliz, 2017-08-10:
Here you go :) It also took me a while to find the schematics. Just not the following, the GPIO pins follow the numbers 1-27 (lower right pin out schematics), but the analog pins labeled as A6-A9 are 34, 35, 32, 33, 25, 28. The board labels on board are pretty accurate except you need to know that those analog pins are not labeled unless you look at the schematic.

Some other quirks I noticed:
Also when you power the board with a 5v USB you get 5v power out of the VIN pin, I used this to power a 5v servo with no problems, even during extended stall draw was ok. Also don't accidentally cross ground on the VIN side with ground on the 3.3v side, you will blow your board when powered via USB.

[SchematicsforESP32.pdf](https://github.com/espressif/arduino-esp32/files/1215494/SchematicsforESP32.pdf)

lonerzzz, 2017-08-13:
@dbachko Did the information provided answer your question?

PITDEVBR, 2017-09-10:
You can talk about the UART Serials, I know have three Ports, but I no can't found the tx1 rx1

copercini, 2017-09-10:
@PITDEVBR by default the pins of serial1 are  `rx1Pin = 9; tx1Pin = 10;` and serial2  `rx2Pin = 16; tx2Pin = 17;`

But you can set it on any pin if you want: Serial1.begin(9600, SERIAL_8N1, gpioX_rx, gpioY_tx);

PITDEVBR, 2017-09-10:
@copercini  but in module ESP32 the DOIT dont have the pins 9 and 10. With is command "Serial1.begin(9600, SERIAL_8N1, gpioX_rx, gpioY_tx);" I can change for anywhere pin in module?

me-no-dev, 2017-09-10:
yes. just make sure that TX is below GPIO34. RX can be above

PITDEVBR, 2017-09-27:
Someone have the code using the firmware the changing the serials 1 for test, a create but no have successful...

copercini, 2017-10-06:
@PITDEVBR like this:
```cpp
HardwareSerial Serial1(1);  // UART1/Serial1 

void setup() {
  Serial.begin(115200);
  Serial1.begin(9600,SERIAL_8N1,9,10); //You can change 9 and 10 to another pins
}
```
Have fun =)

closed 2017-10-06 by copercini

sarwadenj, 2017-10-31:
Hi @liqngliz 

The schematic pdf which you shared is verified , i want to use it in one of design. 
Thanks.

chinswain, 2017-12-10:
"Also don't accidentally cross ground on the VIN side with ground on the 3.3v side, you will blow your board when powered via USB." Could someone explain this to me? Are you saying not to use the GND pin next to Vin with the 3v side? I've been doing just that on a lolin32 ;(

DavidAntliff, 2017-12-21:
@chinswain I'm looking at the schematic and I'm also not sure what @liqngliz is referring to. The schematic doesn't really make a distinction between the grounds so it's not clear to me where the separation is meant to be, or why shorting the grounds would cause an issue. The only thing I can see is that there's just a diode with no current-limiting resistor between the VCCUSB and the VIN pin, which means if you were to short VIN to ground then you'd blow the diode, probably.

I suppose if you tried to impose non-common 5V on both sides of the diode, then the ground on the VIN side would sit one diode bias voltage below the ground on the other, but I'm not sure why that would cause an issue if you connected them, since it would just stop the diode from conducting.

I'd love to know what I'm missing here.

bobmc-rmm, 2017-12-25:
The USB is protected by SS14, a low drop schottky 1 amp diode.  The other input does not have this protection so a battery should not be connected.  I see only 1 ground path.

DavidAntliff, 2017-12-25:
@bobmc-rmm so there should be no issue with connecting the "3.3V output" GND on the left side of the board with the "5V input" GND on the right side of the board?

bobmc-rmm, 2017-12-26:
@DavidAntliff  ..There is only one GND in the schematic. Then consider the 3v regulator which has GND common to it's input and output.  It can only function in a common GND circuit.  There can be parallel inputs to the regulator but every source must have the same GND. Moreover, each souce needs to have a protection diode to prevent reverse flow from competing sources.

DavidAntliff, 2017-12-26:
@bobmc-rmm that makes sense, thanks. So the original statement about blowing the board if *ground* on each side is connected is false? The issue is around managing competing supplies.

bobmc-rmm, 2017-12-27:
@liqngliz .. Arduino for ESP32 has variants/doitESP32devkitV1 has 'pins_arduino.h' with symbold A0..A19 not matching the one and only doit board I have seen. Is there a correct header file somewhere.  I have been using Huzzah32 which costs more than most of these ESP32 but it is better quality and works according to documentation. I have ordered a different ESP32 with more pins.

bobmc-rmm, 2017-12-27:
There is another doit ESP32 with more pins .. 2 X 18 and it is also V1. There is no schematic for it.
http://www.smartarduino.com/view.php?id=95103
I am looking for a header file or pinmap with the GPIO assignment for the symbols printed on each board.

onyx007, 2018-02-27:
@bobmc-rmm I confirm, this the same as https://www.banggood.com/ESP32-Development-Board-WiFiBluetooth-Ultra-Low-Power-Consumption-Dual-Cores-ESP-32-ESP-32S-Board-p-1109512.html?cur_warehouse=CN .. very hard to find the appropriate schematic.

davenippon, 2018-03-03:
@onyx007 @bobmc-rmm Curious if either of you have ever found a pinout or other schematic for the DOIT 36-pin board. I ordered a board for my first Arduino project and that's what showed up. Most everything I find online has a GND pin beside the 3V3 pin on the same side of the board, but this one only has one GND and it is on the opposite side of the board from 3V3. Can I just connect that GND to the ground rail on my breadboard then use a jumper to connect that rail to the ground rail that is on the same side with the 3V3 pin? It's my very first attempt so I have a lot to learn :)

bobmc-rmm, 2018-03-03:
There is only one ground circuit on these ESP boards so you use whichever
pins are marked as GND

davenippon, 2018-03-04:
@bobmc-rmm Thanks! Project complete and working like a champ! (Simple e-paper thermometer I saw on another site I wanted to build to get my feet wet.)

bobmc-rmm, 2018-03-04:
that is good news

toh1000, 2018-03-13:
@davenippon Congrats! I am stuck with connecting the e-paper display (2.9 inch waveshare). Running the new version of the Doit V1 board (with 18 pins on each side). 
Any idea how to wire the display?
Display wires:
BUSY
RST
DC
CS
CLK
DIN
GND
3.3V
Thanks in advance.

davenippon, 2018-03-14:
Hello @toh1000 , Try these and see how it goes: 
(e-paper pin -> E32 board pin):
BUSY -> D4
RST -> D21
DC -> D22
CS -> D5
CLK -> D16
DIN -> D23
GND -> GND
3.3V -> 3V3 

[image]

johansundstrom, 2018-03-19:
Any news on the 2x18pin version of the devboard from DOIT? Schematic or pin documentation?

plugowski, 2018-04-05:
Hi,

I'm looking for a eagle library for that development board, does anyone see that?

ColinMarsh, 2018-04-15:
Good question johansundstrom. I originally purchased some v1's, then some v3's then some more v1's. Every single purchase is different. How can DOIT get away with same number and ver, but 6 more pins. There is only 1 ground pin now as we well. I read somewhere when using pull up/down to 5v side can let out to much smoke. Have a guess where the ground pin is? Yep, right beside the 5v input. I guess that there is going to have to be a sacrificial ESP... I actually lucked onto a site that looked like DOIT. I requested info, but guess that it ended up in a junk filter trap. Sorry about the poor quality pic, but I guess anyone looking will get the picture and be forewarned.
[image]

onyx007, 2018-04-15:
A weird board alright, still looking for a valid breakout diagram .. here is a photo from the front for the fun of it. https://www.flickr.com/photos/138684657@N03/39913883734 . Off topic: I was counting the pins with my pen while the board was live off my Mac , sliding the pen from pin to pin and then from GRND to VIN I blew one board. The good this is, it is not expensive.

ColinMarsh, 2018-04-15:
For what it's worth, here is the web site that I have previously tried to get info. http://www.geekcreit.com/customer_service.html The home page will not load, well, not for me anyway. Not sure if that means anything or not.

BTW. If you are paying more than about AUD9.5/board, you could probably do better. Try Banggood. They modded their price for me last time

onyx007, 2018-04-15:
Thank you for the link Colin, I will use it and ask. I ordered from Banggood.

ColinMarsh, 2018-04-15:
Hi @onyx007, you are welcome

onyx007, 2018-04-15:
If it can be on any help, here is what I found in the Q&A about this product on banggood, it is not completed but getting close.
https://docs.google.com/spreadsheets/d/1VH7ejolhriWXEWxKkzGnvYzx4yaj4Sg_0Dw3ma5qNSk/edit#gid=0

ColinMarsh, 2018-04-16:
Thanks @onyx007. I wonder why DOIT don't define GPIO's contiguously. Does anyone know or heard any discussion on the matter?

Isaquedeveloper, 2018-04-20:
"Also don't accidentally cross ground on the VIN side with ground on the 3.3v side, you will blow your board when powered via USB." Yes. Don´t make this. My ESP32 Died.

johansundstrom, 2018-04-23:
Let us all help onyx007 to fulfill the complete Google document

davenippon, 2018-04-23:
I found a site that seems to have some good info here:
http://esp-idf.readthedocs.io/en/latest/get-started/get-started-devkitc.html

For example, it explains that:
"Some of broken out pins are used internally be the ESP32 module to
communicate with SPI memory. They are grouped on one side of the board
besides the USB connector and labeled D0, D1, D2, D3, CMD and CLK. In
general these pins should be left unconnected or access to the SPI flash
memory / SPI RAM may be disturbed."
and
" GPIO16 and 17 are used internally by the ESP32-WROVER module. They are
broken out and avialable for use only for boards that have the ESP-WROOM-32
module installed."

I'll qualify the above with saying I am very, very new to all this, but
this info looked helpful in mapping out the pins on the ESP32 dev board.
Right now I have one hooked up on a breadboard with a waterproof DS18B20
sensor on my desk to check the temp of the water in my electric kettle and
display it on a 2x2 E-Paper display before making coffee :) I might switch
to a more basic board for that and try to do something cooler with the
ESP32 since it has Wi-Fi, Blue tooth etc.

filchristou, 2018-04-26:
I agree with davenippon here. 
It seems that this particular board is an ESP-DEVKITV1 (or ESP-DEVKITJ), which has some pins left out(or should I say left in), given that they are not of a particular use rather than maybe monitoring.

The full official schematic for the board is here : https://dl.espressif.com/dl/schematics/ESP32-DevKitJ-v1_sch.pdf .

Now, imagine some pins of it never seeing the surface. I believe this particular image can help you through : https://raw.githubusercontent.com/playelek/pinout-doit-32devkitv1/master/pinoutDOIT32devkitv1.png

johansundstrom, 2018-04-27:
I recently sent an email to DOIT.am. I hope to get a response which I will post here. Keep fingers crossed

laurentopia, 2018-04-28:
I am trying to connect a [parallel TFT](https://www.amazon.com/gp/product/B074DRDDX2/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1) to [this board](https://www.amazon.com/gp/product/B079PVCF2G/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1). using the TFT_eSPI library. Oh joy.
First, I learned to stay away from the two TX&RX as they prevent USB upload, maybe linked to the EPROM. Then I wires everything that needed to be within the 0-31 range and good luck on a board with only 24 usable pins.
So far I get a faint pulse, nothing else. Someone told me that this TFT needs 5V signal because it's made for Uno. Would that explain that?

Funny how a couple dollars make us cope with this nonsense board and this nonsense LCD.

volkerpetersen, 2018-05-03:
Has anyone experience powering the DOIT ESP32 with a 3.7V Lithium battery? I've tried putting the 3.7V on Vcc - failed.  Putting it on the USB port fails too.  I even used a low-voltage drop regulator (0.4V at 400mA) and put it on 3V3 - failed.  Last and final test was putting the regulated 3.3V on to Pin 2 (Vout) of the onboard voltage regulator - failed.  
In all cases the board doesn't boot and the current draw cycles between 50 and 100mA with about a 1 second frequency.  Thus, it looks as if the unit starts up, voltage drops, causing the device to reset, current drops and the whole cycle start all over again.

Any idea / suggestions?

laurentopia, 2018-05-04:
@davenippon No I didn't try that one, it's for mega though...

ColinMarsh, 2018-05-04:
@johansundstrom Did you get a response? I have sent 2 emails through their "Support" link and one to a DOIT development engineer's address that I lucked upon. Unfortunately, I am still waiting. Nice product, absolute shit support

onyx007, 2018-05-04:
I sent an email to support as well weeks ago and still waiting ...

johansundstrom, 2018-05-04:
@ColinMarsh No answer at all :-( I signed up for their BBS at http://bbs.smartarduino.com/ and is still pending member since april 24. I keep asking myself - WHY this non responsive behaviour?

lrmoreno007, 2018-05-04:
Why do not you transfer this conversation to another site? It's not the best place.

Thanks.

spiffomatic64, 2018-06-06:
@volkerpetersen my doit v1 board has a ams1117 voltage regulator onboard, and based on a few other pages I found online:

Voltage regulator on board for input voltage (5-12V on Vin pin)

According to the spec sheet https://static.chipdip.ru/lib/552/DOC001552809.pdf , it looks like that wants  4.8V<= VIN <=10.3V

vishnu-chalil, 2018-07-21:
What is the default rx1 tx1 pin in this board

filchristou, 2018-07-21:
There are three UART controllers available on the ESP32 chip.
Usually they are referred as UART0 UART1 UART2. UART0 is used by default for console output.
You asked for default pins on UART1. The answer is that you don't really need them since you always should set them with uart_set_pin().
If you insist please check a detailed answer by kolban : 
https://esp32.com/viewtopic.php?t=1033

vishnu-chalil, 2018-07-21:
@filchristou  how do i use" uart_set_pin()" to set for UART 1 in an ESP-VROOM-32 board

filchristou, 2018-07-23:
You can find documentation here : http://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/uart.html .
An example would be : 

```
#define TXD_PIN (GPIO_NUM_32)
#define RXD_PIN (GPIO_NUM_33)

    //----initialize UART-----//
    uart_config_t uart_config = {}; 
    uart_config.baud_rate = 115200;
    uart_config.data_bits = UART_DATA_8_BITS;
    uart_config.parity = UART_PARITY_DISABLE;
    uart_config.stop_bits = UART_STOP_BITS_1;
    uart_config.flow_ctrl = UART_HW_FLOWCTRL_DISABLE;
    uart_param_config(UART_NUM_1, &uart_config);

    uart_set_pin(UART_NUM_1, TXD_PIN, RXD_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
    uart_driver_install(UART_NUM_1, RX_BUF_SIZE * 10, 0, 0, NULL, 0);
```

You can find more examples on peripherals here https://github.com/espressif/esp-idf/tree/master/examples/peripherals

vishnu-chalil, 2018-08-02:
Thank you @filchristou

cspwcspw, 2018-09-12:
@laurentopia   Your LCD is parallel, it is not SPI (Serial Peripheral Interface).   So I think you are using the wrong library.    I uploaded a project recently using a shield like this at https://github.com/cspwcspw/ESP32_CamToLCD    If you ignore the camera code and just use the LCD part of that there is a simple test suite that should at least show some colours and rectangles.   I don't use any Adafruit libraries,

mcx, 2018-09-26:
@Isaquedeveloper I also did this (i.e. short VIN to GND accidentally), but the board isn't completely dead. You can power it directly from VIN and GND using a power supply (from 5V to 12V). For a long term fix, you would need to replace the SS14 Schottky diode (between the AMS1117 regulator and the micro USB).

csaraivarocha, 2018-10-16:
Hello everybody!
Does anyone here have the DoIt DevKit Esp32 (30pins) fritzing part? I haven't found it at all.
Thanks in advance.

nicechocolate, 2019-02-16:
I found a diagram for this board on Random Nerd tutorials, though mine has the short names silk-screened on anyway.  The 5V and its ground are connected to the modem board, a PIR and a couple of LEDS; the 3.3V and its ground are connected to the RTC and previously also a SSD1306 0.96" OLED display.  The modem needs a good couple of amps when connecting or sending SMS, so there's a backup battery which seems to give enough for that.
I'm using hardwareSerial.h library and UART2 on PIO16 and 17.  Working like a charm, connected to a SIM800C development board.  My SIM880C board has regulator, battery pins and Vmpu input to set the inout/output voltage levels.  RTC via I2C on PIO21 and 22, which are the default in the wire.h library.
Only problem I gave myself was using the wrong Serial monitor baud rate and thinking the modem wasn't working.
Maybe this will help someone.
[image]

ShathyaKodees, 2019-04-25:
Hi All, 
Whether the library <ESPAsyncWebServer.h> will support DOIT ESP32 DEVKIT V1 in Arduino IDE ?

me-no-dev, 2019-04-25:
why wouldn't it? What is special about that board? Library support is architecture based and not board based. Any ESP32/8266 can use that library

NilsBe, 2019-09-23:
> Here you go :) It also took me a while to find the schematics. Just not the following, the GPIO pins follow the numbers 1-27 (lower right pin out schematics), but the analog pins labeled as A6-A9 [...]

I can strongly agree. I already blew three boards by connecting ground on the Vin side with ground on the 3.3v side till I noticed that this was the reason. I still don't understand that there is a different ground. I feel there should be more details about this.

DavidAntliff, 2019-09-23:
>  I already blew three boards by connecting ground on the Vin side with ground on the 3.3v side

This makes no sense to me - see [earlier](https://github.com/espressif/arduino-esp32/issues/544#issuecomment-353897633) [comments](https://github.com/espressif/arduino-esp32/issues/544#issuecomment-370173295). There's only one ground on the board according to the schematics. Are you sure you're not using a Vin that isn't 5V (e.g. 3.3V), then connecting USB and blowing the D1 diode?

[21 more replies]