espressif/arduino-esp32 #545: Unable to use WifiClientSecure to connect to Google services

salqadri, 2017-07-30:
I want to connect to Google servers, so I tried replacing the test_root_ca variable in the WiFiClientSecure.ino example with the Equinox pem file contents but I keep getting a failure. On digging in the it seems mbedtls gives me back error -0x2700 which means that it failed certification verification, but doesn't explain why it failed it. Any ideas on how I can make this work?

copercini, 2017-07-30:
Which google service are you trying to use?
With which certificate?

salqadri, 2017-07-30:
I'm just trying to do a GET on google.com. The root CA for google.com is Equifax Secure Certificate Authority.

salqadri, 2017-07-30:
Nvm; the root CA for google.com turns out to be GeoTrust Global CA, and now it works great.

closed 2017-07-30 by salqadri