espressif/arduino-esp32 #549: Code that compile and run onEsp8266 compile but not run

pbecchi, 2017-07-31:
I am porting a large code from Esp8266. This code was ported to esp8266 from avr to esp8266 last year and run smoothly.
I am trying now to Port it to Esp32.
It compile with no errors but it doesn't run, just before start it give the error:

Guru meditation error of type loadProhibited on  core 0.

This may be due to coding practices , sometime old and deprecated, that are compatible with Esp8266 but not with Esp32.

I discovered one of those:

- use of `extern` for a declaration out of main code.    Having a declaration of a static variable on a library module (eg. `ds1307rtc RTC=ds1307rtc();` on ds1307rtc.cpp with `extern ds1307rtc RTC;` in the main code) lead to the above error, that go away  moving the declaration from the library module to the main code.

This is , as far as I know, quite a common practice(the example is from a well know RealTime clock library) and I ask me why it is not accepted by Esp32.

I wonder if there could be other cases where old coding practices could lead to similar difficult to discover errors

copercini, 2017-07-31:
> Guru meditation error of type loadProhibited on core 0.

Did you decoded it with [EspExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder) ?

pbecchi, 2017-07-31:
Thanks, but I think it is not necessary since I know exactly where the error is.
The error is on the definition `DS1307RTC RTC= DS1307RTC(); // create an instance for the user`
last line of DS1307RTC.cpp .
For some reasons ,the instance ,if created inside the routine code, generate the error.
If I move the above line from the library code to the main code the error disappears and the program run smoothly.

 I have created the issue in order to understand why this append.

igrr, 2017-07-31:
There should not be any issues with extern definitions... It is likely that the issue you see is triggered by something else, but because you get different memory layout with/without extern, the issue is not observed when you remove extern.
Most likely this is memory corruption of some sort, but as @copercini has suggested, it is best to check the exception decoder output.

pbecchi, 2017-08-01:
Thank for your help!
I want to explain better how is the original code and where is the error:

 `/*
 * I2CRTC.cpp - library for I2C RTC
  
  Copyright (c) Michael Margolis 2009
  This library is intended to be uses with Arduino Time.h library functions

  The library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
  
  30 Dec 2009 - Initial release
  5 Sep 2011 updated for Arduino 1.0
  
  23 Dec 2013 -- modified by Ray Wang (Rayshobby LLC) to add support for MCP7940
 */

#include "i2crtc.h"
#include <Wire.h>

static uint8_t _addrs[] = {DS1307_ADDR, MCP7940_ADDR, PCF8563_ADDR};

uint8_t I2CRTC::addr = 0;

I2CRTC::I2CRTC()
{
  Wire.begin();
}

bool I2CRTC::detect()
{
  addr = 0;
  for(uint8_t i = 0;i < sizeof(_addrs);i ++) {
    Wire.beginTransmission(_addrs[i]);
    Wire.write((uint8_t)0x00);
    if(Wire.endTransmission()==0) { // if chip detected successfully
      addr = _addrs[i];
      return true;
    }
  }
  return false;
}

// PUBLIC FUNCTIONS
time_t I2CRTC::get()   // Aquire data from buffer and convert to time_t
{
  tmElements_t tm;
  read(tm);
  return(makeTime(tm));
}

void I2CRTC::set(time_t t)
{
  tmElements_t tm;
  breakTime(t, tm);
  //tm.Second |= 0x80;  // stop the clock   ray: removed this step
  //write(tm); 
  //tm.Second &= 0x7f;  // start the clock  ray: moved to write function
[... 86 lines cut ...]
  return ((num/16 * 10) + (num % 16));
}
I2CRTC RTC = I2CRTC(); // create an instance for the user
`
In the main code there is an`extern I2CRTC RTC;` statement  in order to define and allow use of RTC.

This code give the error
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400d0851  PS      : 0x00060730  A0      : 0x800d0e9c  A1      : 0x3ffe3bc0
A2      : 0x3ffc1ca8  A3      : 0x00000000  A4      : 0x00000016  A5      : 0x000186a0
A6      : 0x3ffaff48  A7      : 0x00000000  A8      : 0x800d084d  A9      : 0x3ffe3ba0
A10     : 0x3ffc1ca8  A11     : 0x00000016  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x3ffc1024  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000001c  LBEG    : 0x400012c5  LEND    : 0x400012d5  LCOUNT  : 0xfffffffc

and decoded:

0x400d0851: I2CRTC::read(tmElements_t&) at d:\Users\pbecc\AppData\Local\Temp\arduino_build_435107\sketch/i2crtc.cpp line 149

where: line 149 is:

uint8_t I2CRTC::bcd2dec(uint8_t num)

The way I correct the error is to take out :`I2CRTC RTC = I2CRTC(); // create an instance for the user` and to put it in the main replacing `extern I2CRTC RTC;`.

copercini, 2017-08-01:
Are you using Arduino as IDF component and you are calling it from a file like main.c ?

pbecchi, 2017-08-01:
No, it is a standard arduino program,

pbecchi, 2017-08-01:
Here the code:
`     /*
 Name:		RTClibTEST.ino
 Created:	30/07/2017 15:15:30
 Author:	pbecc
*/
/*
* TimeRTC.pde
* Example code illustrating Time library with Real Time Clock.
* This example is identical to the example provided with the Time Library,
* only the #include statement has been changed to include the DS3232RTC
* library.
*/
#include <Arduino.h>
#include "i2crtc.h"
#include <Time.h>
#include <Wire.h>      //http://arduino.cc/en/Reference/Wire
//I2CRTC RTC = I2CRTC();
extern I2CRTC RTC;
void setup(void)
{
	Serial.begin(115200);
	setSyncProvider(RTC.get);   // the function to get the time from the RTC
	if (timeStatus() != timeSet)
		Serial.println("Unable to sync with the RTC");
	else
		Serial.println("RTC has set the system time");
}

void loop(void)
{
	digitalClockDisplay();
	delay(1000);
}

void digitalClockDisplay(void)
{
	// digital clock display of the time
	Serial.print(hour());
	printDigits(minute());
	printDigits(second());
	Serial.print(' ');
	Serial.print(day());
	Serial.print(' ');
	Serial.print(month());
	Serial.print(' ');
	Serial.print(year());
	Serial.println();
}

void printDigits(int digits)
{
	// utility function for digital clock display: 
	// prints preceding colon and leading 0
	Serial.print(':');
	if (digits < 10)
		Serial.print('0');
	Serial.print(digits);
}
 `

me-no-dev, 2017-08-01:
as @igrr said, extern is not an issue (look at SPI, Wire, Serial and so on)

pbecchi, 2017-08-01:
Then the issue is last statement on cpp file.
The one that i have relocated to the ino file.

closed 2018-04-19 by everslick