espressif/arduino-esp32 #550: WiFiClientSecure does not connect to some websites

chemicstry, 2017-08-01:
I'm having problem connecting to SSL websites (specifically proxied by CloudFlare) with WiFiClientSecure. Other websites seem to work fine.

Here is console output:
```
[I][ssl_client.cpp:45] start_ssl_client(): Free heap before TLS 191592
[I][ssl_client.cpp:47] start_ssl_client(): Starting socket
[I][ssl_client.cpp:75] start_ssl_client(): Seeding the random number generator
[I][ssl_client.cpp:84] start_ssl_client(): Setting up the SSL/TLS structure...
[I][ssl_client.cpp:108] start_ssl_client(): WARNING: Use certificates for a more secure communication!
[I][ssl_client.cpp:153] start_ssl_client(): Performing the SSL/TLS handshake...
[E][ssl_client.cpp:25] handle_error(): SSL - A fatal alert message was received from our peer
[E][ssl_client.cpp:27] handle_error(): MbedTLS message code: -30592
[E][WiFiClientSecure.cpp:102] connect(): lwip_connect_r: 11
[I][ssl_client.cpp:204] stop_ssl_socket(): Cleaning SSL connection.
[I][ssl_client.cpp:204] stop_ssl_socket(): Cleaning SSL connection.
```

However, when using `https_request` example from esp-idf I can connect to the same website (CloudFlare) successfully. I believe it could be some mbedtls configuration issue in `ssl_client.cpp`

copercini, 2017-08-01:
Quite strange, I tested with codepen.io and laravel.com (which looks like proxied by CloudFlare) and works fine

What URL does this occur in?

chemicstry, 2017-08-01:
Maybe they are using their own certificates? I'm using CF's issued certificates on https://gaben.win/ and it gives me the error above.

copercini, 2017-08-01:
Found it! https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiClientSecure/src/ssl_client.cpp#L138

setting host as "gaben.win" everything works fine

chemicstry, 2017-08-01:
It works now. Thank you.

It was a problem with multi-host certificate right? Is it still WIP?

copercini, 2017-08-01:
It's just pass the "host" from arduino sketch to this mbedtls API.

I'm quite busy right now, but on weekend will make a PR solving it...

copercini, 2017-08-24:
@chemicstry Should be fixed now =)

closed 2017-08-24 by copercini