espressif/arduino-esp32 #551: Linux install instructions not working for me.

CelliesProjects, 2017-08-01:
If I use the installation instructions on https://github.com/espressif/arduino-esp32 for Debian/Ubuntu Linux I get this output:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
git is already the newest version.
The following packages were automatically installed and are no longer required:
  libboost-program-options1.54.0 libcec3 libfdk-aac0 liblockdev1 libplatform1
  libwireshark3 libwiretap3 libwsutil3 python3-markups python3-pyqt4
  python3-sip schroot schroot-common
Use 'apt-get autoremove' to remove them.
0 to upgrade, 0 to newly install, 0 to remove and 82 not to upgrade.
--2017-07-29 16:21:01--  https://bootstrap.pypa.io/get-pip.py
Resolving bootstrap.pypa.io (bootstrap.pypa.io)... 151.101.36.175
Connecting to bootstrap.pypa.io (bootstrap.pypa.io)|151.101.36.175|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1595408 (1,5M) [text/x-python]
Saving to: ‘get-pip.py.2’

100%[======================================>] 1.595.408   6,46MB/s   in 0,2s   

2017-07-29 16:21:02 (6,46 MB/s) - ‘get-pip.py.2’ saved [1595408/1595408]

The directory '/home/cellie/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/cellie/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
/tmp/tmpdlfcQv/pip.zip/pip/_vendor/requests/packages/urllib3/util/ssl_.py:318: SNIMissingWarning: An HTTPS request has been made, but the SNI (Subject Name Indication) extension to TLS is not available on this platform. This may cause the server to present an incorrect TLS certificate, which can cause validation failures. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/security.html#snimissingwarning.
/tmp/tmpdlfcQv/pip.zip/pip/_vendor/requests/packages/urllib3/util/ssl_.py:122: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/security.html#insecureplatformwarning.
Requirement already up-to-date: pip in /usr/local/lib/python2.7/dist-packages
/tmp/tmpdlfcQv/pip.zip/pip/_vendor/requests/packages/urllib3/util/ssl_.py:122: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/security.html#insecureplatformwarning.
The directory '/home/cellie/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/cellie/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Requirement already satisfied: pyserial in /usr/lib/python2.7/dist-packages
/usr/local/lib/python2.7/dist-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:318: SNIMissingWarning: An HTTPS request has been made, but the SNI (Subject Name Indication) extension to TLS is not available on this platform. This may cause the server to present an incorrect TLS certificate, which can cause validation failures. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/security.html#snimissingwarning.
  SNIMissingWarning
/usr/local/lib/python2.7/dist-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:122: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning
fatal: destination path 'esp32' already exists and is not an empty directory.
```
The last error about the missing path is because I retried the install a couple of times.

When compiling an empty sketch for 'ESP32 Dev module' I get this output:
```
Arduino: 1.8.3 (Linux), Board: "ESP32 Dev Module, 80MHz, 921600, None"

/tmp/arduino-1.8.3/arduino-builder -dump-prefs -logger=machine -hardware /tmp/arduino-1.8.3/hardware -hardware /home/cellie/.arduino15/packages -hardware /home/cellie/Arduino/hardware -tools /tmp/arduino-1.8.3/tools-builder -tools /tmp/arduino-1.8.3/hardware/tools/avr -tools /home/cellie/.arduino15/packages -built-in-libraries /tmp/arduino-1.8.3/libraries -libraries /home/cellie/Arduino/libraries -fqbn=espressif:esp32:esp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=none -ide-version=10803 -build-path /tmp/arduino_build_382935 -warnings=none -prefs=build.warn_data_percentage=75 -verbose /tmp/untitled1131754400.tmp/sketch_jul29a/sketch_jul29a.ino
/tmp/arduino-1.8.3/arduino-builder -compile -logger=machine -hardware /tmp/arduino-1.8.3/hardware -hardware /home/cellie/.arduino15/packages -hardware /home/cellie/Arduino/hardware -tools /tmp/arduino-1.8.3/tools-builder -tools /tmp/arduino-1.8.3/hardware/tools/avr -tools /home/cellie/.arduino15/packages -built-in-libraries /tmp/arduino-1.8.3/libraries -libraries /home/cellie/Arduino/libraries -fqbn=espressif:esp32:esp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=none -ide-version=10803 -build-path /tmp/arduino_build_382935 -warnings=none -prefs=build.warn_data_percentage=75 -verbose /tmp/untitled1131754400.tmp/sketch_jul29a/sketch_jul29a.ino
Using board 'esp32' from platform in folder: /home/cellie/Arduino/hardware/espressif/esp32
Using core 'esp32' from platform in folder: /home/cellie/Arduino/hardware/espressif/esp32
Detecting libraries used...
"/home/cellie/.arduino15/packages/Simba/tools/xtensa-esp32-elf-gcc/1.22.0-59/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/config" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/bluedroid" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/app_update" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/bootloader_support" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/bt" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/driver" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/esp32" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/ethernet" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/fatfs" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/freertos" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/jsmn" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/log" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/mdns" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls_port" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/nvs_flash" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/openssl" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/soc" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/spi_flash" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/sdmmc" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/tcpip_adapter" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/ulp" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/vfs" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/wear_levelling" "-I/home/cellie/Arduino/h [... cut ...]
/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/soc/soc/soc.h:163:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((_r), REG_SET_BIT);                                                                    \
             ^
/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/soc/soc/io_mux_reg.h:111:5: note: in expansion of macro 'REG_SET_BIT'
     REG_SET_BIT(PIN_NAME, FUN_PD);
     ^
exit status 1
Error compiling for board ESP32 Dev Module.
```
Am I missing something here?

tobozo, 2017-08-01:
It's loading xtensa from /home/cellie/.arduino15/packages/Simba/tools/xtensa-esp32-elf-gcc/ but it should load it from a subfolder of /home/cellie/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf

lonerzzz, 2017-08-28:
@CelliesProjects Did you manage to resolve this issue?

tobozo, 2017-08-28:
[From the recent commit](https://github.com/CelliesProjects/aquacontrol32/commit/4fe6ae474eed7e3ba47245d1e3500adb619f6cce) a day ago it looks like @CelliesProjects managed to use Arduino IDE after all :-)

CelliesProjects, 2017-08-28:
@lonerzzz Nope. did most definitely NOT. Got it somehow working on my main desktop pc -with a lot of botching- even before I opened this issue, but I was never able to get it working on any of my other devices. ( All running some flavour of Mint )

CelliesProjects, 2017-08-28:
@tobozo See reply I gave lonerzzz.

CelliesProjects, 2017-08-28:
Just tried it again on my laptop.
```
cellie@cellie-ThinkPad-T500 ~/Arduino/hardware/espressif $ rm -R ~/Arduino/hardware/espressif
cellie@cellie-ThinkPad-T500 ~/Arduino/hardware/espressif $ mkdir -p ~/Arduino/hardware/espressif
cellie@cellie-ThinkPad-T500 ~/Arduino/hardware/espressif $ cd ~/Arduino/hardware/espressif
cellie@cellie-ThinkPad-T500 ~/Arduino/hardware/espressif $ clear

cellie@cellie-ThinkPad-T500 ~/Arduino/hardware/espressif $ git clone https://github.com/espressif/arduino-esp32.git esp32
Cloning into 'esp32'...
remote: Counting objects: 5103, done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 5103 (delta 0), reused 2 (delta 0), pack-reused 5098
Receiving objects: 100% (5103/5103), 80.17 MiB | 4.47 MiB/s, done.
Resolving deltas: 100% (2615/2615), done.
Checking connectivity... done.
cellie@cellie-ThinkPad-T500 ~/Arduino/hardware/espressif $ cd esp32/tools/
cellie@cellie-ThinkPad-T500 ~/Arduino/hardware/espressif/esp32/tools $ python get.py
System: Linux, Info: Linux-4.4.0-53-generic-x86_64-with-LinuxMint-18.1-serena
Platform: x86_64-pc-linux-gnu
Downloading xtensa-esp32-elf-linux64-1.22.0-61-gab8375a-5.2.0.tar.gz
Done
Extracting xtensa-esp32-elf-linux64-1.22.0-61-gab8375a-5.2.0.tar.gz
Done
cellie@cellie-ThinkPad-T500 ~/Arduino/hardware/espressif/esp32/tools $ 
```
And again no luck.
Only the 'standard' Arduino's are shown in the `Board:` selection menu.

copercini, 2017-08-28:
@CelliesProjects go to arduino menu: File > preferences
What is your sketchbook location?

CelliesProjects, 2017-08-28:
@copercini On both pc's I have acces to it is '/home/cellie/Arduino'

MarkyAD, 2017-08-30:
Working Mint system here - solution was to forget pip and install pyserial-3.4 from tarball (works here for both arduino-esp32 and esp-idf).

MarkyAD, 2017-08-30:
Working mint system here... not sure if its applicable but the only problems I had were with pip and an old version of pyserial that was already installed. Forgetting pip and installing pyserial-3.4 from tarball got me working in both arduino-esp32 and esp-idf. From Cellies first post, it sure is finding a lot of Simba libraries installed in "/home/cellie/.arduino15/packages/Simba/"... I have nothing esp32 related installed under .arduino15, only in hardware under my sketch root.

CelliesProjects, 2017-08-30:
@MarkyAD This solved my problem! 
So, install python3.4 not from the install script, but from tarball.
Thanks for all the help!

dannybackx, 2017-12-30:
The installation instructions are somewhat flawed. I ended up getting it to work on Ubuntu 17.10, with one change.
The line that mentions "cd /Applications/Arduino_*/Contents/java/hardware/" looks really odd to me. Copied from another platform but untested?

I had unpacked the IDE into ~/src/arduino/arduino-1.8.5 and didn't see "hardware" under "java".
I think from the infamous line above, the "Contents/java" piece should go.

closed 2018-04-19 by CelliesProjects