espressif/arduino-esp32 #553: WiFi.config() can not set DNS.

nnnntosh, 2017-08-02:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			1/Aug/2017
IDE name:							Arduino IDE
Flash Frequency:					?
Upload Speed:						115200

### Description:

I am trying to set DNS with `WiFi.config()`, but DNS lookup ( `gethostbyname()`) fails.
```cpp
WiFi.config(ip, gateway, subnet, dns1, dns2);
```

I tried debugging ESP-IDF. and I have found that DNS response was dropped. 
This is the code below.

esp-idf/components/lwip/core/dns.c
```cpp
        /* Check whether response comes from the same network address to which the
           question was sent. (RFC 5452) */
        if (!ip_addr_cmp(addr, &dns_servers[entry->server_idx])) {
          /* call callback to indicate error, clean up memory and return */
          goto responseerr;
        }
```

`ip_addr_cmp()` checks that `addr->type` matches, but they were not the same.

I checked how to set up DNS with `WiFi.config()`. Nothing is set to `addr->type`, it is undefined value.

arduino-esp32/libraries/WiFi/src/WiFiSTA.cpp
```cpp
    if(dns1 != (uint32_t)0x00000000) {
        // Set DNS1-Server
        d.u_addr.ip4.addr = static_cast<uint32_t>(dns1);
        dns_setserver(0, &d);
    }
```

I also found that DNS works correctly when I modify the code as follows.

```cpp
    if(dns1 != (uint32_t)0x00000000) {
        // Set DNS1-Server
        d.u_addr.ip4.addr = static_cast<uint32_t>(dns1);
        IP_SET_TYPE(&d, IPADDR_TYPE_V4);
        dns_setserver(0, &d);
    }
```

By the way, I think that `ip_addr_set_ip4_u32()` should be used,
It is not possible because the following build error appears due to `gcc option [- Werror = address]`.

```
error: the address of 'd' will always evaluate as 'true' [-Werror=address]
```

me-no-dev, 2017-08-02:
Nice catch!

closed 2017-08-02 by me-no-dev

hutje, 2021-07-05:
WiFi.setDNS() isn't recognized as function... It is once called in the esp32 WIFI library as keyword but never used.