espressif/arduino-esp32 #555: Guru Meditation Error on running WifiScan example.

iAchyut, 2017-08-02:
Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x401123ee  PS      : 0x00060c30  A0      : 0x80112550  A1      : 0x3ffcf474  
A2      : 0x3ffc7190  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3ffc2c13  
A6      : 0xa99afdff  A7      : 0x3ffc66dc  A8      : 0x0008fae7  A9      : 0x3ffcf410  
A10     : 0x3ffc294c  A11     : 0x3ffcf482  A12     : 0x3ffc2940  A13     : 0x3ffcf4a5  
A14     : 0x00000014  A15     : 0x00000000  SAR     : 0x0000001c  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400885a8  LEND    : 0x400885de  LCOUNT  : 0x00000000  

Backtrace: 0x401123ee:0x3ffcf474

CPU halted.

copercini, 2017-08-02:
If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

iAchyut, 2017-08-03:
There was power supply issue and poor usb cable. Thanks for the help

closed 2017-08-03 by iAchyut