espressif/arduino-esp32 #556: ArduinoOTA is borked, after update to IDF libs to 9314bf0

everslick, 2017-08-02:
### Hardware:
Board:							ESP WROOM 32
Core Installation/update date:			02/08/2017

### Problem:
For at least two people on Gitter the

commit ca7f6cc516ce842cf7c6b35dbde4b0d4a5a86404
Author: me-no-dev <hristo@espressif.com>
Date:   Tue Aug 1 08:51:04 2017 +0300

    Update IDF libs to 9314bf0

broke ArduinoOTA. The symptom is that the update file is transfered, as it seems completely, but then the updater stops with error 4 (end failed).

### Sketch:
Happens even with the standard BasicOTA example

Both reporters used a ESP WROOM 32 Breakout Board.

szerwi, 2017-08-02:
I have exactly the same problem.

copercini, 2017-08-03:
Related: https://github.com/espressif/esp-idf/issues/870

Gun-neR, 2017-08-11:
Yes, same issue here.  Wemos Lolin32 v1.0.0  Chip rev.0  IDE 1.8.3  Core? latest I think?

Serial monitor gets to... 98% then Error[4]: End Failed. And IDE shows... [ERROR]: 143614361436.

jrullan, 2017-08-13:
Exactly same issue. Im using the Lolin32, same result as Gun-neR

closed 2017-08-17 by me-no-dev