espressif/arduino-esp32 #557: WiFiAPClass::softAP should fail with a reasonable error code and clean after itself

yaqwsx, 2017-08-02:
When you provide a short password this function fails. Such bugs are hard to debug as the function is black box and there are multiple ways it can fail (and you can learn about them only by studying the source code). Also, when the function fails, it doesn't fail atomically - e.g. in the case of short password, softAP is started with the default values.

Could this function fail with an error code? Or at least, for the sake of backward compatibility, could you provide a function to retrieve last error code/error description?

lonerzzz, 2017-08-15:
@yaqwsx Can you provide the code snippet that reproduces this issue?

yaqwsx, 2017-08-16:
The dead-simple example:

```
#include <Arduino.h>
#include <WiFi.h>

void setup() {
    Serial.begin( 115200 );
    WiFi.softAP( "YouShouldSeeMe", "x" );
    Serial.println( WiFi.softAPIP() ); // This should also fail!
}

void loop() {

}
```

`WiFiAPClass::softAP` fails because of short password:
- however AP with SSID `ESP_XXXXXX` is started and therefore `WiFiAPClass::softAPIP` does not fail.
- there is no way to tell why `WiFiAPClass::softAP` fails - by briefly looking at the code, there are at least 4 ways to fail: cannot change WiFi mode, SSID is wrong, password is wrong or cannot se WiFi configuraiton.

closed 2019-08-15 by stale[bot]