espressif/arduino-esp32 #558: Arduino IDE OTA option disappears after computer reset

szerwi, 2017-08-03:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			03.08.2017, I use 
                                                                        5d927847106f84e4714ee2a50545880b0911c1d8 version
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
When I upload sketch via USB after restarting Arduino IDE there is option to upload sketch via OTA. But, when I restart PC this option disappears.

### Sketch:
BasicOTA example

### Debug Messages:
-

me-no-dev, 2017-08-03:
your computer needs to search for it ;) give it some time

closed 2017-08-03 by me-no-dev

szerwi, 2017-08-03:
But how much time? When I was trying to upload code there was some errors.