espressif/arduino-esp32 #559: SerialToTelnet doesn't show everything

szerwi, 2017-08-03:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			using this version: 5d927847106f84e4714ee2a50545880b0911c1d8
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
Code that should be shown via Telnet: https://pastebin.com/45tFBb26
Code that is shown via Telnet: https://pastebin.com/4q5hHJVd

Half of the code isn't shown.

### Sketch:
TelnetToSerial example and code to display on Serial. Serial and Serial1 are set to 115200 speed.

### Debug Messages:
-

lonerzzz, 2017-08-29:
@szerwi Does this work any differently when you change the baud rate or send your data more slowly?

jestebanes, 2017-09-08:
Hello, I am working on a similar trouble, writes to serial port are asyncronous and maybe if you are writing too fast the serial buffer overflows.

Try to make the buffer greater
In HardwareSerial.cpp, line 29 you can find the serial buffer size declaration, change to 1024 and try again

Best regards

lonerzzz, 2017-09-28:
@szerwi Is there an update on this issue? Please provide more information if you need it kept open.

closed 2018-04-19 by everslick