espressif/arduino-esp32 #560: PlatformIO cannot pick the correct serial port for flashing an ESP-WROVER-KIT

ButchAnton, 2017-08-03:
The ESP-WROVER-KIT has two serial ports (A and B) and only the B port can be used for flashing.  However, when PlatformIO picks the serial port to use for flashing, it uses the first one, and the flashing fails.  PlatformIO needs a way to identify the correct serial port to use for flashing.

Here is the output from an attempt to flash a simple hello_world test using PlatformIO's definitions for the ESP 32 dev board and changing the flash frequency to 40MHz:

platformio run --target upload
[Thu Aug  3 10:19:48 2017] Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
--------------------------------------------------------------------------------
Collected 14 compatible libraries
Looking for dependencies...
Project does not have dependencies
xtensa-esp32-elf-g++ -o .pioenvs/esp32dev/firmware.elf -nostdlib -Wl,-static -u call_user_start_cpu0 -Wl,--undefined=uxTopUsedPriority -Wl,--gc-sections -Wl,-EL -T esp32.common
.ld -T esp32.rom.ld -T esp32.peripherals.ld -Wl,-T"esp32_out.ld" .pioenvs/esp32dev/src/main.o -L/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib -L/
Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/ld -L.pioenvs/esp32dev -Wl,--start-group .pioenvs/esp32dev/libFrameworkArduinoVariant.a .pioenvs/esp32de

v/libFrameworkArduino.a -lgcc -lstdc++ -lapp_update -lbootloader_support -lbt -lbtdm_app -lc -lc_nano -lcoap -lcoexist -lcore -lcxx -ldriver -lesp32 -lethernet -lexpat -lfatfs
-lfreertos -lhal -ljsmn -ljson -llog -llwip -lm -lmbedtls -lmdns -lmicro-ecc -lnet80211 -lnewlib -lnghttp -lnvs_flash -lopenssl -lphy -lpp -lrtc -lsdmmc -lsmartconfig -lsoc -ls
pi_flash -ltcpip_adapter -lulp -lvfs -lwear_levelling -lwpa -lwpa2 -lwpa_supplicant -lwps -lxtensa-debug-module -Wl,--end-group
Checking program size
text       data     bss     dec     hex filename
103554    21376    2384  127314   1f152 .pioenvs/esp32dev/firmware.elf
MethodWrapper(["upload"], [".pioenvs/esp32dev/firmware.bin"])
Auto-detected: /dev/cu.usbserial-4013613A
"/Users/butch/.platformio/penv/bin/python2.7" "/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/esptool.py" --before default_reset --after hard_reset --chip
 esp32 --port "/dev/cu.usbserial-4013613A" --before default_reset --after hard_reset --baud 115200 write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 "
/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/sdk/bin/bootloader.bin" 0x8000 "/Users/i831533/src/ESP32/wrover/hello_world/.pioenvs/esp32dev/partitions.bi
n" 0xe000 "/Users/butch/.platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin" 0x10000 .pioenvs/esp32dev/firmware.bin
esptool.py v2.0-beta3
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____
 
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
*** [upload] Error 2
========================== [ERROR] Took 38.09 seconds ==========================

ButchAnton, 2017-08-03:
Ivan addressed this in an upcoming PlatformIO update.

closed 2017-08-03 by ButchAnton

wordyallen, 2018-06-23:
is there a setting in the `ini` to edit this?