espressif/arduino-esp32 #562: Timeout Issue in MQTTClient through ESP32

architmuchhal12, 2017-08-04:
Hello,

I am using a Sparkfun ESP32 with Adafruti FONA 808 to connect to an MQTT Server. 

I am using the following 2 libraries. 
1. TinyGSM [(https://github.com/vshymanskyy/TinyGSM)](url)
2. PubSub Client [(https://github.com/knolleary/pubsubclient)](url)

I have modified the TinyGSM/examples/MqttClient/MQTTClient.ino accordingly.

I am using hardware serial at 115200 baud rate. The issue I am facing is the serial monitor displays the reason for no connection as (-4) which is due to Timeout Value. I also tried increasing the timeout value in the library but with no success.

I know this issue isn't relevant to ESP32 but if I could get any help, it would be really great.

Please suggest me a solution!
Thanks!

**Init Code:**

```
// Select your modem:
#define TINY_GSM_MODEM_SIM800
//#define TINY_GSM_MODEM_SIM900
//#define TINY_GSM_MODEM_A6
//#define TINY_GSM_MODEM_M590

#include <TinyGsmClient.h>
#include <PubSubClient.h>
#include "Adafruit_FONA.h"

// Default pins for Feather 32u4 FONA
#define FONA_RST 25
HardwareSerial fonaSerial(1);
Adafruit_FONA fona = Adafruit_FONA(FONA_RST);

// Your GPRS credentials
// Leave empty, if missing user or pass
const char apn[]  = "wholesale";
const char user[] = "0";
const char pass[] = "0";

TinyGsm modem(fonaSerial);
TinyGsmClient client(modem);
PubSubClient mqtt(client);

const char* broker = "iot.wabashnational.com";
const char* userName = "******";
```

FONA808 initialization: 

```
// Set FONA module baud rate
  fonaSerial.begin(115200, SERIAL_8N1, 33, 32);
  if (! fona.begin(fonaSerial)) {
    Serial.println(F("Couldn't find FONA"));
    while (1);
  }
  Serial.println(F("FONA is OK"));
  Serial.println(F("Found FONA808 (v2) "));
  delay(1000);
```

Serial Monitor Output:

```
Attempting to open comm with ATs
	---> AT
	<--- BU��
	---> AT
	<--- ÿ
	---> AT
	<--- 
	---> AT
	<--- ÿ
	---> AT
	<--- AT
	---> AT
	<--- AT
	---> ATE0
	<--- ATE0
	---> ATE0
	<--- OK
	---> AT+CVHU=0
	<--- OK
	---> ATI
	<--- SIM808 R14.18

OK

FONA is OK
Found FONA808 (v2) 
Initializing modem...
Waiting for network... OK
Connecting through(apn)---> wholesale----->>> OK and Connected to GPRS
Setting up server----->>  Server Set
Connecting to: iot.wabashnational.com ----->> Fail
Error code number for no connection: -4
```

androidcho, 2017-08-06:
Your library is not initialised properly.
Take a look here - https://pubsubclient.knolleary.net/api.html

architmuchhal12, 2017-08-07:
I am not sure what error I am making in initializing the PubSub client. Here is my code for the connection to MQTT Broker

Code - 

```
boolean mqttConnect() {
  Serial.print("Connecting to: ");
  Serial.println(broker);
  if (mqtt.connect("ESP32Client", "user", "password")) {
    Serial.println(" ----->> Fail!");
  }
  else {
    Serial.print("Error code number for no connection: ");
    Serial.println(mqtt.state());
    return false;
  }
```

I have also tried other combinations from the library for the "connect" function, but none of them returned a success.

androidcho, 2017-08-09:
I can't where your WiFi library is initialised as client. Take a look at this tutorial, it'll get you up and running.
https://techtutorialsx.com/2017/04/24/esp32-publishing-messages-to-mqtt-topic/

copercini, 2017-08-09:
@androidcho He are using an external GSM modem (SIM808) instead of Wifi, so don't need the wifi stuff =)

chris-gunawardena, 2017-08-14:
@architmuchhal12 [This](Could even be a network issue or a wrong port, Have you tried with a different mqtt server like ) is a working example on the ESP32. This could be a network or port issue, have you tried connecting to a local server using something like [mosca](https://github.com/mcollina/mosca)?

closed 2018-04-19 by everslick