espressif/arduino-esp32 #564: Ping library

pbecchi, 2017-08-06:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Describe your problem here

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

everslick, 2017-08-06:
While I'm in no position to answer your question, I'm interested in testing your lib. If you are willing to share it, you could just put it in here as a comment or attachment.

pbecchi, 2017-08-06:
Basic functions are working , but I am in process of writing a library with similar footprint as the ESP8266 one.
I have not implemented the function handler that the Esp8266 library have: to implement those I probably have to rewrite it! If it important to do it that way, you have to wait a little more!

everslick, 2017-08-06:
no hurry, thanks!

bill-orange, 2017-08-08:
I can also use your library.  I have odd router disconnects.  After 60+ hours of normal operation, the router will not allow a connection to the ESP32 or EPS8266 devices but the connection status on the ESP side shows the connection is good.  Since the ESP believes the connection is good even my watchdog is not triggered.  

I solved this problem on the ESP8266 systems by having them PING the router once every 6 min. or so.  If the PING fails the EPS8266 restarts and all is well.  Restarting the connection with a new begin() statement would probably work too.

Thus, for the ESP32 systems I need the PING library.

bill-orange, 2017-11-05:
Has there been any progress on this project?

pbecchi, 2017-11-06:
Few days more....and I will provide a library ! It will not 100% compatible with the ESP8266 one!

bill-orange, 2017-11-06:
That would be super!

lonerzzz, 2017-11-06:
Definitely looking forward to seeing it.

bill-orange, 2017-11-08:
Excellent work.  Thank you. Pinging the router is the only reliable way I have found to test the WiFi connection.  I think this is a router issue rather than an EPP32/8266 issue.  I rearranged you example into a function as shown below.

bool pingtest (IPAddress ia) {
  //uint8_t first_byte = IP[0];
  //int ia[4] = { 192, 168, 0, 1 };
  int  i = 0;
  while (Serial.available()) {

    char c = Serial.read();
    delay(100);
    int  val = 0;
    while (c != '.' &&  c != 10 && c != 255) {
      if (c >= '0' && c <= '9') {
        val = val * 10 + (c - '0');
      }
      c = Serial.read();
    }

    ia[i++] = val ;
  }
  IPAddress adr = IPAddress(ia[0], ia[1], ia[2], ia[3]);
  Serial.printf("Ping : %d . %d . %d . %d ->", ia[0], ia[1], ia[2], ia[3]);
  if (ping_start(adr, 4, 0, 0, 5)) {
    //Serial.println("OK");
    return true;
  }
  else {
    //Serial.println("FAILED");
    return false;

  }
}

marian-craciunescu, 2018-02-06:
I can also confirm that your library is working.
Maybe you could do a small wrapper on it

tferrin, 2018-02-17:
Very nice!  Useful for tracking down network problems.

Since the ping library works this is really not an Arduino ESP32 "issue" and so this thread should be marked closed.

bill-orange, 2018-02-17:
Agreed, let’s close it!

robjen, 2018-08-13:
same problem here ... ESP8266 running for many days but lost network connection after few hours so added this ping library as a keep alive on network ... 

now crashes after 40-45 minutes (4 times in a row now)

remove library and happily running well 3 days+

so have to agree there must be a nasty memory leak in there somewhere

closed 2019-08-15 by stale[bot]