espressif/arduino-esp32 #566: Any HTTPS connection results in a stack overflow error

BluCodeGH, 2017-08-07:
### Hardware:
Board:							MHET-Live ESP32-MINI-KIT using ESP-WROOM-32
Core Installation/update date:			9/jul/2017
IDE name:							IDF component
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
Whenever I try to connect to a website over HTTPS, using either the HTTPClient or the WiFiClientSecure, I get a stack overflow error when I call http.begin() or client.connect(): 

```
***ERROR*** A stack overflow in task loopTask has been detected.
abort() was called at PC 0x4008747c on core 1

Backtrace: 0x40087364:0x3ffcf2d0 0x40087463:0x3ffcf2f0 0x4008747c:0x3ffcf310 0x4008519c:0x3ffcf330 0x40086760:0x3ffcf350 0x4008671a:0x00000000
```

And here is what I get with xtensa-esp32-elf-addr2line:

```
$ xtensa-esp32-elf-addr2line -e build/Watch.elf 0x40087364
ESP32/esp-idf/components/esp32/./panic.c:546
$ xtensa-esp32-elf-addr2line -e build/Watch.elf 0x40087463
ESP32/esp-idf/components/esp32/./panic.c:546
$ xtensa-esp32-elf-addr2line -e build/Watch.elf 0x4008747c
ESP32/esp-idf/components/esp32/./panic.c:546
$ xtensa-esp32-elf-addr2line -e build/Watch.elf 0x4008519c
ESP32/esp-idf/components/freertos/./tasks.c:4476
$ xtensa-esp32-elf-addr2line -e build/Watch.elf 0x40086760
??:?
$ xtensa-esp32-elf-addr2line -e build/Watch.elf 0x4008671a
??:?
```
It may be worthwhile to note that my program is very large and divided into multiple files, so this is not the main file. I have also used `make menuconfig` to double the size of all the stacks I could find, which made no difference whatsoever. HTTP connections work fine.

### Sketch:
```cpp

#include "Arduino.h"
#include "email.h"

//#include "WiFiClientSecure.h"
#include "HTTPClient.h"

//WiFiClient client;

String host = "www.howsmyssl.com";
String file = "/a/check";

const char* ca = //Cert

void https() { //Called after connected to wifi and tested connection
  Serial.println("Connecting to: " + String(host));

  HTTPClient http;

  http.begin(String("https://") + host + file, ca);
  int httpCode = http.GET();

  if (httpCode > 0) {
    Serial.print("HTTP Code: ");
    Serial.println(httpCode);

    if(httpCode == HTTP_CODE_OK) {
      String payload = http.getString();
      Serial.println(payload);
    }
  } else {
    Serial.print("Failed, code: ");
    Serial.println(http.errorToString(httpCode));
  }

  http.end();

  /*
  if (client.connect(host.c_str(), 443)) {
    // Connection Succeed.
    Serial.println("Fetching Data.");

    // Get the contents of the csv file
    client.print(String("GET ") + file + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "Connection: close\r\n\r\n");

    unsigned long timeout = millis();
    while (client.available() == 0) {
      if (millis() - timeout > 5000) {
        Serial.println("Client Timeout !");
        client.stop();
        return;
      }
    }
    // Once the response is available,
    // check stuff
    while (client.connected()) {
      String line = client.readStringUntil('\n');
      Serial.println(line);
      if (line == "\r") {
        Serial.println("headers received");
        break;
      }
    }

    while (client.available()) {
      // read line till /n
      String line = client.readStringUntil('\n');

      Serial.println(line);
    }
  } else {
    Serial.println("Connection Failed.");
  }
  */
}
```

### Debug Messages:
I'm not using the IDE and I have set my bootloader and default log level to warning and up. Right now I don't have time to recompile, but if you *really* need it I can get the full boot messages.

lonerzzz, 2017-08-12:
@BluCodeGH Can you add the following code to your setup and have it run every 5 to 10 seconds in your main loop? This will show how your program is using the stack and the heap over time.

`
      UBaseType_t uxHighWaterMark = uxTaskGetStackHighWaterMark( NULL );
      Serial.print( "Memory: " );
      Serial.print(uxHighWaterMark );
      Serial.print( " ");
      Serial.println( esp_get_free_heap_size() );
`

It will allow you to determine how close you are to the limit of the stack and using your heap from the beginning. If you are using too much of the stack, one option is to create another thread with its own stack and do some of your work there.

BluCodeGH, 2017-08-13:
I have updated my code to attempt to find the issue, here is the new code.

```c
#include "Arduino.h"

#include "WiFi.h"
#include "WiFiMulti.h"
#include "WiFiClientSecure.h"

const char* ssid     = "myssid";
const char* password = "mypwrd";

const char*  server = "www.howsmyssl.com";  // Server URL

/*const char* test_root_ca= "not using this";*/

// You can use x.509 client certificates if you want
//const char* test_client_key = "";   //to verify the client
//const char* test_client_cert = "";  //to verify the client

WiFiClientSecure client;
WiFiMulti wifiMulti;

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(115200);
  delay(100);

  Serial.print("Attempting to connect to SSID: ");
  Serial.println(ssid);
  wifiMulti.addAP(ssid, password);

  // attempt to connect to Wifi network:
  int status = wifiMulti.run();
  if (status != WL_CONNECTED) {
    Serial.println(status);
    // uh oh. better sleep and let the user reset
    delay(99000);
  }

  Serial.print("Connected to ");
  Serial.println(ssid);

  //client.setCACert(test_root_ca);
  //client.setCertificate(test_client_key); // for client verification
  //client.setPrivateKey(test_client_cert); // for client verification

  Serial.println("\nStarting connection to server...");
  UBaseType_t uxHighWaterMark = uxTaskGetStackHighWaterMark( NULL );
  Serial.print( "Memory: " );
  Serial.print(uxHighWaterMark );
  Serial.print( " ");
  Serial.println( esp_get_free_heap_size() );
  delay(2000);
  if (!client.connect(server, 443))
    Serial.println("Connection failed!");
  else {
    Serial.println("Connected to server!");
    // Make a HTTP request:
    client.println("GET https://www.howsmyssl.com/a/check HTTP/1.0");
    client.println("Host: www.howsmyssl.com");
    client.println("Connection: close");
    client.println();

    while (client.connected()) {
      String line = client.readStringUntil('\n');
      if (line == "\r") {
        Serial.println("headers received");
        break;
      }
    }
    // if there are incoming bytes available
    // from the server, read them and print them:
    while (client.available()) {
      char c = client.read();
      Serial.write(c);
    }

    client.stop();
  }
}

void loop() {
  Serial.println(".");
  delay(2000);
}
```

Here is the relevant output:
```
Starting connection to server...
Memory: 1920 163976
***ERROR*** A stack overflow in task loopTask has been detected.
abort() was called at PC 0x40087378 on core 1
```

and the rest is the same as before. Thanks for your help, but it seems I have plenty of heap space left....

lonerzzz, 2017-08-14:
Was checking both stack and heap. You seem to have lots of both. I modified your code as follows and am not getting your stack overflow on an ESP-WROOM-32. I just get the connection, a periodic dump from the site and the connection closed.

Do you have the latest code from this repo?

```
#include "Arduino.h"

#include "WiFi.h"
#include "WiFiMulti.h"
#include "WiFiClientSecure.h"

const char* ssid     = "myssid";
const char* password = "mypasswd";

const char*  server = "www.howsmyssl.com";  // Server URL

/*const char* test_root_ca= "not using this";*/

// You can use x.509 client certificates if you want
//const char* test_client_key = "";   //to verify the client
//const char* test_client_cert = "";  //to verify the client

WiFiClientSecure client;
WiFiMulti wifiMulti;
uint8_t wifiStatus = 0;

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(115200);
  delay(100);

  wifiMulti.addAP(ssid, password);

  Serial.println("\nStarting connection to server...");
  UBaseType_t uxHighWaterMark = uxTaskGetStackHighWaterMark( NULL );
  Serial.print( "Memory: " );
  Serial.print(uxHighWaterMark );
  Serial.print( " ");
  Serial.println( esp_get_free_heap_size() );
}

void loop()
{
  Serial.print("Attempting to connect to SSID: ");
  Serial.println(ssid);

  // Attempt to connect to Wifi network:
  if ((wifiStatus != WL_CONNECTED) && ((wifiStatus = wifiMulti.run()) != WL_CONNECTED))
  {
    Serial.println(wifiStatus);
    // uh oh. better sleep and let the user reset
    delay(10000);
    return;
  }

  Serial.print("Connected to ");
  Serial.println(ssid);

  //client.setCACert(test_root_ca);
  //client.setCertificate(test_client_key); // for client verification
  //client.setPrivateKey(test_client_cert); // for client verification

  Serial.println("\nStarting connection to server...");

  if (!client.connect(server, 443))
  {
    Serial.println("Connection failed!");
  }
  else
  {
    Serial.println("Connected to server!");
    // Make a HTTP request:
    client.println("GET https://www.howsmyssl.com/a/check HTTP/1.0");
    client.println("Host: www.howsmyssl.com");
    client.println("Connection: close");
    client.println();

    while (client.connected())
    {
      String line = client.readStringUntil('\n');
      if (line == "\r")
      {
        Serial.println("headers received");
        break;
      }
    }
    // if there are incoming bytes available
    // from the server, read them and print them:
    while (client.available())
    {
      Serial.write( client.read() );
    }

    client.stop();
  }

  UBaseType_t uxHighWaterMark = uxTaskGetStackHighWaterMark( NULL );
  Serial.print( "Memory: " );
  Serial.print(uxHighWaterMark );
  Serial.print( " ");
  Serial.println( esp_get_free_heap_size() );

  delay( 10000 );
}
```

lonerzzz, 2017-08-25:
@BluCodeGH Any update on this issue?

BluCodeGH, 2017-08-30:
Sorry I have been on vacation with limited internet for the past two weeks. I have the latest code as of three or four weeks ago, but I will pull, flash and get back to you asap.

copercini, 2017-08-30:
@BluCodeGH I saw that your core installation is of the day 9/jul/2017,

(Day 1 of August, the stack of loop task was increased)[https://github.com/espressif/arduino-esp32/commit/4495659ac53bcdc4f8e59d6918b3e472903d63a7] due some problems with WiFiClientSecure...

Maybe just updating your core will solve all the things ;D

BluCodeGH, 2017-08-30:
I certainly hope so!

BluCodeGH, 2017-08-31:
The new code works... kinda. The relevant code:

```cpp
    unsigned long timeout = millis();
    while (client.available() == 0) {
      if (millis() - timeout > 5000) {
        Serial.println("Client Timeout !");
        client.stop();
        return;
      }
    }
    // Once the response is available,
    // check stuff
    while (client.connected()) {
      String line = client.readStringUntil('\n');
      Serial.println(line);
      if (line == "\r") {
        Serial.println("headers received");
        break;
      }
    }

    Serial.println("Here.");

    while (client.available()) {
      // read line until /n
      String line = client.readStringUntil('\n');

      Serial.println(line);
    }
```

When I run this, after connecting to wifi I get the headers, it prints `Here.` and then doesn't print anything from the third while loop. Also it seems that it often has trouble connecting to wifi with WifiMulti, every other time I try to connect I get `[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 1 - UNSPECIFIED` and then it hangs, which seems to be related to #511, but I don't have any interrupts that are running that fast AFAIK.

lonerzzz, 2017-12-18:
@BluCodeGH The reason that that the loop doesn't print anything is because you used the getStream() function instead of the getStreamPtr() function. When you use getStream, you are executing the available() function for the WiFiClient instead of the one for WiFiClientSecure and you get zero as the amount of bytes available.

@copercini What do you think about adding a flag to the httpClient so that it knows which type of transport it is using and generates an error if a person calls getStream() when they should call getStreamPtr(). Alternatively, could we make a getStreamFromSSL() function that returns a reference to WiFiClientSecure instead of the getStreamPtr? Other thoughts?

BluCodeGH, 2017-12-18:
Wow, this brings up some old memories. Where exactly am I calling getStream()? I can't seem to find in the code snippets.

lonerzzz, 2017-12-18:
@BluCodeGH My apologies. Too much stuff going on around my home when I wrote that. I confused the explanation for what I was observing and its not quiet what you are seeing. A possible cause of your issue might be the same mixup of WiFiClientSecure and WiFiClient as they are linked together, but the issue seems to be in underlying code and why I had included copercini in the first place.

BluCodeGH, 2017-12-18:
Ah, that makes more sense. For now is there any other method to use HTTPS? I've been putting of my project, partially because of this, but also cause I'm lazy 😄.

lonerzzz, 2017-12-18:
The httpClient is now working with SSL/TLS and the server root certificate. Just make sure to see the original comment I made that you saw regarding getting the stream so you are not wondering why nothing is coming.

BluCodeGH, 2017-12-18:
Awesome, thank you. Let's hope we can get this issue fixed too, or at the very least make it known to not work in the documentation

copercini, 2017-12-18:
@BluCodeGH testing your code https://github.com/espressif/arduino-esp32/issues/566#issuecomment-322074080, it's working without any problems here:

```
Starting connection to server...
Memory: 6440 151704
Connected to server!
headers received
{"given_cipher_suites":["TLS_ECDHE_ECDSA_WITH_AES_256_GCM_SHA384","TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384","TLS_DHE_RSA_WITH_AES_256_GCM_SHA384","TLS_ECDHE_ECDSA_WITH_AES_256_CCM","TLS_DHE_RSA_WITH_AES_256_CCM","TLS_ECDHE_ECDSA_WITH_AES_256_CBC_SHA384","TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA384","TLS_DHE_RSA_WITH_AES_256_CBC_SHA256","TLS_ECDHE_ECDSA_WITH_AES_256_CBC_SHA","TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA","TLS_DHE_RSA_WITH_AES_256_CBC_SHA","TLS_ECDHE_ECDSA_WITH_AES_256_CCM_8","TLS_DHE_RSA_WITH_AES_256_CCM_8","TLS_ECDHE_ECDSA_WITH_AES_128_GCM_SHA256","TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256","TLS_DHE_RSA_WITH_AES_128_GCM_SHA256","TLS_ECDHE_ECDSA_WITH_AES_128_CCM","TLS_DHE_RSA_WITH_AES_128_CCM","TLS_ECDHE_ECDSA_WITH_AES_128_CBC_SHA256","TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256","TLS_DHE_RSA_WITH_AES_128_CBC_SHA256","TLS_ECDHE_ECDSA_WITH_AES_128_CBC_SHA","TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA","TLS_DHE_RSA_WITH_AES_128_CBC_SHA","TLS_ECDHE_ECDSA_WITH_AES_128_CCM_8","TLS_DHE_RSA_WITH_AES_128_CCM_8","TLS_RSA_WITH_AES_256_GCM_SHA384","TLS_RSA_WITH_AES_256_CCM","TLS_RSA_WITH_AES_256_CBC_SHA256","TLS_RSA_WITH_AES_256_CBC_SHA","TLS_ECDH_RSA_WITH_AES_256_GCM_SHA384","TLS_ECDH_RSA_WITH_AES_256_CBC_SHA384","TLS_ECDH_RSA_WITH_AES_256_CBC_SHA","TLS_ECDH_ECDSA_WITH_AES_256_GCM_SHA384","TLS_ECDH_ECDSA_WITH_AES_256_CBC_SHA384","TLS_ECDH_ECDSA_WITH_AES_256_CBC_SHA","TLS_RSA_WITH_AES_256_CCM_8","TLS_RSA_WITH_AES_128_GCM_SHA256","TLS_RSA_WITH_AES_128_CCM","TLS_RSA_WITH_AES_128_CBC_SHA256","TLS_RSA_WITH_AES_128_CBC_SHA","TLS_ECDH_RSA_WITH_AES_128_GCM_SHA256","TLS_ECDH_RSA_WITH_AES_128_CBC_SHA256","TLS_ECDH_RSA_WITH_AES_128_CBC_SHA","TLS_ECDH_ECDSA_WITH_AES_128_GCM_SHA256","TLS_ECDH_ECDSA_WITH_AES_128_CBC_SHA256","TLS_ECDH_ECDSA_WITH_AES_128_CBC_SHA","TLS_RSA_WITH_AES_128_CCM_8","TLS_EMPTY_RENEGOTIATION_INFO_SCSV"],"ephemeral_keys_supported":true,"session_ticket_supported":true,"tls_compression_supported":false,"unknown_cipher_suite_supported":false,"beast_vuln":false,"able_to_detect_n_minus_one_splitting":false,"insecure_cipher_suites":{},"tls_version":"TLS 1.2","rating":"Probably Okay"}ÿ.
```

copercini, 2017-12-18:
@lonerzzz Your analysis is right, testing with `WiFiClient stream = http.getStream();` will get a empty answer and with `WiFiClient * stream = http.getStreamPtr();` will works.

What do you think of both solutions? generate an error when a person calls getStream() over SSL and also have getStreamFromSSL() for this

lonerzzz, 2017-12-18:
@copercini We can do both to remove as many landmines for users as possible. Was the use of getStreamPtr based on trying to line up with other API implementations or just a work around? If it is not standard, I would suggest removing it in favor of getStream() and getStreamFromSSL().

copercini, 2017-12-18:
@lonerzzz getStreamPtr comes from [ESP8266 implementation](https://github.com/esp8266/Arduino/blob/355232ff6d327a86e1bd0b5a924bbf44ce6b492f/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp#L580), also has an example using it [here](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/examples/StreamHttpClient/StreamHttpClient.ino), so I think it's interesting to keep

lonerzzz, 2017-12-18:
@copercini I will submit an update for review.

lonerzzz, 2017-12-21:
@copercini I added the error handling but not the getStreamFromSSL() function as there is an assignment operator for the secure wifi client that would be affected. I put up the changes and they were merged in. Not sure what might all be using the operator.

closed 2019-08-15 by stale[bot]