espressif/arduino-esp32 #568: TG1WDT_SYS_RESET following BLE call to esp_ble_gatts_start_service

nkolban, 2017-08-08:
I have a BLE library that runs fine in ESP-IDF.  It also compiles cleanly under the Arduino IDE.  However, when I run it as an Arduino application, we fail with the following message shown at reboot:

```
rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:708
load:0x40078000,len:0
load:0x40078000,len:12168
entry 0x40078a68
```

The app restarts, runs for about 20 seconds doing its initialization and then reboots again.
In my debugging, I have pinned it down to a call to a BLE API called "esp_ble_gatts_start_service".

d3cline, 2017-08-08:
Is this the BLE.zip file @copercini posted? I Was able to get the BLE_Server example working without any modification to the core libs, I only had to add something to the loop and pull the github repo for arduino-esp32

nkolban, 2017-08-08:
Howdy @d3cline ... I'll double check with mr @copercini ... him and I have been chatting via Skype for the last couple of days and at the end of yesteday, I believe he was experiencing the same error ... but that may mean we are BOTH using the Makefile recipe used to create the ESP32_BLE.zip and neither of us are using what @copercini created originally.  However, you have given a fantastic clue and we'll take them all.  If the BLE.zip posted to this issue works then we can examine that carefully to distinguish the difference.  All options are open for research/study.

As always, this is a community effort, if others want to jump in, please feel free.  All participants welcome and have equal votes.

d3cline, 2017-08-08:
I can verify BLE_server is working, I am even able to dynamically update a characteristic from a sensor and read it on the phone. Give me a few minutes and I can post my functional code (just have to clean up the I2C stuff I was sending) It's not much but it can be verified.

d3cline, 2017-08-08:
```
#include <BLE.h>
#include <BLEUtils.h>
#include <BLEServer.h>

BLE ble;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

// Defining it here means we can update it later
BLECharacteristic *pCharacteristic;

void setup()
{
  Serial.begin(115200);

  Serial.println("Starting BLE work!");

  ble.initServer("MyESP32");
  BLEServer *pServer = new BLEServer();
  BLEService *pService = pServer->createService(BLEUUID(SERVICE_UUID));
// slightly different then the provided example
  pCharacteristic = pService->createCharacteristic(
                                         BLEUUID(CHARACTERISTIC_UUID),
                                         BLECharacteristic::PROPERTY_READ | BLECharacteristic::PROPERTY_WRITE
                                       );

  pCharacteristic->setValue("QBOrX9k4VFV5Aect01x1LcdFU7tL5fC9OaAQ5TUFcIbbPr35EC5ypjZC0DX9acZ");
  pService->start();
  BLEAdvertising *pAdvertising = pServer->getAdvertising();
  pAdvertising->start();

  
}

void loop()
{

    float Red = 123456;
    String RED = String(Red);
    char str[50];
    RED.toCharArray(str, 50);
    pCharacteristic->setValue(str);
  

  Serial.println();

  //delay(500);
}
```

neilbags, 2017-08-08:
Hi @nkolban, @d3cline,

I just tried to run the above code, linked to the BLE.zip from @copercini, but it crashes at 
` BLEService *pService = pServer->createService(BLEUUID(SERVICE_UUID));`

Can you share the BLE library you are using? Or is a matter of changing the sdkconfig.h? I'm really new to this but would love to jump in if I can!

d3cline, 2017-08-09:
What commit are you guys on? I am on 41e36a72f309fbfa5e2bdeebdca93661728bddb3 Aug 4 14:30. The commit has to be right and I had no build issues.

neilbags, 2017-08-09:
@d3cline i am also up-to-date (same commit). no build issues but the code crashed on my board

d3cline, 2017-08-09:
@neilbags I wonder if reversion plays any role or if we can query any other statistics to compare hardware. I am using the ESP-WROOM-32 on a Wemos lion battery breakout. I have an older blue version on a different PCB I can get my UART cable and test it. Does the BLE_scan build and function?  that example I had no build issues with originally.

d3cline, 2017-08-09:
Flashed the older board, it worked. I am willing to do any other version comparison requested to see if there is a difference in build environments. I am on Ubuntu 64bit, 4.8.0-58-generic. Arduino 1.8.0.

neilbags, 2017-08-09:
@d3cline this is also a ESP-WROOM-32 from this ad:
https://www.aliexpress.com/item/Lolin-ESP32-wemos-ESP32-WiFi-Modules-Bluetooth-Dual-ESP-32-ESP-32S-ESP8266/32807523301.html

I'm on Debian 9 (4.9.0-3-amd64) with Arduino 1.8.1. I have the GATTServer example from the IDF working. I'll try @nkolban's CPP examples as well when I get a chance. I havent tried to debug the arduino BLE library yet, as I don't know C++ or the framework. I'm just starting out - its possible I've done something very stupid

neilbags, 2017-08-09:
@d3cline just to clarify - are you using the BLE.zip from issue #423 ?

d3cline, 2017-08-09:
Yes, https://github.com/espressif/arduino-esp32/files/1203237/BLE.zip

neilbags, 2017-08-09:
BLE_scan builds and runs. And now BLE_notify works and so does your code!

I swear I didn't change anything. Maybe I have a dodgy module/board

nkolban, 2017-08-09:
After an evening of study, we pinned it down to a result which is going to mean that we close out this issue.

While testing the BLE libraries, we had enabled BLE diagnostics/debugging.  Internally, the BLE C++ libraries utilize the ESP_LOGx macros to perform logging.  Within the Arduino environment, these are implicitly disabled.  We had come up a with a recipe that "seemed' to enable them and, in unit testing, it seemed to work.  However, we seem to have hit a problem.   In Arduino land, debugging is commonly performed within user application code with logical such as:

```
Serial.begin(115200);
Serial.println("Here is some text!");
```
What this means is that the Arduino libraries "utilize" the UART/Serial.  It also appears that the ESP_LOGx utilize a slightly different interface to the same serial.  And this is where things start to get grey.

When we run a lot of BLE code (eg. being a BLE server) with debugging on, we appear to have a watchdog exception raised and halt.  If we run the exact same code with the ESP_LOGx switched off (the default in Arduino), then the application runs perfectly with no watchdog exception.

Thus the immediate cure to this particular puzzle is to switch of ESP_LOGx debugging.  While I would like to consider the inability to run ESP_LOGx debugging within the BLE a defect, unfortunately (at this time) ESP_LOGx debugging is not supported in the Arduino environment and hence,  isn't eligible for defect support.   Thus we are closing this issue and claiming (at least for the time being) that one should **not** attempt to switch on ESP_LOGx debugging.

closed 2017-08-09 by nkolban