espressif/arduino-esp32 #569: ESP32 EEPROM Example "EEPROM Fail to initialise" Problem

studuino, 2017-08-09:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			N/A
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
Serial output:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:2016
load:0x40078000,len:7780
ho 0 tail 12 room 4
load:0x40080000,len:252
entry 0x40080034
start...
failed to initialise EEPROM

### Sketch:
```cpp

//Change the code below by your sketch
/*
   EEPROM Write
   Stores random values into the EEPROM.
   These values will stay in the EEPROM when the board is
   turned off and may be retrieved later by another sketch.
*/

#include "EEPROM.h"

// the current address in the EEPROM (i.e. which byte
// we're going to write to next)
int addr = 0;
#define EEPROM_SIZE 64
void setup()
{
  Serial.begin(115200);
  Serial.println("start...");
  if (!EEPROM.begin(EEPROM_SIZE))
  {
    Serial.println("failed to initialise EEPROM"); delay(1000000);
  }
  Serial.println(" bytes read from Flash . Values are:");
  for (int i = 0; i < EEPROM_SIZE; i++)
  {
    Serial.print(byte(EEPROM.read(i))); Serial.print(" ");
  }
  Serial.println();
  Serial.println("writing random n. in memory");
}

void loop()
{
  // need to divide by 4 because analog inputs range from
  // 0 to 1023 and each byte of the EEPROM can only hold a
  // value from 0 to 255.
  // int val = analogRead(10) / 4;
  int val = byte(random(10020));
  // write the value to the appropriate byte of the EEPROM.
  // these values will remain there when the board is
  // turned off.
  EEPROM.write(addr, val);
  Serial.print(val); Serial.print(" ");
  // advance to the next address.  there are 512 bytes in
  // the EEPROM, so go back to 0 when we hit 512.
  // save all changes to the flash.
  addr = addr + 1;
  if (addr == EEPROM_SIZE)
  {
    Serial.println();
    addr = 0;
    EEPROM.commit();
    Serial.print(EEPROM_SIZE);
    Serial.println(" bytes written on Flash . Values are:");
    for (int i = 0; i < EEPROM_SIZE; i++)
    {
      Serial.print(byte(EEPROM.read(i))); Serial.print(" ");
    }
    Serial.println(); Serial.println("----------------------------------");
  }

  delay(100);
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```
esptool.py v2.0-dev
Connecting...
Uploading stub...
Running stub...
Stub running...
Attaching SPI flash...
Configuring flash size...
Compressed 10128 bytes to 6242...

Writing at 0x00001000... (100 %) 
Wrote 10128 bytes (6242 compressed) at 0x00001000 in 0.6 seconds (effective 138.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 105...

Writing at 0x00008000... (100 %) 
Wrote 3072 bytes (105 compressed) at 0x00008000 in 0.0 seconds (effective 768.0 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %) 
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4369.1 kbit/s)...
Hash of data verified.
Compressed 205280 bytes to 70811...

Writing at 0x00010000... (20 %) 
Writing at 0x00014000... (40 %) 
Writing at 0x00018000... (60 %) 
Writing at 0x0001c000... (80 %) 
Writing at 0x00020000... (100 %) 
Wrote 205280 bytes (70811 compressed) at 0x00010000 in 6.3 seconds (effective 260.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

copercini, 2017-08-09:
Are you using the last version of core?
What is in your `\Arduino\hardware\espressif\esp32\tools\partitions\dafault.csv` file?

studuino, 2017-08-12:
Hi copercini,

ESP-WROOM-32 (ets Jun  8 2016 00:22:57) <- any information for esp32s core version?

defaults.csv contents is as below
# Name | Type | SubType | Offset | Size | Flags
-- | -- | -- | -- | -- | --
nvs | data | nvs | 0x9000 | 0x5000 |  
otadata | data | ota | 0xe000 | 0x2000 |  
app0 | app | ota_0 | 0x10000 | 1M |  
app1 | app | ota_1 |   | 1M |  
spiffs | data | spiffs |   | 0x1F0000

Thanks.

lonerzzz, 2017-08-12:
@studuino You will get the latest supported version if you download or pull the latest ESP32 arduino code from this repo and run through the setup instructions. That is the first thing to try.

I am actively using the EEPROM code on an ESP-WROOM-32 without issue, especially without problems from the .begin function. However, I have .begin and .end surrounding my reads and a .begin and .commit around my writes. May be worth a try to try some changes that way.

copercini, 2017-08-12:
So, there is no EEPROM partition in your file... 

I recommend you [update to the latest version of core](https://github.com/espressif/arduino-esp32/blob/master/doc/windows.md#how-to-update-to-the-latest-code) which already [has this partition by default](https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/default.csv)

lonerzzz, 2017-08-15:
@studuino any update on this issue?

studuino, 2017-08-19:
Hi lonerzzz,
Sorry for my late response. I will try out and update.
Thanks for reminder.

studuino, 2017-08-19:
@lonerzzzz and @copercini 
after arduino esp32 core update, i works like a charm.
and I also found preference library for the similar purpose and will weigh pros and cons for my specific needs.
[image]

Thanks alot.

closed 2017-08-19 by studuino