espressif/arduino-esp32 #570: ESP32 suddenly stop appearing in serial port

kosmic21, 2017-08-09:
### Hardware:
Board: ESP32 Dev Module
Core Installation/update date: 03 Aug 2017
IDE name: Arduino IDE
Flash Frequency: 80Mhz
Upload Speed: 115200

### Description:
The first day I got the ESP32 it worked perfectly I made a small led circuit which I controlled with an app on my phone but today out of know where the Arduino IDE Suddenly refuses to see the board in the serial port and I can't upload my sketches anymore

lonerzzz, 2017-08-09:
Here are a few questions that might help:

Are your board settings unchanged from the last time you used it within the board view?
Are you able to connect with another terminal application, TeraTerm for example?
Have you checked the baud rate settings to ensure they are correct?
Are you using USB to Serial adapter? Have you tried plugging it in again?
Have checked the voltage levels on the communications lines to ensure that they are correct?
Can you try to connect with either a volt meter or oscilloscope attached to either the transmit or receive lines and ground to see if the voltage changes are happening on the line? 

Jason

kriswiner, 2017-08-15:
This is likely a driver issue common to Windows and other laptops. Might need to close the IDE, put the ESP32 in boot mode, open the IDE, try to flash a blink program.

If you are using an FTDI chip as USB interface you might have to reinstall or update the FTDI driver, etc.

lonerzzz, 2017-08-21:
@kosmic21 Any update on the problem?

clowrey, 2017-08-29:
Was this the official dev board or some one of the clones? I have two clones that the USB port has died/came broken, I suspect they are using fake CP2102 USB-Serial ICs, and or they have not designed some part of the PCB correctly.

kosmic21, 2017-08-30:
Thanks very much man I’m currently away ill let you know when I get back
Jordan North
java@jordannorth.co.za

lonerzzz, 2017-09-28:
@kosmic21 Any update on this item? Do you have any further information about it?

kosmic21, 2017-09-29:
It looks like when I use a ftdi ti interface with the board it works

me-no-dev, 2017-09-29:
Mac or PC :D

kosmic21, 2017-09-30:
Mac

me-no-dev, 2017-09-30:
this is a kinda "known" issue on some Macs :) (also mac user here). I will advise you to use a nice USB hub to connect your serial devices ;) something in the timing causes the USB controller of the Mac to freak out and disable the port (or certain devices)

kosmic21, 2017-09-30:
Awesome Thanks for all the help dude

closed 2017-10-20 by copercini

tolo69, 2017-11-01:
I face similiar Problem with ESP32DevC with CP2102 under win10. The 64-bit Driver was installed sucessfully but the VCP doesn't appear in device Manager. I have repetetly connected/disconnected the dev-board. The Laptop makes the USB-Sound but it doesn't appear as VCP. Under WIN7 the same ESP32DevC works. Seems to be a WIN10 driver problem?

tolo69, 2018-01-22:
Problem is solved now. The issue was, that USB-port can not provide sufficient supply current.  When supplying the ESP-module with separate 5V, the VCP works stable.

michapixel, 2018-02-27:
I'm experiencing nearly the same.
I have a esp32 ttgo with batterypack and a 0.96" oled. 
attached is a dht22. 

i'm working / flashing with chillipepper (http://chilipeppr.com/esp32). 
I have been nearly done with my little project - templogger - when suddenly i cannot connect to the esp anymore.

I can see following output in my console:
```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:4436
load:0x40078000,len:11080
load:0x40080000,len:252
entry 0x40080034
 [0;32mI (44) boot: ESP-IDF v3.0-dev-20-g9b955f4c 2nd stage bootloader [0m
 [0;32mI (44) boot: compile time 15:47:33 [0m
 [0;32mI (44) boot: Enabling RNG early entropy source... [0m
 [0;32mI (62) boot: SPI Speed      : 40MHz [0m
 [0;32mI (75) boot: SPI Mode       : DIO [0m
 [0;32mI (87) boot: SPI Flash Size : 4MB [0m
 [0;32mI (99) boot: Partition Table: [0m
 [0;32mI (110) boot: ## Label            Usage          Type ST Offset   Length [0m
 [0;32mI (133) boot:  0 nvs              WiFi data        01 02 00009000 00006000 [0m
 [0;32mI (156) boot:  1 phy_init         RF data          01 01 0000f000 00001000 [0m
 [0;32mI (180) boot:  2 factory          factory app      00 00 00010000 00100000 [0m
 [0;32mI (203) boot:  3 spiffs           unknown          c2 00 00110000 002f0000 [0m
 [0;32mI (226) boot: End of partition table [0m
 [0;32mI (239) boot: Disabling RNG early entropy source... [0m
 [0;32mI (256) boot: Loading app partition at offset 00010000 [0m
 [0;32mI (1471) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512)  [0m
 [0;32mI (1471) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x20b94 (134036) map [0m
 [0;32mI (1488) boot: segment 2: paddr=0x00040ba4 vaddr=0x3ffc0000 size=0x0209c (  8348) load [0m
 [0;32mI (1518) boot: segment 3: paddr=0x00042c48 vaddr=0x40080000 size=0x00400 (  1024) load [0m
 [0;32mI (1541) boot: segment 4: paddr=0x00043050 vaddr=0x40080400 size=0x160dc ( 90332) load [0m
 [0;32mI (1610) boot: segment 5: paddr=0x00059134 vaddr=0x400c0000 size=0x00064 (   100) load [0m
 [0;32mI (1611) boot: segment 6: paddr=0x000591a0 vaddr=0x00000000 size=0x06e68 ( 28264)  [0m
 [0;32mI (1628) boot: segment 7: paddr=0x00060010 vaddr=0x400d0018 size=0x81248 (528968) map [0m
 [0;32mI (1654) cpu_start: Pro cpu up. [0m
 [0;32mI (1666) cpu_start: Starting app cpu, entry point is 0x40080fc4 [0m
 [0;32mI (0) cpu_start: App cpu up. [0m
 [0;32mI (1698) heap_alloc_caps: Initializing. RAM available for dynamic allocation: [0m
 [0;32mI (1721) heap_alloc_caps: At 3FFAFF10 len 000000F0 (0 KiB): DRAM [0m
 [0;32mI (1741) heap_alloc_caps: At 3FFC90E8 len 00016F18 (91 KiB): DRAM [0m
 [0;32mI (1762) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM [0m
 [0;32mI (1784) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM [0m
 [0;32mI (1805) heap_alloc_caps: At 400964DC len 00009B24 (38 KiB): IRAM [0m
 [0;32mI (1826) cpu_start: Pro cpu start user code [0m
 [0;32mI (206) cpu_start: Starting scheduler on APP CPU. [0m

Mounting flash filesystem...
 [0;32mI (386) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE [0m
```
and then it 'disconnects', the COM-port stays open as i still can see itz in the device manager, but windows makes a device-disconnect-sound. if i turn the device off and then on it connects (sound) thenb 'disconnects' (again sound)

i can erase and reflash, but as soon as the esp is turned on, i have the output interrupt like above. 
My wild guess is: the onboard flash is corrupted. What do you guys think?

michapixel, 2018-03-01:
Ah, tried another computer, so basically another powersource, and boom, everything works as expected, so yeah, it very clearly to be a power issue.