espressif/arduino-esp32 #572: esp_deep_sleep timer wakeup not accurate for longer durations?

ConnorMullaly, 2017-08-09:
### Hardware:
Board:							esp32 dev module
Core Installation/update date:			8/9/2017
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						256000

### Description:
I am looking to put the esp32 into deep sleep for 6 hours at a time.  What I found when deep sleeping for 21600000000 microseconds was that it would wake up every 2 minutes about.  It was accurate when sleeping for shorter durations, but could never sleep for longer durations.  I am using the example code for deep sleep timer wakeup but just changing the time to sleep to 21600

### Sketch:
```cpp

//Change the code below by your sketch
/*
Simple Deep Sleep with Timer Wake Up
=====================================
ESP32 offers a deep sleep mode for effective power
saving as power is an important factor for IoT
applications. In this mode CPUs, most of the RAM,
and all the digital peripherals which are clocked
from APB_CLK are powered off. The only parts of
the chip which can still be powered on are:
RTC controller, RTC peripherals ,and RTC memories
This code displays the most basic deep sleep with
a timer to wake it up and how to store data in
RTC memory to use it over reboots
This code is under Public Domain License.
Author:
Pranav Cherukupalli <cherukupallip@gmail.com>
*/

#define uS_TO_S_FACTOR 1000000  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  21600        /* Time ESP32 will go to sleep (in seconds) */

RTC_DATA_ATTR int bootCount = 0;

/*
Method to print the reason by which ESP32
has been awaken from sleep
*/
void print_wakeup_reason(){
  esp_deep_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_deep_sleep_get_wakeup_cause();

  switch(wakeup_reason)
  {
    case 1  : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case 2  : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case 3  : Serial.println("Wakeup caused by timer"); break;
    case 4  : Serial.println("Wakeup caused by touchpad"); break;
    case 5  : Serial.println("Wakeup caused by ULP program"); break;
    default : Serial.println("Wakeup was not caused by deep sleep"); break;
  }
}

void setup(){
  Serial.begin(115200);
  delay(1000); //Take some time to open up the Serial Monitor

  //Increment boot number and print it every reboot
  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));

  //Print the wakeup reason for ESP32
  print_wakeup_reason();

  /*
  First we configure the wake up source
  We set our ESP32 to wake up every 5 seconds
  */
  esp_deep_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  Serial.println("Setup ESP32 to sleep for every " + String(TIME_TO_SLEEP) +
  " Seconds");

  /*
  Next we decide what all peripherals to shut down/keep on
  By default, ESP32 will automatically power down the peripherals
  not needed by the wakeup source, but if you want to be a poweruser
  this is for you. Read in detail at the API docs
  http://esp-idf.readthedocs.io/en/latest/api-reference/system/deep_sleep.html
  Left the line commented as an example of how to configure peripherals.
  The line below turns off all RTC peripherals in deep sleep.
  */
  //esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
  //Serial.println("Configured all RTC Peripherals to be powered down in sleep");

  /*
  Now that we have setup a wake cause and if needed setup the
  peripherals state in deep sleep, we can now start going to
  deep sleep.
  In the case that no wake up sources were provided but deep
  sleep was started, it will sleep forever unless hardware
  reset occurs.
  */
  Serial.println("Going to sleep now");
  esp_deep_sleep_start();
  Serial.println("This will never be printed");
}

void loop(){
  //This is not going to be called
}

lonerzzz, 2017-08-09:
This may be better raised in the IDF forum as the Arduino has no control over this accuracy to my understanding. The IDF code may allow for tweaking the clock division to better improve accuracy.

eyaleb, 2017-08-10:
Your constants are 32 bits (no LL suffix), so
21600000000 = 0x50775d800
truncating to 32 bits 0x775d800 = 125163520us ~ 125 seconds

ConnorMullaly, 2017-08-10:
This seems to be the culprit.  I will test the code with the change.  In my other software I was setting the sleep time like so: uint64_t sleep_time_us = 21600 * 1000000;  This could have also been truncating to 32 bits and then storing into a 64 bit.  I change this by first storing both numbers as 64 bit and then multiplying.

lonerzzz, 2017-08-12:
@ConnorMullaly Was the issue due to the truncation? Can we close the issue?

ConnorMullaly, 2017-08-14:
@lonerzzz yes this issue was due to truncation.  Thank you

lonerzzz, 2017-08-14:
Closing based on feedback that the issue was resolved.

closed 2017-08-14 by lonerzzz

jayrawatrj, 2021-07-26:
I have exactly the same issue. esp goes to sleep and never woke up.. anyone has a solution for this?

jayrawatrj, 2021-07-26:
#define uS_TO_S_FACTOR 1000000ULL 
I have the updated example code..