espressif/arduino-esp32 #575: howsmyssl reports Bad due to insecure cipher suites

bbx10, 2017-08-11:
### Hardware:
Board: Adafruit Huzzah32
Core Installation/update date: 10/Aug/2017
IDE name: Arduino IDE 1.8.2
Flash Frequency: 80Mhz
Upload Speed: 921600

### Description:
https://howsmyssl.com reports overall result of Bad due to insecure cipher suites.

```
  "insecure_cipher_suites": {
    "TLS_RSA_WITH_3DES_EDE_CBC_SHA": [
      "uses 3DES which is vulnerable to the Sweet32 attack but was not configured as a fallback in the ciphersuite order"
    ],
    "TLS_RSA_PSK_WITH_3DES_EDE_CBC_SHA": [
      "uses 3DES which is vulnerable to the Sweet32 attack but was not configured as a fallback in the ciphersuite order"
    ],
    "TLS_DHE_PSK_WITH_3DES_EDE_CBC_SHA": [
      "uses 3DES which is vulnerable to the Sweet32 attack but was not configured as a fallback in the ciphersuite order"
    ],
    "TLS_DHE_RSA_WITH_3DES_EDE_CBC_SHA": [
      "uses 3DES which is vulnerable to the Sweet32 attack but was not configured as a fallback in the ciphersuite order"
    ],
    "TLS_ECDHE_ECDSA_WITH_3DES_EDE_CBC_SHA": [
      "uses 3DES which is vulnerable to the Sweet32 attack but was not configured as a fallback in the ciphersuite order"
    ],
    "TLS_ECDHE_PSK_WITH_3DES_EDE_CBC_SHA": [
      "uses 3DES which is vulnerable to the Sweet32 attack but was not configured as a fallback in the ciphersuite order"
    ],
    "TLS_ECDHE_RSA_WITH_3DES_EDE_CBC_SHA": [
      "uses 3DES which is vulnerable to the Sweet32 attack but was not configured as a fallback in the ciphersuite order"
    ],
    "TLS_ECDH_ECDSA_WITH_3DES_EDE_CBC_SHA": [
      "uses 3DES which is vulnerable to the Sweet32 attack but was not configured as a fallback in the ciphersuite order"
    ],
    "TLS_ECDH_RSA_WITH_3DES_EDE_CBC_SHA": [
      "uses 3DES which is vulnerable to the Sweet32 attack but was not configured as a fallback in the ciphersuite order"
    ],
    "TLS_PSK_WITH_3DES_EDE_CBC_SHA": [
      "uses 3DES which is vulnerable to the Sweet32 attack but was not configured as a fallback in the ciphersuite order"
    ]
  }
```

### Sketch:
```cpp
/**
 * BasicHTTPClient.ino
 *
 *  Created on: 24.05.2015
 *
 */

#include <Arduino.h>

#include <WiFi.h>
#include <WiFiMulti.h>

#include <HTTPClient.h>
#include <ArduinoJson.h>

#define USE_SERIAL Serial

WiFiMulti WiFiMulti;

const char* root_ca_cert =
"-----BEGIN CERTIFICATE-----\n"
"MIIEkjCCA3qgAwIBAgIQCgFBQgAAAVOFc2oLheynCDANBgkqhkiG9w0BAQsFADA/\n"
"MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT\n"
[... 267 lines cut ...]
    ],
    "TLS_DHE_RSA_WITH_3DES_EDE_CBC_SHA": [
      "uses 3DES which is vulnerable to the Sweet32 attack but was not configured as a fallback in the ciphersuite order"
    ],
    "TLS_ECDHE_ECDSA_WITH_3DES_EDE_CBC_SHA": [
      "uses 3DES which is vulnerable to the Sweet32 attack but was not configured as a fallback in the ciphersuite order"
    ],
    "TLS_ECDHE_PSK_WITH_3DES_EDE_CBC_SHA": [
      "uses 3DES which is vulnerable to the Sweet32 attack but was not configured as a fallback in the ciphersuite order"
    ],
    "TLS_ECDHE_RSA_WITH_3DES_EDE_CBC_SHA": [
      "uses 3DES which is vulnerable to the Sweet32 attack but was not configured as a fallback in the ciphersuite order"
    ],
    "TLS_ECDH_ECDSA_WITH_3DES_EDE_CBC_SHA": [
      "uses 3DES which is vulnerable to the Sweet32 attack but was not configured as a fallback in the ciphersuite order"
    ],
    "TLS_ECDH_RSA_WITH_3DES_EDE_CBC_SHA": [
      "uses 3DES which is vulnerable to the Sweet32 attack but was not configured as a fallback in the ciphersuite order"
    ],
    "TLS_PSK_WITH_3DES_EDE_CBC_SHA": [
      "uses 3DES which is vulnerable to the Sweet32 attack but was not configured as a fallback in the ciphersuite order"
    ]
  }
}
```

skandragon, 2017-08-15:
Would it be as simple as editing `components/mbedtls/port/include/mbedtls/esp_config.h` and commenting out the `#define MBEDTLS_DES_C` around line 1701?

me-no-dev, 2017-08-17:
I feel this should go to espressif/esp-idf instead.

projectgus, 2017-08-17:
We have a fix for this coming in IDF (disabling 3DES by default). I've added a link to this issue to the commit so there will be a notification when it lands in IDF master branch.

For the record, the Sweet32 attack which leads to the "Bad" result requires the attacker to capture around 785GB of traffic from a single client session. Which is probably a challenge for most ESP32-based devices! But that's no reason we shouldn't disable it by default (it also saves some code size, which is nice.)

bbx10, 2017-09-13:
Since ba929be27a88b573ff632c5878f984a5eb1e073f howsmyssl reports "Probably Okay" which is the highest rating. The latest versions of Firefox and Chrome receive the same rating.

```
{
  "rating": "Probably Okay",
  "tls_version": "TLS 1.2",
  "given_cipher_suites": [
    "TLS_ECDHE_ECDSA_WITH_AES_256_GCM_SHA384",
    "TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384",
    "TLS_DHE_RSA_WITH_AES_256_GCM_SHA384",
    "TLS_ECDHE_ECDSA_WITH_AES_256_CCM",
    "TLS_DHE_RSA_WITH_AES_256_CCM",
    "TLS_ECDHE_ECDSA_WITH_AES_256_CBC_SHA384",
    "TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA384",
    "TLS_DHE_RSA_WITH_AES_256_CBC_SHA256",
    "TLS_ECDHE_ECDSA_WITH_AES_256_CBC_SHA",
    "TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA",
    "TLS_DHE_RSA_WITH_AES_256_CBC_SHA",
    "TLS_ECDHE_ECDSA_WITH_AES_256_CCM_8",
    "TLS_DHE_RSA_WITH_AES_256_CCM_8",
    "TLS_ECDHE_ECDSA_WITH_AES_128_GCM_SHA256",
    "TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256",
    "TLS_DHE_RSA_WITH_AES_128_GCM_SHA256",
    "TLS_ECDHE_ECDSA_WITH_AES_128_CCM",
    "TLS_DHE_RSA_WITH_AES_128_CCM",
    "TLS_ECDHE_ECDSA_WITH_AES_128_CBC_SHA256",
    "TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256",
    "TLS_DHE_RSA_WITH_AES_128_CBC_SHA256",
    "TLS_ECDHE_ECDSA_WITH_AES_128_CBC_SHA",
    "TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA",
    "TLS_DHE_RSA_WITH_AES_128_CBC_SHA",
    "TLS_ECDHE_ECDSA_WITH_AES_128_CCM_8",
    "TLS_DHE_RSA_WITH_AES_128_CCM_8",
    "TLS_RSA_WITH_AES_256_GCM_SHA384",
    "TLS_RSA_WITH_AES_256_CCM",
    "TLS_RSA_WITH_AES_256_CBC_SHA256",
    "TLS_RSA_WITH_AES_256_CBC_SHA",
    "TLS_ECDH_RSA_WITH_AES_256_GCM_SHA384",
    "TLS_ECDH_RSA_WITH_AES_256_CBC_SHA384",
    "TLS_ECDH_RSA_WITH_AES_256_CBC_SHA",
    "TLS_ECDH_ECDSA_WITH_AES_256_GCM_SHA384",
    "TLS_ECDH_ECDSA_WITH_AES_256_CBC_SHA384",
    "TLS_ECDH_ECDSA_WITH_AES_256_CBC_SHA",
    "TLS_RSA_WITH_AES_256_CCM_8",
    "TLS_RSA_WITH_AES_128_GCM_SHA256",
    "TLS_RSA_WITH_AES_128_CCM",
    "TLS_RSA_WITH_AES_128_CBC_SHA256",
    "TLS_RSA_WITH_AES_128_CBC_SHA",
    "TLS_ECDH_RSA_WITH_AES_128_GCM_SHA256",
    "TLS_ECDH_RSA_WITH_AES_128_CBC_SHA256",
    "TLS_ECDH_RSA_WITH_AES_128_CBC_SHA",
    "TLS_ECDH_ECDSA_WITH_AES_128_GCM_SHA256",
    "TLS_ECDH_ECDSA_WITH_AES_128_CBC_SHA256",
    "TLS_ECDH_ECDSA_WITH_AES_128_CBC_SHA",
    "TLS_RSA_WITH_AES_128_CCM_8",
    "TLS_EMPTY_RENEGOTIATION_INFO_SCSV"
  ],
  "ephemeral_keys_supported": true,
  "session_ticket_supported": true,
  "tls_compression_supported": false,
  "unknown_cipher_suite_supported": false,
  "beast_vuln": false,
  "able_to_detect_n_minus_one_splitting": false,
  "insecure_cipher_suites": {}
}

```

closed 2017-09-13 by bbx10