espressif/arduino-esp32 #576: Missing Pin in

Thro42, 2017-08-11:
### Hardware:
Board:							Noduino Quantum
Core Installation/update date:			11/jul/2017
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
 I miss the definition of Port A1 and A2 in arduino-esp32/variants/quantum/pins_arduino.h

lonerzzz, 2017-08-12:
@Thro42 Can you attach an example of what you are expecting to see in that file that is missing?

Thro42, 2017-08-18:
As you can see in arduino-esp32/variants/quantum/pins_arduino.h, the definition of Port A1 and A2 is missing.
....
static const uint8_t A0 = 36;
static const uint8_t A3 = 39;
static const uint8_t A4 = 32;
static const uint8_t A5 = 33;
.....
As I known the full Mapping must be:
static const uint8_t A0 = 36;
**static const uint8_t A1 = 37;
static const uint8_t A2 = 38;**
static const uint8_t A3 = 39;
static const uint8_t A4 = 32;
static const uint8_t A5 = 33;

copercini, 2017-08-19:
But there are no GPIO37 and 38 in Noduino quantum....
[image]

lonerzzz, 2017-08-21:
@Thro42 Did copercini answer your question or is additional information available to better describe the situation?

copercini, 2017-09-08:
Since Noduino Quantum don't expose these pins and even if exposed they do not seem to work as ADC when connected to the 270pF sampling capacitors as described in https://espressif.com/sites/default/files/documentation/esp32_hardware_design_guidelines_en.pdf session 3.1.6

closed 2017-09-08 by copercini