espressif/arduino-esp32 #577: rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2)) waiting for download

studuino, 2017-08-12:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			NA
IDE name:							Arduino IDE 1.8.1
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
Sketch Uploading is successful.
But when I check in serial monitor (press reset button) the message says

**ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download**

```

### Debug Messages:
esptool.py v2.0-dev
Connecting...
Uploading stub...
Running stub...
Stub running...
Attaching SPI flash...
Configuring flash size...
Compressed 10128 bytes to 6242...

Writing at 0x00001000... (100 %) 
Wrote 10128 bytes (6242 compressed) at 0x00001000 in 0.6 seconds (effective 146.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 105...

Writing at 0x00008000... (100 %) 
Wrote 3072 bytes (105 compressed) at 0x00008000 in 0.0 seconds (effective 1536.0 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %) 
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4096.0 kbit/s)...
Hash of data verified.
Compressed 530128 bytes to 279810...

Writing at 0x00010000... (5 %) 
Writing at 0x00014000... (11 %) 
Writing at 0x00018000... (16 %) 
Writing at 0x0001c000... (22 %) 
Writing at 0x00020000... (27 %) 
Writing at 0x00024000... (33 %) 
Writing at 0x00028000... (38 %) 
Writing at 0x0002c000... (44 %) 
Writing at 0x00030000... (50 %) 
Writing at 0x00034000... (55 %) 
Writing at 0x00038000... (61 %) 
Writing at 0x0003c000... (66 %) 
Writing at 0x00040000... (72 %) 
Writing at 0x00044000... (77 %) 
Writing at 0x00048000... (83 %) 
Writing at 0x0004c000... (88 %) 
Writing at 0x00050000... (94 %) 
Writing at 0x00054000... (100 %) 
Wrote 530128 bytes (279810 compressed) at 0x00010000 in 24.9 seconds (effective 170.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

lonerzzz, 2017-08-12:
@studuino What you are seeing is the output of a successful write to the ESP32. What issue do you believe that you are hitting? The rst.. line indicates that the device is ready to receive bytes.

studuino, 2017-08-12:
Hi lonerzzz,

I am trying to run WiFi sketch. It uploads successfully but when I open serial monitor after upload, the messages said "ets Jun 8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download" 

This is self assembled board (using ESP-WROOM-32 module).
Previously it works just fine. Today I got this problem.
I have to do manual reset(EN) and IO0 to gnd before sketch upload.

everslick, 2017-08-12:
i'm not sure if it is really required, but i have a 10k pullup on IO0 to make sure it is not floating.

lonerzzz, 2017-08-12:
As @everslick is alluding to to, it would seem your IO00 is being held low or floating at the wrong level which is causing you to stay in bootloader mode. You need to make sure that IO00 stays high when your toggle the enable pin to reset.

studuino, 2017-08-13:
@everslick  I followed the wiring of components from https://pcbreflux.blogspot.com/2016/11/esp-32-getting-to-blinky-to-get-to.html . 

@lonerzzz I will check IO0 level after sketch upload.

Thanks.

studuino, 2017-08-13:
I read IO0 at 550mV.
I noticed that sometimes I do not even need to manually do press sequence of EN(Reset) and IO0 for uploading sketches. I also checked the EN voltage which is stable at 3V.

lonerzzz, 2017-08-13:
@studuino That voltage is definitely too low. You need to figure out why IO0 is low because it is essentially holding you in bootloader mode. If you have other circuitry on that pin, it may be better moved to another pin to free up IO0 for bootloading purposes. If this is not possible, you need to ensure that on startup IO0 is kept high. If you are otherwise not using that pin but getting that voltage, a 10K pullup will help the situation.

studuino, 2017-08-13:
Thank you very much @lonerzzz .
I found out my poor re-soldering that shorted usb/serial(ft232rl)'s txd and gnd pin has caused the problem. Now i read IO0 at 3V and manual press combination work just perfect as before.

closed 2017-08-13 by studuino

SERIDJ, 2020-03-30:
hi,

i have the same problem, when i changed in menu config Offset of Partition Table to 0x10000

[image]

atanisoft, 2020-03-30:
@SERIDJ Don't hijack closed issues with unrelated issues. The default address that the bootloader jumps to for the app is 0x10000. You can't change the offset of the partition table to this address.

SERIDJ, 2020-03-31:
@atanisoft 
what i know if we enable flash encryption  bootloader app binary  may exceed the default bootloader size limit. And (partition table offset 0x8000 - bootloader offset 0x1000). so that why we need to set partition table offset to a higher value than 0x8000. to place the partition table later in the flash and this increases the space available for the bootloader ;) ;)

atanisoft, 2020-03-31:
@SERIDJ open a new issue if you want to continue discussions.

SERIDJ, 2020-03-31:
@atanisoft  yes yes  thanks

rshah18, 2020-06-02:
i had this problem because my ground and io0 were still connected after loading the program. unplug the ground from the pin 0 to make it work

EdCaneda, 2020-07-12:
Thank you for having this discussion. I had this issue too. 
After reading this thread, I remove the jumper wire between GPIO0 to GND and reset the ESP32-CAM again.
There the issue was resolved.

LucaH360, 2023-01-14:
j'essaye de faire marcher une ESP32 caméra webserveur mais dans le moniteur sa meut mais ça : 
17:29:34.313 -> ets Jul 29 2019 12:21:46
17:29:34.313 -> 
17:29:34.313 -> rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
17:29:34.313 -> waiting for download

Wilddieb, 2024-03-14:
Same Issue, what I have done to solve it was:

1. 10K Pullup to IO0 
2. Firmware install
3. Remove 10K Pullup
4. Unplug and replug power 

problem solved for me.

timkatta5, 2024-09-26:
I am having the same issue with my heltec Lora 32 after having installed meshtastic on it.

I get a similar instalation log from the first post. But after I reset or turn it on again, screen is off and serial monitor gives me this message:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x1 (DOWNLOAD_BOOT(UART0/UART1/SDIO_FEI_REO_V2))

waiting for download

ets Jun 8 2016 00:22:57

The pins are not soldered to the chip and I am accessing it from USB.

Darren777, 2025-05-22:
I found 2 solutions to the problem which seems to be cause by a small amount of code being regarded as no code by the esp32 so it just stays in boot mode thinking that it has no code to run.

Firstly, holding the boot button on the ESP32 down prior to connecting the USB cable, and then only releasing it once the sketch starts uploading. this solved the problem.

Secondly, this solution only works in Arduino IDE, before uploading the sketch go to the upload options and choose the "large app" option