espressif/arduino-esp32 #580: HTTPClient POST getting stuck after few requests when used with deep sleep

chris-gunawardena, 2017-08-14:
### Hardware:
Board:							Sparkfun ESP32 thing
Core Installation/update date:		Latest stable
IDE name:						Platform.io
Flash Frequency:					40Mhz
Upload Speed:					115200

### Description:
I have a simple sketch that resumes from sleep every 60 seconds and performs a https POST request, but it seems to get stuck after 5 to 10 requests. I have tried with `setReuse` true/false and with ` HTTPClient http;` in `setup()` and inside the `loop()`

I tried the same request without deep sleep and it's working fine. (sketch below)

### Sketch that fails after 5-10 tries:
```cpp
#include <Arduino.h>
#include <WiFi.h>
#include <WiFiClientSecure.h>
#include <HTTPClient.h>

const char* firebase_ca =     "-----BEGIN CERTIFICATE-----\n" \
    "MIIEKDCCAxCgAwIBAgIQAQAhJYiw+lmnd+8Fe2Yn3zANBgkqhkiG9w0BAQsFADBC\n" \
    "MQswCQYDVQQGEwJVUzEWMBQGA1UEChMNR2VvVHJ1c3QgSW5jLjEbMBkGA1UEAxMS\n" \
    "R2VvVHJ1c3QgR2xvYmFsIENBMB4XDTE3MDUyMjExMzIzN1oXDTE4MTIzMTIzNTk1\n" \
    "OVowSTELMAkGA1UEBhMCVVMxEzARBgNVBAoTCkdvb2dsZSBJbmMxJTAjBgNVBAMT\n" \
    "HEdvb2dsZSBJbnRlcm5ldCBBdXRob3JpdHkgRzIwggEiMA0GCSqGSIb3DQEBAQUA\n" \
    "A4IBDwAwggEKAoIBAQCcKgR3XNhQkToGo4Lg2FBIvIk/8RlwGohGfuCPxfGJziHu\n" \
    "Wv5hDbcyRImgdAtTT1WkzoJile7rWV/G4QWAEsRelD+8W0g49FP3JOb7kekVxM/0\n" \
    "Uw30SvyfVN59vqBrb4fA0FAfKDADQNoIc1Fsf/86PKc3Bo69SxEE630k3ub5/DFx\n" \
    "+5TVYPMuSq9C0svqxGoassxT3RVLix/IGWEfzZ2oPmMrhDVpZYTIGcVGIvhTlb7j\n" \
    "gEoQxirsupcgEcc5mRAEoPBhepUljE5SdeK27QjKFPzOImqzTs9GA5eXA37Asd57\n" \
    "r0Uzz7o+cbfe9CUlwg01iZ2d+w4ReYkeN8WvjnJpAgMBAAGjggERMIIBDTAfBgNV\n" \
    "HSMEGDAWgBTAephojYn7qwVkDBF9qn1luMrMTjAdBgNVHQ4EFgQUSt0GFhu89mi1\n" \
    "dvWBtrtiGrpagS8wDgYDVR0PAQH/BAQDAgEGMC4GCCsGAQUFBwEBBCIwIDAeBggr\n" \
    "BgEFBQcwAYYSaHR0cDovL2cuc3ltY2QuY29tMBIGA1UdEwEB/wQIMAYBAf8CAQAw\n" \
    "NQYDVR0fBC4wLDAqoCigJoYkaHR0cDovL2cuc3ltY2IuY29tL2NybHMvZ3RnbG9i\n" \
    "YWwuY3JsMCEGA1UdIAQaMBgwDAYKKwYBBAHWeQIFATAIBgZngQwBAgIwHQYDVR0l\n" \
    "BBYwFAYIKwYBBQUHAwEGCCsGAQUFBwMCMA0GCSqGSIb3DQEBCwUAA4IBAQDKSeWs\n" \
    "12Rkd1u+cfrP9B4jx5ppY1Rf60zWGSgjZGaOHMeHgGRfBIsmr5jfCnC8vBk97nsz\n" \
    "qX+99AXUcLsFJnnqmseYuQcZZTTMPOk/xQH6bwx+23pwXEz+LQDwyr4tjrSogPsB\n" \
    "E4jLnD/lu3fKOmc2887VJwJyQ6C9bgLxRwVxPgFZ6RGeGvOED4Cmong1L7bHon8X\n" \
    "fOGLVq7uZ4hRJzBgpWJSwzfVO+qFKgE4h6LPcK2kesnE58rF2rwjMvL+GMJ74N87\n" \
    "L9TQEOaWTPtEtyFkDbkAlDASJodYmDkFOA/MgkgMCkdm7r+0X8T/cKjhf4t5K7hl\n" \
    "MqO5tzHpCvX2HzLc\n" \
    "-----END CERTIFICATE-----\n";;

const char* ssid = "xxx";
const char* password = "yyy";
RTC_DATA_ATTR int bootCount = 1;

void setup(){
  Serial.begin(115200);
  delay(5000);
  Serial.println("Boot number: " + String(bootCount++));

  // WiFi
  WiFi.begin(ssid, password); 
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to WiFi..");
  }
  Serial.println("Connected to the WiFi network");

  // POST  
  HTTPClient http;
  http.setReuse(true);
  http.begin("https://water-9dbfa.firebaseio.com/users/chris/readings.json", firebase_ca);
  http.addHeader("Content-Type", "application/json");
  int httpResponseCode = http.POST("{ \"level\": " + String(bootCount) + ", \"timestamp\": {\".sv\": \"timestamp\"} }");   
  Serial.println("httpResponseCode: " + http.errorToString(httpResponseCode));
  http.end();
[... 166 lines cut ...]

  // WiFi
  WiFi.begin(ssid, password); 
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to WiFi..");
  }
  Serial.println("Connected to the WiFi network");

}

void loop(){
  Serial.println("Boot number: " + String(bootCount++));

  // POST  
  HTTPClient http;
  http.setReuse(true);
  http.begin("https://water-9dbfa.firebaseio.com/users/chris/readings.json", firebase_ca);
  http.addHeader("Content-Type", "application/json");
  int httpResponseCode = http.POST("{ \"level\": " + String(bootCount) + ", \"timestamp\": {\".sv\": \"timestamp\"} }");   
  Serial.println("httpResponseCode: " + http.errorToString(httpResponseCode));
  http.end();
  delay(5000);
}
```

chris-gunawardena, 2017-08-14:
Maybe this has something to do with the connected sensor and the voltage divider, closing this until I confirm this.

chris-gunawardena, 2017-08-14:
Adding these lines to my deep sleep config seems to fix this issue:
```cpp
  esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_SLOW_MEM, ESP_PD_OPTION_OFF);
  esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_FAST_MEM, ESP_PD_OPTION_OFF);
  esp_deep_sleep_pd_config(ESP_PD_DOMAIN_MAX, ESP_PD_OPTION_OFF);
```

chris-gunawardena, 2017-08-14:
Spoke too soon, the above just made about 15 requests instead of the usual 5 before getting stuck.

lonerzzz, 2017-08-14:
Does the same program work without the HTTP Post call? Can you have it simply wake up and print something out? That way we can see if the HTTP Post is part of the equation at all.

chris-gunawardena, 2017-08-14:
Thanks, just checked and it works fine for hours without the http stuff, also seems to work without SSL doing a http post, but I'll test more tomorrow. esp_get_free_heap_size() is reporting the same size after every wake, so no memory leaks or anything What I did notice was that something goes wrong in the call before it gets stuck. eg: 
```
[E][ssl_client.cpp:25] handle_error(): X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[E][ssl_client.cpp:27] handle_error(): MbedTLS message code: -9984
[E][WiFiClientSecure.cpp:102] connect(): lwip_connect_r: 11
httpResponseCode: connection refused
```

lonerzzz, 2017-08-14:
You may also want to check the stack high water mark where you are checking the memory.

Also, try to handle the error and successfully complete the POST before allowing the sleep to occur. The issue might be due to some corruption from SSL failure but you don't see it before sleeping again. This would reduce the issue to being HTTP over SSL related.

lonerzzz, 2017-08-15:
I ran a modified version of the code for a number of hours with debug level set to warn. What I saw was the following:

Connecting to WiFi..[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 4 - ASSOC_EXPIRE

What I never saw was the error that you are seeing. Here is the code

```
#include <Arduino.h>
#include <HTTPClient.h>
#include <WiFi.h>
#include <WiFiClientSecure.h>

const char* firebase_ca =     "-----BEGIN CERTIFICATE-----\n" \
    "MIIEKDCCAxCgAwIBAgIQAQAhJYiw+lmnd+8Fe2Yn3zANBgkqhkiG9w0BAQsFADBC\n" \
    "MQswCQYDVQQGEwJVUzEWMBQGA1UEChMNR2VvVHJ1c3QgSW5jLjEbMBkGA1UEAxMS\n" \
    "R2VvVHJ1c3QgR2xvYmFsIENBMB4XDTE3MDUyMjExMzIzN1oXDTE4MTIzMTIzNTk1\n" \
    "OVowSTELMAkGA1UEBhMCVVMxEzARBgNVBAoTCkdvb2dsZSBJbmMxJTAjBgNVBAMT\n" \
    "HEdvb2dsZSBJbnRlcm5ldCBBdXRob3JpdHkgRzIwggEiMA0GCSqGSIb3DQEBAQUA\n" \
    "A4IBDwAwggEKAoIBAQCcKgR3XNhQkToGo4Lg2FBIvIk/8RlwGohGfuCPxfGJziHu\n" \
    "Wv5hDbcyRImgdAtTT1WkzoJile7rWV/G4QWAEsRelD+8W0g49FP3JOb7kekVxM/0\n" \
    "Uw30SvyfVN59vqBrb4fA0FAfKDADQNoIc1Fsf/86PKc3Bo69SxEE630k3ub5/DFx\n" \
    "+5TVYPMuSq9C0svqxGoassxT3RVLix/IGWEfzZ2oPmMrhDVpZYTIGcVGIvhTlb7j\n" \
    "gEoQxirsupcgEcc5mRAEoPBhepUljE5SdeK27QjKFPzOImqzTs9GA5eXA37Asd57\n" \
    "r0Uzz7o+cbfe9CUlwg01iZ2d+w4ReYkeN8WvjnJpAgMBAAGjggERMIIBDTAfBgNV\n" \
    "HSMEGDAWgBTAephojYn7qwVkDBF9qn1luMrMTjAdBgNVHQ4EFgQUSt0GFhu89mi1\n" \
    "dvWBtrtiGrpagS8wDgYDVR0PAQH/BAQDAgEGMC4GCCsGAQUFBwEBBCIwIDAeBggr\n" \
    "BgEFBQcwAYYSaHR0cDovL2cuc3ltY2QuY29tMBIGA1UdEwEB/wQIMAYBAf8CAQAw\n" \
    "NQYDVR0fBC4wLDAqoCigJoYkaHR0cDovL2cuc3ltY2IuY29tL2NybHMvZ3RnbG9i\n" \
    "YWwuY3JsMCEGA1UdIAQaMBgwDAYKKwYBBAHWeQIFATAIBgZngQwBAgIwHQYDVR0l\n" \
    "BBYwFAYIKwYBBQUHAwEGCCsGAQUFBwMCMA0GCSqGSIb3DQEBCwUAA4IBAQDKSeWs\n" \
    "12Rkd1u+cfrP9B4jx5ppY1Rf60zWGSgjZGaOHMeHgGRfBIsmr5jfCnC8vBk97nsz\n" \
    "qX+99AXUcLsFJnnqmseYuQcZZTTMPOk/xQH6bwx+23pwXEz+LQDwyr4tjrSogPsB\n" \
    "E4jLnD/lu3fKOmc2887VJwJyQ6C9bgLxRwVxPgFZ6RGeGvOED4Cmong1L7bHon8X\n" \
    "fOGLVq7uZ4hRJzBgpWJSwzfVO+qFKgE4h6LPcK2kesnE58rF2rwjMvL+GMJ74N87\n" \
    "L9TQEOaWTPtEtyFkDbkAlDASJodYmDkFOA/MgkgMCkdm7r+0X8T/cKjhf4t5K7hl\n" \
    "MqO5tzHpCvX2HzLc\n" \
    "-----END CERTIFICATE-----\n";;

const char* ssid = "myssid";
const char* password = "mypasswd";
RTC_DATA_ATTR int bootCount = 1;

void setup()
{
  Serial.begin(115200);
  delay(100);
  Serial.println("Boot number: " + String(bootCount++));

  // WiFi
  WiFi.begin(ssid, password); 
  Serial.print("Connecting to WiFi.");
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(1000);
    Serial.print( "." );
  }
  Serial.println();
  Serial.println("Connected");

  // POST  
  HTTPClient http;
  int httpResponseCode;
  http.setReuse(false);
  do
  {
    http.begin("https://water-9dbfa.firebaseio.com/users/chris/readings.json", firebase_ca);
    http.addHeader("Content-Type", "application/json");
    httpResponseCode = http.POST("{ \"level\": " + String(bootCount) + ", \"timestamp\": {\".sv\": \"timestamp\"} }");
    if (httpResponseCode != 200)
    {
      Serial.println( "httpResponseCode: " + http.errorToString( httpResponseCode ) );
    }
    http.end();
    
  } while (httpResponseCode != 200);
  
  UBaseType_t uxHighWaterMark = uxTaskGetStackHighWaterMark( NULL );
  Serial.print( "Memory: " );
  Serial.print(uxHighWaterMark );
  Serial.print( " ");
  Serial.println( esp_get_free_heap_size() );
  Serial.println();

  delay(100);

  // sleep
  esp_deep_sleep_enable_timer_wakeup(20000000);
  esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
  esp_deep_sleep_start();
}

void loop()
{
  
}
```

chris-gunawardena, 2017-08-15:
Thanks @lonerzzz, even with your sketch with only the ssid and password changed it was getting stuck, might have something to do with the bad internet connection or the router here (I'm on vacation). I ran out of patience and changed the https call to a MQTT call which has been working for a few hours now doing calls every minute. I created a small server to forward the mqtt request to firebase https and reply back with the response via mqtt [here](https://github.com/chris-gunawardena/http2mqtt.js).  Thanks again for your help, I'll retest with https when I get home and reopen this issue if needed.

closed 2017-08-15 by chris-gunawardena

copercini, 2017-08-15:
@chris-gunawardena Let me try understand:
How it stuck?
The ESP32 stop of run? or just stops of call firebase URL?

chris-gunawardena, 2017-08-17:
@copercini The sketch simply hangs on this line:
`int httpResponseCode = http.POST("{ \"level\": " + String(bootCount) + ", \"timestamp\": {\".sv\": \"timestamp\"} }");` I tried switching on the onboard LED before this line and switching it off after, the LED stays lit which means it never returns from the http.POST().

aallman, 2018-12-19:
@chris-gunawardena Did you ever figure out why the http.POST call got stuck?  I've got a similar situation but it only gets stuck on the call if the previous post returned a read timeout error (-11).

leonbrag, 2020-06-05:
I am using ArduinoHttpClient and it returns the same error on HTTPS post:  9984

Using Uno WiFi rev 2 with certs uploaded

atanisoft, 2020-06-05:
@leonbrag I don't believe the  Uno WiFi Rev 2 uses an esp32. I would suggest opening an issue against the the Uno WiFi repository.

leonbrag, 2020-06-05:
@atanisoft  thank you, I opened against ArduinoHttpClient. Using WiFiNINA with the ssl does work.