espressif/arduino-esp32 #581: Missing files in tools directory of Arduino-ESP32 set

rrobinet, 2017-08-14:
Hi,
After installing the current version of the Arduino-ESP32 set.
I have 2 compilation errors due to missing files

fork/exec ... /hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++: no such file or directory
Error compiling for board WEMOS LOLIN32.

fork/exec ... /hardware/espressif/esp32/tools/esptool: no such file or directory
Error compiling for board WEMOS LOLIN32.

By copying these 2 files from a previous version, everything  look OK 

Robert

chris-gunawardena, 2017-08-14:
Did you follow [these](https://github.com/espressif/arduino-esp32#installation-instructions) instructions? Specially the `python get.py` bit?

rrobinet, 2017-08-15:
OK I didn't run the python script.
Thanks for the update
Robert

closed 2017-08-15 by rrobinet