espressif/arduino-esp32 #582: ESP32-D2WD partition table does not get generated correctly (+verbose doesnt work?)

demultiplexer, 2017-08-15:
### Hardware:
Custom board

### Description:
For much more detail: please read https://www.esp32.com/viewtopic.php?f=19&t=2708

It seems like Arduino does not generate a correct partition table for ESP32-D2WD (or, probably, any other <4MB configuration). I had to manually change default.csv with other partition information to get any sketch to work. Frustratingly as well, contrary to when compiling with ESP-IDF, the compiler also seems to ignore compiler verbosity flags and gives no indication whatsoever that this could be a partitioning error; it simply outputs 'user code done' when trying to run an erroneous flash. This made me waste way more time on this problem than I would have, if it were obvious that I was flashing a broken partition table.

me-no-dev, 2017-08-17:
working with 2MB flash would mean that some things need to be disabled (like OTA and SPIFFS). As for IDF errors, they are enabled at level ERROR for the libs and NONE for bootloader in order to emit less messages on boot.

demultiplexer, 2017-08-17:
I see. Now, if I would make my own boards.txt for this hardware, is it documented somewhere how I would go about configuring it not to have the OTA and SPIFFS partitions AND notifying the user of this? -D2WD has serious board space advantages and is well large enough to hold the types of programs it will run, and I am building this as an open source project, so having Arduino support would be nice.

For reference, how large is the OTA firmware? How tight is it to pack it in anyway?

I don't quite understand why there is a menu option for a bunch of error levels, but - if I understand this correctly - these are simply ignored and always set to ERROR/NONE? Why have that option in boards.txt and the interface then? Is this a non-ESP32 artefact?

Thanks, by the way, for your great work on the project :)

me-no-dev, 2017-08-17:
The debug levels are for Arduino's debug system (log_e, log_d and so on). Those are based purely on defines so if not enabled, code will not be compiled. IDF works a bit different. We can add another partition scheme for 2MB flash and give it as an option for the dev board. How does that sound?

demultiplexer, 2017-08-17:
Yeah, that is probably the easiest immediate solution.

me-no-dev, 2017-08-17:
Done! https://github.com/espressif/arduino-esp32/commit/856823ef195ef5c28516395356e8a30f8af8e917 Please close the ticket if you find it that works :)

closed 2017-08-17 by demultiplexer